undera/pylgbst #1: gattlib

g255, 2017-12-28:
HI, We'd love to use some of your code with my son on his Lego Boost. Unfortunately we're having some issues with dependencies using python 3.6. Apologies if this is not the correct forum, we're new to this and will learn quickly!

$ python3 demo.py
Traceback (most recent call last):
  File "demo.py", line 4, in <module>
    from pylgbst import *
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pylgbst/__init__.py", line 1, in <module>
    from pylgbst.movehub import *
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pylgbst/movehub.py", line 5, in <module>
    from pylgbst.comms import BLEConnection
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pylgbst/comms.py", line 11, in <module>
    from gattlib import DiscoveryService, GATTRequester
ModuleNotFoundError: No module named 'gattlib'

undera, 2017-12-28:
Did you install it using `pip install pylgbst`? It should have installed gattlib for you...

undera, 2017-12-28:
In general it's library dependency problem. You need to have `gattlib`. I usually install it like `pip install gattlib`.

g255, 2017-12-28:
ah! thanks, I tried a few things based on what you said.. but now I think its a compatibility problem with mac high sierra (10.13.1) - perhaps I should compile gattlib locally (this is a little beyond me at the moment!)  the error I get is:

$ pip install gattlib
Collecting gattlib
  Using cached gattlib-0.20150805.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/pg/xkjd9zl934gddp91v_3blk_c0000gn/T/pip-build-dTFt5d/gattlib/setup.py", line 48, in <module>
        raise OSError("Not supported OS")
    OSError: Not supported OS

undera, 2017-12-28:
Sorry, I can't help with gattlib on Mac, I'm using Linux

undera, 2017-12-28:
For reference: 
- https://github.com/IanHarvey/bluepy/issues/44
- https://bitbucket.org/OscarAcena/pygattlib/issues/12/add-osx-support

lmerory, 2017-12-28:
after quite a bit of wrangling of dependencies (libboost, libboost-thread-dev...) i got gattlib installed with pip.. so now i get:

  File "test1.py", line 2, in <module>
    hub = MoveHub()
  File "/home/vagrant/boost/pylgbst/pylgbst/movehub.py", line 60, in __init__
    self.connection.set_notify_handler(self._notify)
  File "/home/vagrant/boost/pylgbst/pylgbst/comms.py", line 101, in set_notify_handler
    raise RuntimeError("No requester available")

i'm running the code on a ubuntu virtualbox on a windows laptop with BLE 4.2, I can run Lego Boost app on it without problems so I think bluetooth works ok.. any suggestions?

undera, 2017-12-29:
Looks like a bug, I am reproducing it. I will fix it right away.

undera, 2017-12-29:
Actually, can you try invoking it like this: `hub=MoveHub(BLEConnection().connect())` ? It will workaround this bug.

undera, 2017-12-29:
The actual fix for the bug is in master branch of source now: https://github.com/undera/pylgbst/commit/dc863f11b1db24afa94d50964971961a100c7f94

fettikz, 2018-01-01:
Hi, after this fix I get:
    hub = MoveHub()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/movehub.py", line 40, in __init__
    connection = BLEConnection().connect()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 85, in connect
    devices = service.discover(1)
RuntimeError: Set scan parameters failed (are you root?)

undera, 2018-01-01:
Yes, you need to use it with root permission to access bluetooth device. Use `sudo` to invoke it.

atravert, 2018-01-01:
Hello,
First, many thanks for this library ! 
On my side (running on raspberry pi 3 / raspbian), I have also 
(i) installed gattlib 
(ii) patched the BLEConnection().connect() issue
(iii) run a test file using `sudo`

and I got the following error:
```
pi@raspberrypi:~ $ sudo python3.5 ./mhub.py
Exception in thread Notify queue dispatcher:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/comms.py", line 50, in _dispatch_notifications
    self.notification_sink(handle, data)
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/movehub.py", line 91, in _notify
    log.debug("Notification on %s: %s", handle, str2hex(orig))
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/constants.py", line 14, in str2hex
    return binascii.hexlify(data).decode("utf8")
TypeError: a bytes-like object is required, not 'str'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/comms.py", line 52, in _dispatch_notifications
    log.warning("Data was: %s", str2hex(data))
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/constants.py", line 14, in str2hex
    return binascii.hexlify(data).decode("utf8")
TypeError: a bytes-like object is required, not 'str'

WARNING:movehub:Got only these devices: (None, None, None, None, None, Button on port 0, None, None)
Traceback (most recent call last):
  File "./mhub.py", line 8, in <module>
    hub = MoveHub()
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/movehub.py", line 62, in __init__
    self._wait_for_devices()
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/movehub.py", line 81, in _wait_for_devices
    raise RuntimeError("Failed to obtain all builtin devices")
RuntimeError: Failed to obtain all builtin devices

```

any hint ?

undera, 2018-01-02:
Can you try this with python 2.7?

atravert, 2018-01-02:
Many thanks: it works fine with python 2.7 !

Do you have an idea as to why there's a problem with python 3.5 ?   

NB: I had some issues in building gattlib for python 3.5: I have used libbboost_python-3.5.so instead of the  default (probably libbboost_python-3.4.so).

undera, 2018-01-02:
This might be the reason for the failures...

closed 2018-01-02 by undera

rikponne, 2018-05-21:
After a lot of trying I've finally managed to install gattlib on my Raspberry Pi 3B / Raspian. However, using Python 3.5, I am now getting the following:

from pylgbst import MoveHub
hub = MoveHub()
Traceback (most recent call last):
  File "<pyshell#10>", line 1, in <module>
    hub = MoveHub()
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/movehub.py", line 60, in __init__
    self.connection.set_notify_handler(self._notify)
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/comms.py", line 101, in set_notify_handler
    raise RuntimeError("No requester available")
RuntimeError: No requester available

And with the workaround mentioned above:

hub=MoveHub(BLEConnection().connect())
Traceback (most recent call last):
  File "<pyshell#11>", line 1, in <module>
    hub=MoveHub(BLEConnection().connect())
NameError: name 'BLEConnection' is not defined

Maybe the same issue as atravert? Anyway, any suggestion is welcome. Quite new to this.

rikponne, 2018-05-21:
Okay, thanks! Working now, with Python 3.5.