undera/pylgbst #4: Feature request: Support of BlueGiga adapters

birgander2, 2018-01-26:
It would be really cool to additionally support BlueGiga BLED112 adpators through [pygatt](https://github.com/peplin/pygatt/tree/master/pygatt) just like the [pyb00st package](https://github.com/JorgePe/pyb00st). This would allow the usuage of your package for people who have only such an adaptor (like me) and enables support for Windows, where pygattlib is not available.

Probably not that complex if you analyse a bit the implementation in pyb00st. If I find some time, I'll maybe try if I can get something running.

undera, 2018-01-28:
This is great idea. I welcome any code contributions here!

undera, 2018-01-28:
Also, one thing I don't like about pygatt is that it relies on `gatttool` executable, which means everything will be much slower.

sidikalamini, 2018-02-07:
pygatt use gattool for BlueZ, for BLED112 will use BGAPI. not sure whether BGAPI is faster/slower though

undera, 2018-02-07:
It would be great if somebody would try implementing this and providing PR. My attempts were unsuccessfull.

birgander2, 2018-02-07:
I made some tries. I got the notifier running, but the wait_for_devices didn't work as expected. However, I must admit that I have very few experience with all this gatt/bluetooth stuff.

undera, 2018-02-07:
@birgander2 Can you show your code changes? It might be first step into right direction

netding, 2018-06-03:
Dear Andrey, 
congratualations to this very professional work. I plan to use the Lego Boost set to teach BLE communication, sensor technology and robotics at Reutlingen University.
So I adapted the BlueGiga part of the pygatt library to your pylgbst library. It took me some time, but now your library also runs on Windows 7 using the BLE112 dongle. Unfortunately gattlib als well as bluepy did not work for me on a Raspberry Pi3 and Raspberry Pi Zero W. So I switched for the Linux part to the gatt-python (gatt) library.
You can find the source code als well as a (sorry german language) student instruction on my web site https://www.tec.reutlingen-university.de/prof-mack/projekte/python-und-bluetooth-low-energy-lernen-mit-lego-boost/
I had to simplify your library a little bit to get it on a bachelor mechatronics student level ;-)
Thank you again for your great work on the Lego Boost robotics set! 
Stefan

undera, 2018-06-04:
That's cool, Stefan!
I will review your code for potential improvements of main repo

undera, 2018-07-18:
I have just released a big change with version 0.6, making changes from @netding code example.
 It allows now to use different backend libraries, not only `gattlib`. Can somebody please experiment with installing other libs and trying 4 available connection classes, possibly one of them will work for you?
I simply have no Windows OS machine, nor BlueGiga adapters...

closed 2019-02-13 by undera