undera/pylgbst #6: RuntimeError: Device is not responding!

npl12, 2018-03-09:
Hello,
I really like this library. I cloned it and executed it on a raspberry pi. 
I start the DebugServer with DebugServer(BLEConnection().connect()).start()

Most of the time everything works fine, and I can execute commands on the movehub.
But every once in a while I get the following error:

ERROR:comms:Failed to handle cmd: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 197, in _handle_conn
    self._handle_cmd(json.loads(line))
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 205, in _handle_cmd
    self.ble.write(cmd['handle'], unhexlify(cmd['data']))
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 115, in write
    return self.requester.write_by_handle(handle, data)
RuntimeError: Device is not responding!

In the issue before, you wrote that the battery could be an issue.
I checked the battery and even used brand new ones, but I still get the error.

Do you have any idea how to get rid of this error?

undera, 2018-03-09:
It's not because of battery itself, it is because of loss of signal. I also get this problem sometimes, but just reconnecting helps. I have no idea what makes it to lose BLE signal. It is obviously happening out of control for pylgbst library, just inside BLE communication layer.

undera, 2018-03-09:
It's duplicate of issue #2

closed 2018-03-09 by undera