undera/pylgbst #7: Discovering devices without result

dasoho, 2018-03-20:
Actually this is not an issue but I am hoping for help :-)

I am very happy that I have found your library, I am sure it will bring fun to my son and me :-)

I am trying to get it running on my Raspberry 3

What I did:
`sudo pip install gattlib`
Following depedencies were necessary to get gattlib compiled:
`sudo apt-get install libglib2.0-dev`
`sudo apt-get install libboost-all-dev`
`pip install gTTS`
`pip install https://github.com/undera/pylgbst/archive/0.5.tar.gz`

Python version is: Python 2.7.13

Starting on of the test programs while pressing the Move Hub button leads to:
```
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...
```
and loops infinite without finding the Move Hub.

Using the scan funtionality of the `blutoothctl `tool I can see the Move Hub:
```
[NEW] Device 90:A4:DE:F7:70:C9 Panasonic Viera
[NEW] Device E8:B1:FC:47:99:8C LENOVO
[NEW] Device 00:16:53:AC:DA:6B LEGO Move Hub
```

Strange thing is that when I use the command `hcitool scan` I do not see the Move Hub but the other BT devices:
```
pi@raspberrypi:~/legoboost $ hcitool scan
Scanning ...
	E8:B1:FC:47:99:8C	LENOVO
	90:A4:DE:F7:70:C9	Panasonic Viera
```
Is there anything I need to configure additionally to get library call `hub = MoveHub(connection)` finding the LEGO Move Hub?

Any advice is welcome to get the the problem on my side analyzed.

Many thanks,
Holger

undera, 2018-03-21:
Is `hci0` interface present on your computer?

undera, 2018-03-21:
I believe if `hcitool scan` can't find MoveHUB, then `pylgbst` also won't find it.

dasoho, 2018-03-21:
Yes, hci0 is the interface to the Rhaspberry 3 built-in BT hardware.
I really don't know what can prevent `hcitool scan` from discovering Movehub while finding the other BT devices in my living room?
Unfortunately, I do not have much experience with BT, T think I need to figure out what is the actual difference between `bluetoothctl` and `hcitool scan`  ....

undera, 2018-03-21:
Also, if you have alternative BLE adapter to attach to your Pi, try it. This looks like hardware+OS problem.

dasoho, 2018-03-21:
... many thanks for your feedback. If I get a solution I will post it here.

cpseager, 2018-03-21:
Move hub is a BLE device, so with hcitool

    sudo hcitool lescan

dasoho, 2018-03-21:
Thanks, sounds reasonable, I will give it a try tonight.

dasoho, 2018-03-21:
Just confirmed that `sudo hcitool lescan` does find the Movehub:
pi@raspberrypi:~ $ sudo hcitool lescan
LE Scan ...
00:16:53:AC:DA:6B (unknown)
00:16:53:AC:DA:6B LEGO Move Hub
Based on that I would assume that the BT subsystem on my Raspberry is configured properly and I need to dig more into the direction how it's been accessed from python.

What I recognize is that the messages
```
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...
```
are being printed very fast (maybe each 10ms) but the comment in the code in `comms.py` says there is a 1 second timeout:

```
class BLEConnection(Connection):
    """
    Main transport class, uses real Bluetooth LE connection.
    Loops with timeout of 1 seconds to find device named "Lego MOVE Hub"

    :type requester: Requester
    """

    def __init__(self):
        super(BLEConnection, self).__init__()
        self.requester = None

    def connect(self, bt_iface_name='hci0', hub_mac=None):
        service = DiscoveryService(bt_iface_name)

        while not self.requester:
            log.info("Discovering devices using %s...", bt_iface_name)
            devices = service.discover(1)
            [...]
```
Is the timeout specified by `devices = service.discover(1)` maybe just referring to 1ms? This eventually isn't enough time to discover the movehub?
Is the `Discovering devices` message coming on your side also with high frequency < 10ms?
Or any other advice?

Thanks,
Holger

undera, 2018-03-21:
The timeout is exactly in seconds, you can find it here: https://bitbucket.org/OscarAcena/pygattlib/src/a858e8626a93cb9b4ad56f3fb980a6517a0702c6/src/gattservices.cpp?at=default&fileviewer=file-view-default

In my regular PC experiments, if there is no MoveHub enabled, I get poll messages exactly once a second.

The fact it responds very fast means that it's likely some kind of failure happens inside `discover()` that makes it to respond so fast.

PS. I'd suggest you to maximize logging verbosity for your troubleshooting. It's done like this in Python: `import logging; logging.basicConfig(level=logging.DEBUG);`

dasoho, 2018-03-22:
OK, thanks.
Seems that pygattlib does not to work properly on my system.
Hope that enabling logging sheds some light on the issue.
Will give it a try tonight.

dasoho, 2018-03-22:
Enabled logging as suggested but no significant hint.
Crosschecked with gatttool and got an immediate connection to the Movehub:
```
pi@raspberrypi:~/legoboost $ sudo hcitool lescan
LE Scan ...
00:16:53:AC:DA:6B (unknown)
00:16:53:AC:DA:6B LEGO Move Hub
pi@raspberrypi:~/legoboost $ gatttool -I
[                 ][LE]> connect 00:16:53:AC:DA:6B
Attempting to connect to 00:16:53:AC:DA:6B
Connection successful
```
... and the Movehub LED turned blue.

No idea why it doesn't work through pygattlib. Even tried a "clean" install by cloning the repository and compiling the code by
```
hg clone https://bitbucket.org/OscarAcena/pygattlib
cd pygattlib
sudo python setup.py install
```
but ended up with the same result.

For now I hit the pause button :-)

Thank you very much for your help.

Holger

cpseager, 2018-03-22:
Did you try to connect via bluetoothctl itself?

     [bluetooth]# power on
     [bluetooth]# connect 00:16:53:AC:DA:6B

see if it connects and then disconnect it again

     [bluetooth]# disconnect

I have a vague memory I had to do that with a different BLE device to make it a 'known device' on the pi before I could use it properly

dasoho, 2018-03-24:
Tried your suggestion and the connection via `bluetoothctl `works perfectly, but a subsequent connection through pygattlib is still without success.
I share the opinion of undera, there is some combatibility issue within discover() of pygattlib within my system as it returns immediately without respecting the specified timeout of 1 second.
On the other hand I have no idea how to solve it. The pygattlib code got compiled on my system so everything should fit together?!

Thanks for your suggestions.

undera, 2018-03-24:
You would need to use `gdb` to find exact reason of a failure... Starting simple Python script via gdb...

undera, 2018-07-18:
I have just released a big change with version 0.6. It allows you to use different backend libraries, not only `gattlib`. Can you please experiment with installing other libs and trying 4 available connection classes, possibly one of them will work for you?

dasoho, 2018-07-18:
Tank you for your effort. I am currently in vacation but I will try the new lib options as soon as I return early August.

undera, 2018-09-13:
I see "No module named pexpect" - can you try installing "pip install fexpect"? Then try again with pygatt.

Also, try just using `get_connection_auto()`, probably with first parameter of your BLE interface name. It defaults to `hci0`, but there is a chance your BLE interface is named differently on Pi?

dasoho, 2018-09-13:
I removed gatt just to make sure that `get_connection_auto()` is forced to use pygatt. With gatttool (which I think is used in pygatt) I already connected to the movehub successfully from command line (see comment further above).
As you recommended, I additionally installed fexpect and used `get_connection_auto()` .

Now I get the following error:

```
ERROR:pygatt.backends.gatttool.gatttool:Unexpected error when scanning: Set scan parameters failed: Operation not permitted

Traceback (most recent call last):
  File "test.py", line 13, in <module>
    conn=get_connection_auto()  # ! don't put this into `try` block
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 52, in get_connection_auto
    raise Exception("Failed to autodetect connection, make sure you have installed prerequisites")
Exception: Failed to autodetect connection, make sure you have installed prerequisites
```
I also tried using sudo but then I get a strange error:

```
pi@raspberrypi:~/pylgbst/pylgbst/examples $ sudo python test.py
Traceback (most recent call last):
  File "test.py", line 10, in <module>
    from pylgbst import get_connection_auto
ImportError: No module named pylgbst
```
Seems that the library is not properly installed for user root?

Again, any idea?

dasoho, 2018-09-13:
Just had a look into the pygatt code, they are using `hcitool lescan` which indeed requires root permission. (see https://github.com/undera/pylgbst/issues/7#issuecomment-375465196)
So, the question is why running the python script with root does not work?

undera, 2018-09-14:
Looks like some specifics with Python class search paths...
Did you install `pylgbst` using `sudo pip install ...`  or just `pip install`? 
If latter is the case, I'd uninstall non-root version and would try installing sudo way.

undera, 2018-09-14:
That's great! If you see that your experience can be integrated into project docs, feel free to suggest a PR!

closed 2018-09-14 by undera