undera/pylgbst #8: Renamed device does not connect

cpseager, 2018-03-23:
The Lego app allows you to rename the Move hub, As this lib currently searches by default name a 'renamed' hub will not be found. 

The workaround for now is to tweak the name here:

https://github.com/undera/pylgbst/blob/master/pylgbst/comms.py#L18

undera, 2018-03-23:
You don't have to patch the source code. It is Python, so you can just write in your code `pylgbst.comms.LEGO_MOVE_HUB="new hub name"

cpseager, 2018-03-23:
Thanks yes I understand that. However if you have more than one hub with different names you have to remember to keep changing it!

It is easier to not connect via name (if it can change) when the advert also contains other fixed unique data that does not change. For instance the lego app itself appears to connect via looking for the GAP data type  0x07 (Complete List of 128-bit Service Class UUID) not the 0x09 (Local Name). So it looks for this instead.

07 23 d1 bc ea 5f 78 23 16 de ef 12 12 23 16 00 00

undera, 2018-03-23:
I'm not sure I understood what you explained. Can you provide a PR illustrating better approach to identify move hub?

undera, 2019-03-19:
1 year with no reply :(

cpseager, 2019-03-19:
Hi , I moved onto other repos so haven't used this for some time, but what you need to reproduce this functionality - i.e. identify via 'fixed' UUID instead of the 'changeable' name - is shown here: 

https://github.com/nathankellenicki/node-poweredup/blob/master/src/boostmovehub.ts#L29

closed 2020-06-27 by undera