undera/pylgbst #10: pylgbst on RPI3 with Raspbian stretch

vwone, 2018-07-15:
Hi, Thank you for your great job and very usefull to learn python to children.
Thank to S. Mack for "Lego boost roboter steuern mit python unter windows oder linux" (in german :( )

But would it be possible to do a tutorial for RPI3 dummy users like me please?
I'm facing a lot of issues...

Let's start from the beginning.
I'm using RPI3 (enhanced with bluetooth) under Raspbian stretch.
I updated with : 
    **sudo apt-get update 
    sudo apt-get upgrade -y**

Next i failed to install gattlib:
 **sudo pip install gattlib**

Collecting gattlib
  Using cached https://files.pythonhosted.org/packages/be/2f/5b1aecec551b42b59d8b399ad444b5672972efb590ca83d784dbe616a3e1/gattlib-0.20150805.tar.gz
    Complete output from command python setup.py egg_info:
    Package glib-2.0 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `glib-2.0.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'glib-2.0' found
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-RfY5iW/gattlib/setup.py", line 12, in <module>
        "pkg-config --cflags glib-2.0".split()).decode('utf-8')
      File "/usr/lib/python2.7/subprocess.py", line 219, in check_output
        raise CalledProcessError(retcode, cmd, output=output)
    subprocess.CalledProcessError: Command '['pkg-config', '--cflags', 'glib-2.0']' returned non-zero exit status 1

    ----------------------------------------
**Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-RfY5iW/gattlib/**

This cmds didn't figure out the issue
    **sudo apt-get install glib-2.0**
    **sudo pip install --upgrade setuptools --user python**

The next step succeeded
    **pip install https://github.com/undera/pylgbst/archive/0.5.tar.gz**

but it seems to be necessary to make a:
    **git clone  https://github.com/undera/pylgbst/**
to run examples?

last step failed : 
    **python demo.py**
but of course => **ImportError: No module named gattlib**

thank's for your help.

vwone, 2018-07-15:
It seems that dependencies are missing. Its advised to download and compile from source. 

 **sudo apt-get install mercurial**

 **hg clone https://bitbucket.org/OscarAcena/pygattlib**

 **cd pygattlib**

 **cat DEPENDS**

Install all dependencies:
   **sudo apt-get install libboost-thread-dev libboost-python-dev libbluetooth-dev libglib2.0-dev python-dev**

now install gattlib like any other python module.
    **sudo python setup.py install
    sudo pip install gattlib**

Now:
**sudo python demo.py**
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...
...and so on

boost is not detected...
however i can detect my lego boost:
    **sudo bluetoothctl**
[NEW] Controller B8:27:EB:89:23:7E raspberrypi [default]
[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:89:23:7E Discovering: yes
[NEW] Device 2C:41:A1:13:41:2D LE-Bose Micro SoundLink
[NEW] Device 5C:F9:38:B1:2C:7A 5C-F9-38-B1-2C-7A
**[NEW] Device 00:16:53:AF:85:B1 LEGO Move Hub**

but this doesn't work
   **sudo hcitool scan**...nothing ?!

    **sudo gatttool -b 00:16:53:AF:85:B1 -I**
[00:16:53:AF:85:B1][LE]> connect
Attempting to connect to 00:16:53:AF:85:B1
Connection successful
[00:16:53:AF:85:B1][LE]>quit

Please how to run demo.py?
thanks

undera, 2018-07-16:
Hi,
After installing `glib-2.0` you should have tried to do `sudo pip install gattlib`

undera, 2018-07-16:
Try to get more debug info by changing this line inside `demo.py`: `logging.basicConfig(level=logging.INFO)`. Make it `DEBUG` instead of `INFO`.

Then, run it again, and share here the log you got.

undera, 2018-07-16:
Also, this is perhaps duplicate of #7 , read there for possible hints

vwone, 2018-07-16:
Hi,
thank you very much

Here is my feedback #7 etc...

**hcitool scan**
Scanning ...
        78:BD:BC:62:48:C5       [TV] UE48JU6410
        38:37:8B:4F:20:52       n/a
        D0:66:7B:2C:2C:AA       DTVBluetooth
but nothing about Lego MoveHub?!

hci0 seems up
pi@raspberrypi:~ $ **hciconfig -a**
hci0:   Type: Primary  Bus: UART
        BD Address: B8:27:EB:89:23:7E  ACL MTU: 1021:8  SCO MTU: 64:1
        **UP RUNNING**
        RX bytes:5417 acl:0 sco:0 events:97 errors:0
        TX bytes:4388 acl:0 sco:0 commands:70 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT
        Name: 'raspberrypi'
        Class: 0x6c0000
        Service Classes: Rendering, Capturing, Audio, Telephony
        Device Class: Miscellaneous,
        HCI Version: 4.1 (0x7)  Revision: 0x168
        LMP Version: 4.1 (0x7)  Subversion: 0x2209
        Manufacturer: Broadcom Corporation (15)

**sudo hcitool lescan**

LE Scan ...
00:16:53:AF:85:B1 (unknown)
00:16:53:AF:85:B1 LEGO Move Hub
7C:43:7C:36:47:D4 (unknown)
78:BD:BC:62:48:C5 [TV] UE48JU6410

In this case LEGO Move Hub is visible ?!
I notice that [TV] UE48JU6410 is still visible contrary to the LEGO Move Hub ?!

**sudo bluetoothctl**
[NEW] Controller B8:27:EB:89:23:7E raspberrypi [default]
[NEW] Device 00:16:53:AF:85:B1 LEGO Move Hub
[bluetooth]#
[bluetooth]# **power on**
Changing power on succeeded
[bluetooth]# **connect 00:16:53:AF:85:B1**
Attempting to connect to 00:16:53:AF:85:B1
[CHG] Device 00:16:53:AF:85:B1 Connected: yes
Connection successful
[NEW] Primary Service
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service0001
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
[NEW] Characteristic
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service0001/char0002
        00002a05-0000-1000-8000-00805f9b34fb
        Service Changed
[NEW] Descriptor
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service0001/char0002/desc0004
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Primary Service
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service000c
        00001623-1212-efde-1623-785feabcd123
        Vendor specific
[NEW] Characteristic
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service000c/char000d
        00001624-1212-efde-1623-785feabcd123
        Vendor specific
[NEW] Descriptor
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service000c/char000d/desc000f
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[CHG] Device 00:16:53:AF:85:B1 ServicesResolved: yes
[LEGO Move Hub]#
[LEGO Move Hub]#

this time the connection succeeded on the first try.
but i prevously enter in an another try:
**sudo bluetoothctl**
[bluetooth]# **agent on**
[bluetooth]# **default-agent**
[bluetooth]# **scan on**
[bluetooth]# **connect 00:16:53:AF:85:B1**

then 
[LEGO Move Hub]# **quit**
[DEL] Controller B8:27:EB:89:23:7E raspberrypi [default]
[DEL] Primary Service
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service0001
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
[DEL] Characteristic
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service0001/char0002
        00002a05-0000-1000-8000-00805f9b34fb
        Service Changed
[DEL] Descriptor
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service0001/char0002/desc0004
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[DEL] Primary Service
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service000c
        00001623-1212-efde-1623-785feabcd123
        Vendor specific
[DEL] Characteristic
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service000c/char000d
        00001624-1212-efde-1623-785feabcd123
        Vendor specific
[DEL] Descriptor
        /org/bluez/hci0/dev_00_16_53_AF_85_B1/service000c/char000d/desc000f
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration

 finally:
**sudo python demo.py**
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...
INFO:comms:Discovering devices using hci0...

and here is the result of changing this line inside demo.py
logging.basicConfig(level=logging.INFO) => logging.basicConfig(level=logging.DEBUG)

 **sudo python demo.py > out 2>&1**

WARNING:root:Failed to use debug server: Traceback (most recent call last):
  File "demo.py", line 186, in <module>
    connection = DebugServerConnection()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 213, in __init__
    self.sock.connect(('localhost', port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

INFO:comms:Discovering devices using hci0...
DEBUG:comms:Devices: {}
INFO:comms:Discovering devices using hci0...
DEBUG:comms:Devices: {}
...
Traceback (most recent call last):
  File "demo.py", line 189, in <module>
    connection = BLEConnection().connect()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 85, in connect
    devices = service.discover(1)
KeyboardInterrupt

:(
according to #1 it seems like that Rikponne and Atravert have succeeded connecting Boost with RPI 3
 but I do not understand how.

Atravert  : " (ii) patched the BLEConnection().connect() issue ???
I have used libbboost_python-3.5.so instead of the default (probably libbboost_python-3.4.so). ???"

All for this evening.
Thanks.

vwone, 2018-07-17:
Hi,
I hope @atravert or @rikponne could have a suggestion?
Thanks

undera, 2018-07-18:
I have just released a big change with version 0.6. It allows you to use different backend libraries, not only `gattlib`. Can you please experiment with installing other libs and trying 4 available connection classes, possibly one of them will work for you?

vwone, 2018-07-18:
**pip uninstall pylgbst**
Uninstalling pylgbst-0.4:
  /home/pi/.local/lib/python2.7/site-packages/pylgbst-0.4.egg-info
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.pyc
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/comms.py
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/comms.pyc
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/constants.py
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/constants.pyc
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/movehub.py
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/movehub.pyc
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/peripherals.py
  /home/pi/.local/lib/python2.7/site-packages/pylgbst/peripherals.pyc
Proceed (y/n)? y
  Successfully uninstalled pylgbst-0.4

**pip install https://github.com/undera/pylgbst/archive/0.6.tar.gz**
Collecting https://github.com/undera/pylgbst/archive/0.6.tar.gz
  Downloading https://github.com/undera/pylgbst/archive/0.6.tar.gz
     / 40kB 350kB/s
Installing collected packages: pylgbst
  Running setup.py install for pylgbst ... done
Successfully installed pylgbst-0.6

**sudo python demo.py > out 2>&1**

WARNING:root:Failed to use debug server: Traceback (most recent call last):
  File "demo.py", line 186, in <module>
    connection = DebugServerConnection()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 213, in __init__
    self.sock.connect(('localhost', port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

INFO:comms:Discovering devices using hci0...
DEBUG:comms:Devices: {}
INFO:comms:Discovering devices using hci0...
DEBUG:comms:Devices: {}
Traceback (most recent call last):
  File "demo.py", line 189, in <module>
    connection = BLEConnection().connect()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 85, in connect
    devices = service.discover(1)
KeyboardInterrupt

installing other libs and trying 4 available connection classes???

Retry #1
**pip install gattlib**
Collecting gattlib
  Using cached https://files.pythonhosted.org/packages/be/2f/5b1aecec551b42b59d8b399ad444b5672972efb590ca83d784dbe616a3e1/gattlib-0.20150805.tar.gz
Building wheels for collected packages: gattlib
  Running setup.py bdist_wheel for gattlib ... done
  Stored in directory: /home/pi/.cache/pip/wheels/a9/c7/5a/799e5065ceb9e39ed2be02f9899cdc84fbb32d5f54e0055838
Successfully built gattlib
Installing collected packages: gattlib
Successfully installed gattlib-0.20150805

Patched connection issue:
**hub=MoveHub(connection) => hub=MoveHub(BLEConnection().connect())**
 **sudo python demo.py**
WARNING:root:Failed to use debug server: Traceback (most recent call last):
  File "demo.py", line 186, in <module>
    connection = DebugServerConnection()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 213, in __init__
    self.sock.connect(('localhost', port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

INFO:comms:Discovering devices using hci0...
DEBUG:comms:Devices: {}
INFO:comms:Discovering devices using hci0...
DEBUG:comms:Devices: {}
INFO:comms:Discovering devices using hci0...
Traceback (most recent call last):
  File "demo.py", line 189, in <module>
    connection = BLEConnection().connect()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms.py", line 85, in connect
    devices = service.discover(1)
KeyboardInterrupt

**sudo pip install gatt**
Collecting gatt
  Downloading https://files.pythonhosted.org/packages/96/d0/d66154053d5b47996731d80ee66f65bdf7b790258addc0b6a5f50bcc3579/gatt-0.2.7.tar.gz
Building wheels for collected packages: gatt
  Running setup.py bdist_wheel for gatt ... done
  Stored in directory: /root/.cache/pip/wheels/17/83/e3/72f896e9a4f0cd046ae413d4c6a1033e7d11b6adfed46fae78
Successfully built gatt
Installing collected packages: gatt
**Successfully installed gatt-0.2.**7

/pylgbst/tests $ **sudo python test_gatt.py**
Traceback (most recent call last):
  File "test_gatt.py", line 7, in <module>
    from pylgbst.comms_gatt import CustomDevice
ImportError: No module named comms_gatt

 **sudo pip install comms_gatt**
Collecting comms_gatt
  Could not find a version that satisfies the requirement comms_gatt (from versions: )
No matching distribution found for comms_gatt

/pylgbst/tests $ **sudo python test_movehub.py**
Traceback (most recent call last):
  File "test_movehub.py", line 6, in <module>
    from tests import log, HubMock, ConnectionMock, Thread
ImportError: No module named tests

/pylgbst/tests $ **sudo python test_pygatt.py**
Traceback (most recent call last):
  File "test_pygatt.py", line 9, in <module>
    from pylgbst.comms_pygatt import GattoolConnection
ImportError: No module named comms_pygatt

/pylgbst/examples $ **sudo python demo.py**
Traceback (most recent call last):
  File "demo.py", line 191, in <module>
    connection = get_connection_auto()
NameError: name 'get_connection_auto' is not defined

**I wrote this  : LegoBoost.py**
```
from pylgbst import MoveHub
hub = MoveHub()
for device in hub.devices:
        print(device)
```

**sudo python LegoBoost.py**
1
2
50
55
56
57
58
59
60

**so i continued with this code :** 
```
from pylgbst import MoveHub
from pylgbst.comms import BLEConnection
from time import sleep

hub = MoveHub(BLEConnection().connect())
sleep(60)

```
**sudo python LegoBoost_2.py**

connection succeeded for 60 [s]

It seems that i must have to learn python by stripping step by step all your code ...
to be continued

undera, 2018-07-19:
All your attempts look strange, looks like your installation did not upgrade somehow.
THe thing is `BLEConnection` does not exist anymore, and `demo.py` has been updated to reflect that. `from pylgbst import MoveHub` can't work anymore, because you can only import `from pylgbst.movehub import MoveHub`.

This makes me think your `pip install ...` did not succeed...

vwone, 2018-07-22:
sorry, I'm absent for a vacation ... resumed in SeptemberBack soon ;)____________________De : "Andrey Pokhilko" Date : 19 juil. 2018 à 08:16:18À : undera/pylgbstCc : vpw@neuf.fr, author@noreply.github.comSujet : Re: [undera/pylgbst] pylgbst on RPI3 with Raspbian stretch (#10)All your attempts look strange, looks like your installation did not upgrade somehow.
THe thing is BLEConnection does not exist anymore, and demo.py has been updated to reflect that. from pylgbst import MoveHub can't work anymore, because you can only import from pylgbst.movehub import MoveHub.
This makes me think your pip install ... did not succeed...

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/undera/pylgbst/issues/10#issuecomment-406166510",
"url": "https://github.com/undera/pylgbst/issues/10#issuecomment-406166510",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [undera/pylgbst] pylgbst on RPI3 with Raspbian stretch (#10)",
"sections": [
{
"text": "",
"activityTitle": "**Andrey Pokhilko**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@undera",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"undera/pylgbst\",\n\"issueId\": 10,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"undera/pylgbst\",\n\"issueId\": 10\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/undera/pylgbst/issues/10#issuecomment-406166510"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 356608518\n}"
}
],
"themeColor": "26292E"
}
]

vwone, 2018-08-20:
Update bluetooth backend :
**pip install --upgrade pygatt gatt gattlib**

remove previous pylgbst package 
**pip unsinstall pylgbst**

 **pip install https://github.com/undera/pylgbst/archive/0.8.tar.gz**
Collecting https://github.com/undera/pylgbst/archive/0.8.tar.gz
  Downloading https://github.com/undera/pylgbst/archive/0.8.tar.gz
     / 40kB 371kB/s
Installing collected packages: pylgbst
  Running setup.py install for pylgbst ... done
Successfully installed pylgbst-0.8

**pip show pylgbst**
Name: pylgbst
Version: 0.8
Summary: Python library to interact with LEGO Move Hub (from Lego BOOST set)
Home-page: UNKNOWN
Author: Andrey Pokhilko
Author-email: apc4@ya.ru
License: UNKNOWN
Location: /home/pi/.local/lib/python2.7/site-packages
Requires:

 **python demo.py**
Traceback (most recent call last):
  File "demo.py", line 5, in <module>
    from pylgbst import *
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 4, in <module>
    from pylgbst.comms import DebugServer
ImportError: No module named comms

**sudo python demo.py**
Traceback (most recent call last):
  File "demo.py", line 191, in <module>
    connection = get_connection_auto()
NameError: name 'get_connection_auto' is not defined

undera, 2018-08-20:
You need to import `get_connection_auto()` in your code to use it

vwone, 2018-08-21:
so it seems that i must use sudo,
on the other hand I do not understand why I have to modify your file "demo.py"?
but I added this line to your own code "demo.py":

import time
from time import sleep

from pylgbst import *
**from pylgbst import get_connection_auto**
from pylgbst.comms import DebugServerConnection
from pylgbst.movehub import MoveHub, COLORS, COLOR_BLACK
from pylgbst.peripherals import EncodedMotor, TiltSensor, Amperage, Voltage

and i got this result : 
Traceback (most recent call last):
  File "My_Demo.py", line 6, in <module>
    from pylgbst import get_connection_auto
ImportError: cannot import name get_connection_auto

commit f078d188 (2019-01-16): Fix #10

Fixes #10
closed 2019-02-13 by undera