undera/pylgbst #12: Missing contents of pylgbst.comms?

talbertc-usgs, 2018-08-12:
I just installed the 0.7 release on Windows but I'm getting an error:
```
C:\Anaconda3\envs\boost\python.exe C:/Anaconda3/envs/boost/Lib/site-packages/pylgbst/__init__.py
Traceback (most recent call last):
  File "C:/Anaconda3/envs/boost/Lib/site-packages/pylgbst/__init__.py", line 4, in <module>
    from pylgbst.comms import DebugServer
  File "C:\Anaconda3\envs\boost\lib\site-packages\pylgbst\__init__.py", line 4, in <module>
    from pylgbst.comms import DebugServer
ImportError: cannot import name 'DebugServer'
```
It appears that the contents of https://github.com/undera/pylgbst/blob/master/pylgbst/comms.py  is indeed empty.  Did this get deleted by accident?

Excited to try out this library as a tool for teaching Python coding to my daughter.  Thank you for building this and your excellent videos.

undera, 2018-08-12:
Thanks for reporting this.
I have made updated release 0.8 that should work fine. Let me know if it helps.
BTW  `DebugServer` is currently in a broken state, just use `get_connection_auto()` or simply use `MoveHub()`.

undera, 2018-08-12:
Huh, I see you use Windows. Do you use BlueGiga dongle? It's the only supported BLE device on Windows...

talbertc-usgs, 2018-08-13:
I just ordered one, but thanks for the heads up!

Colin Talbert
Data Manager, Analyst and Developer
US Geological Survey, Fort Collins Science Center
2150 Centre Ave. Bldg. C
Fort Collins, CO 80526
(970) 226-9425
http://orcid.org/0000-0002-9505-1876

Work schedule:
Monday, Thursday       -   7:00 - 5:00
Tuesday - Wednesday -   7:00 - 3:10
Friday (Telework)         -   7:00 - 5:00

dasoho, 2018-09-11:
I just wanted to continue with issue#7 by using the newest version 0.8 but I get the same error like the starter of this issue.
`
pi@raspberrypi:~/pylgbst/pylgbst/examples $ python test.py
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    from pylgbst.movehub import MoveHub
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/__init__.py", line 4, in <module>
    from pylgbst.comms import DebugServer
ImportError: No module named comms
`
The error happens already in the very first line when trying to import MoveHub.
Any further advice?

closed 2019-02-13 by undera