undera/pylgbst #13: Unable to run MoveHub() in multiple environments

shashankp, 2018-11-10:
On Windows:

```
D:\code\boost\pylgbst>pip install https://github.com/undera/pylgbst/archive/0.9.tar.gz
Collecting https://github.com/undera/pylgbst/archive/0.9.tar.gz
  Downloading https://github.com/undera/pylgbst/archive/0.9.tar.gz
     / 40kB 181kB/s
Installing collected packages: pylgbst
  Running setup.py install for pylgbst ... done
Successfully installed pylgbst-0.9
You are using pip version 9.0.1, however version 18.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

D:\code\boost\pylgbst>python
Python 3.6.3 |Anaconda custom (64-bit)| (default, Oct 15 2017, 03:27:45) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pylgbst.comms_gatt import GattConnection
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pylgbst.comms_gatt'
>>> from pylgbst.movehub import MoveHub
>>> hub = MoveHub()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\code\boost\pylgbst\pylgbst\movehub.py", line 41, in __init__
    connection = get_connection_auto()
  File "D:\code\boost\pylgbst\pylgbst\__init__.py", line 52, in get_connection_auto
    raise Exception("Failed to autodetect connection, make sure you have installed prerequisites")
Exception: Failed to autodetect connection, make sure you have installed prerequisites

```

On Windows Subsystem for Linux:
```

shash@PC:~$ pip3 install https://github.com/undera/pylgbst/archive/0.9.tar.gz
Collecting https://github.com/undera/pylgbst/archive/0.9.tar.gz
  Downloading https://github.com/undera/pylgbst/archive/0.9.tar.gz
     / 40kB 180kB/s
Installing collected packages: pylgbst
  Running setup.py install for pylgbst ... done
Successfully installed pylgbst-0.9
shash@PC:~$ python3
Python 3.6.6 (default, Sep 12 2018, 18:26:19)
[GCC 8.0.1 20180414 (experimental) [trunk revision 259383]] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pylgbst.movehub import MoveHub
>>> hub = MoveHub()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/shash/.local/lib/python3.6/site-packages/pylgbst/movehub.py", line 41, in __init__
    connection = get_connection_auto()
  File "/home/shash/.local/lib/python3.6/site-packages/pylgbst/__init__.py", line 52, in get_connection_auto
    raise Exception("Failed to autodetect connection, make sure you have installed prerequisites")

```
What are the pre-requisites as directed by the error message?
Is it OS related?

undera, 2018-11-11:
You should read the prerequisites section here: https://github.com/undera/pylgbst/blob/master/README.md#general-notes

On Windows, only BlueGiga adapters are supported.

romarro, 2018-11-14:
Be aware that the WSL doesn't have access to Bluetooth.
https://wpdev.uservoice.com/forums/266908-command-prompt-console-windows-subsystem-for-l/suggestions/33277174-add-suppor-to-bluetooth-on-wsl

shashankp, 2018-11-20:
@undera , @romarro 

thanks, will try it out on OSX

closed 2018-11-20 by shashankp

ludi81, 2022-01-15:
How can I connect with osx to the lego boost?
if i use python3 I receive the following error:

`
python3 demo.py 
22	INFO	root	Trying get_connection_bluepy
23	INFO	root	Trying get_connection_bluegiga
24	INFO	root	Trying get_connection_gatt
25	INFO	root	Trying get_connection_bleak
288	INFO	comms-bleak	Discovering devices... Press green button on Hub
10491	WARNING	hub	Got only these devices: (None, None, None, None, None, None, None)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 50, in <lambda>
    self._connection_thread = threading.Thread(target=lambda: asyncio.run(self._bleak_thread()))
  File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 60, in _bleak_thread
    await bleak.connect(self.hub_mac, self.hub_name)
  File "/usr/local/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 157, in connect
    raise ConnectionError('Device not found.')
ConnectionError: Device not found.
`

If I use python
`python demo.py 
Traceback (most recent call last):
  File "demo.py", line 5, in <module>
    from pylgbst import *
ImportError: No module named pylgbst
`

Does anyone have an idea?  The links above doesn't work any more.