undera/pylgbst #14: Any possibility for LEGO App-Controlled Batmobile 76112?

liangqi, 2018-11-22:
Haven't bought yet, just query. Thanks for your work.

undera, 2018-11-22:
Sorry, I don't have that set to experiment with it. But let me know if you'll try. I heard the protocol is more or less compatible.

undera, 2018-11-22:
FYI There's also https://github.com/nathankellenicki/node-poweredup working on this, using JS.

ElTentakel, 2018-12-08:
Hi, I tried to build a minimal class for the communication with the Powered Up Hub, but there are some issues:

```
import time
from struct import pack

from pylgbst.comms.cpygatt import GattoolConnection

from pylgbst import get_connection_auto
from pylgbst.constants import *
from pylgbst.utilities import str2hex

ENABLE_NOTIFICATIONS_HANDLE = 0x000f
ENABLE_NOTIFICATIONS_VALUE = b'\x01\x00'

class poweredup:
    def __init__(self, connection=None):
        if not connection:
            connection = get_connection_auto()

        self.connection = connection	
        self._wait_for_devices()
        
    def _wait_for_devices(self):
        self.connection.enable_notifications()

    def send(self, msg_type, payload):
        cmd = pack("<B", PACKET_VER) + pack("<B", msg_type) + payload
        self.connection.write(MOVE_HUB_HARDWARE_HANDLE, pack("<B", len(cmd) + 1) + cmd)

    def _notify(self, handle, data):
	print ("notification received")
		
    def shutdown(self):
	self.send(MSG_DEVICE_SHUTDOWN, b'')
			
    def write_to_hub(self, msg_type, port, params):
        cmd = pack("<B", port) + params
        self.send(msg_type, cmd)

conn = GattoolConnection("hci0")
print ("connect")
conn.connect("90:84:2B:03:FD:03")
print ("connected")

hub = poweredup(conn)

i = 0x60
#                                                              Speed			  Time	              Swing Mode
cmd = pack("<B", 0x11) + pack("<B", 0x60) + pack("<B", 0x00) + pack("<B", i) + pack("<B", 0x20) + pack("<B", 0x00)
hub.write_to_hub(MSG_SET_PORT_VAL, 0x00, cmd)
time.sleep(0.1)

# does not work at all:
while (i > 0):
	cmd = pack("<B", 0x11) + pack("<B", 0x60) + pack("<B", 0x00) + pack("<B", i) + pack("<B", 0x20) + pack("<B", 0x00)
	hub.write_to_hub(MSG_SET_PORT_VAL, 0x00, cmd)
	time.sleep(0.1)
	i-=0x10         
	

hub.shutdown()
```

The Issues are, that the connection is very unstable. Most of the time i get:

```
connect
ERROR:pygatt.backends.gatttool.gatttool:Timed out connecting to 90:84:2B:03:xx:xx after 5.0 seconds.
Traceback (most recent call last):
  File "test.py", line 40, in <module>
    conn.connect("90:84:2B:xx:xx")
  File "/home/xx/.local/lib/python2.7/site-packages/pylgbst/comms/cpygatt.py", line 39, in connect
    self._conn_hnd = adapter.connect(address)
  File "/home/xx/.local/lib/python2.7/site-packages/pygatt/backends/gatttool/gatttool.py", line 388, in connect
    raise NotConnectedError(message)
pygatt.exceptions.NotConnectedError: Timed out connecting to 90:84:2B:03:xx:xx after 5.0 seconds.

```

And if it works, it only works for the first commands. The connection closes immediately. But the motor moves for a fraction of a second - so fundamentally it works. 

Device Detection and Notifications seems not to work at all.

undera, 2018-12-08:
Connection instability is a real thing, but from what I heard, it's specific to devices you use. For example, my setup only has connection stability issues when battery is drained.

ElTentakel, 2018-12-08:
Interesting. The thing is, that the following command works quite well, but disconnects also after some seconds:
```
 gatttool -b 90:84:2B:03:xx:xx --char-write-req --handle 0x0e --value 0800810011510020
```
and connecting with gattool works completely fine:

```
gatttool -b 90:84:2B:03:xx:xx -I
[90:84:2B:03:xx:xx][LE]> connect
Attempting to connect to 90:84:2B:03:xx:xx
[90:84:2B:03:xx:xx][LE]> primary
Command Failed: Disconnected
Connection successful
[90:84:2B:03:xx:xx][LE]> primary
attr handle: 0x0001, end grp handle: 0x0004 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x0005, end grp handle: 0x000b uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x000c, end grp handle: 0x000f uuid: 00001623-1212-efde-1623-785feabcd123
[90:84:2B:03:xx:xx][LE]> 
```

ElTentakel, 2018-12-09:
So i tried to sniff the traffic between the BT remote and the Hub. It seems that those devices have a different protocol and also they are much more chatty. A message is send multiple times a second. Is it possible, that the connection has to be kept alive actively?

undera, 2018-12-09:
It's BLE, so connection is on all the time. You should join Slack channel of Lego Boost and read history there. One guy was experimenting around train a lot, he shared some findings.

romarro, 2018-12-10:
here is a link to a js library for PoweredUp devices (includes Boost and the other ones as well):
https://github.com/nathankellenicki/node-poweredup

ElTentakel, 2018-12-12:
Thank you for your recommendation. I've already tested the library, but I would prefer a Python Library.

undera, 2018-12-13:
I'm all for adding PoweredUp support into `pylgbst` library, I just don't have device to validate it.
I'm ready to work with whoever able to contribute PR to support it. I have expectation (coming from JS library owner sharings) that protocol is quite similar and there can be single library implementing all devices support.

add30417, 2019-01-27:
I would also looking forward to use pylgbst together with PoweredUp/Train system. Node-poweredup working well, but I want to include this into a existing python project.
Do you have any time plans for it? If is only the missing HW issue, I could support you by the development.

undera, 2019-01-27:
I don't have plans to invest much time and I don't have hardware.
But I can accept some code contribution, assuming it's done with the same level of quality and architecture. For that, you'd need to first explain what and how you plan to change in the code and if everything fits the idea of pylgbst - implement it and bring for review.

undera, 2019-12-26:
Closing as stale

closed 2019-12-26 by undera