undera/pylgbst #15: TypeError: 'NoneType' object is not iterable

asbayer, 2018-11-29:
Hi, 
Do you have an idea how i get rid of that messages? 
I did everything you wrote on your page setting up the environment but get stuck with the demo. Here the output (demo.py is taken from you page):
/usr/bin/python2.7 /root/PycharmProjects/mypy/lego/demo.py
INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port /dev/ttyACM0
INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port /dev/ttyACM0
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port /dev/ttyACM0
INFO:pygatt.backends.bgapi.bgapi:Running receiver
INFO:pygatt.backends.bgapi.bgapi:Disabling advertising
INFO:comms-pygatt:Discovering devices...
INFO:pygatt.backends.bgapi.bgapi:Starting an active scan
INFO:pygatt.backends.bgapi.bgapi:Pausing for 1s to allow scan to complete
INFO:pygatt.backends.bgapi.bgapi:Stopping scan
INFO:pygatt.backends.bgapi.bgapi:Discovered 1 devices: [{'rssi': -55, 'packet_data': {'scan_response_packet': {'tx_power_level': bytearray(b'\x00'), 'slave_connection_interval_range': bytearray(b'\x10\x00 \x00'), 'complete_local_name': u'LEGO Move Hub'}, 'connectable_advertisement_packet': {'complete_list_128-bit_service_class_uuids': [u'0x00:00:16:23:12:12:EF:DE:16:23:78:5F:EA:BC:D1:23'], 'flags': bytearray(b'\x06'), 'manufacturer_specific_data': bytearray(b'\x97\x03\x00@\x06\xfeA\x00')}}, 'name': u'LEGO Move Hub', 'address': u'00:16:53:A5:5A:0B'}]
INFO:root:Found LEGO Move Hub at 00:16:53:A5:5A:0B
INFO:pygatt.backends.bgapi.bgapi:Connecting to device at address 00:16:53:A5:5A:0B (timeout 5s)
INFO:pygatt.backends.bgapi.bgapi:Connection status: handle=0x0, flags=(11, 90, 165, 83, 22, 0), address=0x0b5aa5531600, connection interval=206.250000ms, timeout=830, latency=22 intervals, bonding=0x0
INFO:pygatt.backends.bgapi.bgapi:Connected to 00:16:53:A5:5A:0B
INFO:pygatt.backends.bgapi.bgapi:Fetching characteristics for connection 0
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/bgapi/bgapi.py", line 594, in _receive
    packet_type, args = self._lib.decode_packet(packet)
TypeError: 'NoneType' object is not iterable

Traceback (most recent call last):
  File "/root/PycharmProjects/mypy/lego/demo2.py", line 200, in <module>
    connection.disconnect()
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms/cpygatt.py", line 48, in disconnect
    self._conn_hnd.disconnect()
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/bgapi/device.py", line 21, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/bgapi/device.py", line 134, in disconnect
    self._backend.expect(ResponsePacketType.connection_disconnect)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/bgapi/bgapi.py", line 532, in expect
    return self.expect_any([expected], *args, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/bgapi/bgapi.py", line 568, in expect_any
    raise exc
pygatt.backends.bgapi.exceptions.ExpectedResponseTimeout: Timed out after 1.000000s waiting for [<ResponsePacketType.connection_disconnect: 29>]

Process finished with exit code 1

I would be very happy if you can give me a hint of how to deal with that.
Thank you very much, Andreas

undera, 2018-11-29:
This looks similar to https://github.com/peplin/pygatt/issues/196

What exactly do you have in your `demo.py` file?

asbayer, 2018-11-29:
I just copied your demo.py file
https://github.com/undera/pylgbst/blob/master/examples/demo.py

After the first trail I changed the line:
  connection = get_connection_auto()
to 
  connection = get_connection_bluegiga(hub_mac='00:16:53:A5:5A:0B')
Both scenarios prduce the output shown above. 
I'm using a Kali-Linux system.

undera, 2018-11-29:
That demo file has no job to do, because it has `demo_all()` commented out. Try uncommenting it and run again.

asbayer, 2018-11-29:
This I already did. But nothing changed

undera, 2018-11-29:
Ok, I see. Can you try with another backend BLE lib?

undera, 2018-11-29:
pygatt might require `sudo` to run properly

asbayer, 2018-11-29:
I am logged in as user root on that machine. But I also tried now:
`sudo python2.7 demo.py`
unfortunately with the same result. I will continue investigating

closed 2019-02-13 by undera