undera/pylgbst #16: App v1.7 features: Motor-Power & Acceleration

bjoernbundschuh, 2018-12-13:
I am using your code on some of my MOCs and get good results.
A month ago, Lego released a new APP-Version wich inlcudes programming blocks to set the motor power (e.g. the force, not the speed!) of each motor and other blocks to set the acceleration and deceleration of each motor.
Do you happen to know which command to use to do the same in python?

undera, 2018-12-13:
Does new app version require some firmware upgrade of HW part?

undera, 2018-12-13:
I guess acceleration is the mode that I've found in the past by brute-forcing motor: https://github.com/undera/pylgbst/blob/master/pylgbst/peripherals.py#L161

I just didn't find its parameters and their meanings.

Regarding "force" mode I doubt it's true, since applying more force for motor means rotating faster (simple physics). That might be just a programming GUI concept that uses known commands inside.

bjoernbundschuh, 2018-12-29:
Hi,

No, no HW upgrade required. At least I did not realize any upgrade.

Sorry for the delay of my response. Your message was caught in the SPAM filter.

Regards,

Björn Bundschuh
Am Stadtgraben 6
64823 Groß Umstadt

bjoernbundschuh, 2018-12-29:
Hi Andrey,

Thanks for the reply. Unfortunately your mail landed in SPAM so I only saw it today.

I will experiment with the acceleration bit.

With reduced force setting the motor does turn a bit slower but besides the reduced voltage, the motor turns off as soon as the amperage exceeds a certain value. So I guess you have to subscribe to the amperage of a particular motor.

Best regards,

Björn Bundschuh
Am Stadtgraben 6
64823 Groß Umstadt

undera, 2018-12-29:
Cool. Let me know if we can improve something in the library code.

bjoernbundschuh, 2019-01-11:
BTW: the 3 values of the ColorDistance Sensor in "STREAM_3_VALUES" mode are the RGB values.
However, read-out is quite unreliable depending on distance and ambient lighting.

closed 2019-02-13 by undera