undera/pylgbst #21: Invalid syntax in peripherals.py?

CaptainMojito, 2019-02-12:
Hello
I'm trying to get the library working with my Boost.
Just using the simple instantiate code gives me an error:

```
Traceback (most recent call last):
  File "demo.py", line 1, in <module>
    from pylgbst.movehub import MoveHub
  File "C:\Python37-32\lib\site-packages\pylgbst\movehub.py", line 7, in <module>
    from pylgbst.peripherals import Button, EncodedMotor, ColorDistanceSensor, LED, TiltSensor, Voltage, Peripheral, \
  File "C:\Python37-32\lib\site-packages\pylgbst\peripherals.py", line 64
    def subscribe(self, callback, mode, granularity=1, async=False):
                                                           ^
SyntaxError: invalid `syntax`

```
On the screen the up-arrow is pointing to the async=False parameter.
Any clues?

I'm working in Windows 10. I have Python 3.7.2. I have installed pygatt. I know the Boost is working because I can use the Windows 10 Lego Boost app.

undera, 2019-02-12:
This is something already fixed. I've made 0.10 release just now.

closed 2019-02-12 by undera

CaptainMojito, 2019-02-12:
Thanks, I now have Vernie running around my desk.
Just a note, the library still reports as version 0.9 when I do a pip list.
Also, another note in case anyone else finds this- I had to make a small tweak to the pygatt BGAPI library to get the BLED112 device to work correctly in Windows- the tweak is described in https://github.com/peplin/pygatt/issues/118 (adding a time delay).

undera, 2019-02-13:
Can you point to exact place in code where you put that `sleep()`?

CaptainMojito, 2019-02-14:
It goes in the bgapi.py file which is part of the pygatt library backends. The `sleep()` goes in the `start()` method right before the `open_serial_port()`. I basically did exactly as described in https://github.com/peplin/pygatt/issues/118#issuecomment-292058009 (however because I already had the pygatt library installed I already had the bgapi.py file so I didn't need to do the step described in https://github.com/peplin/pygatt/issues/118#issuecomment-293048737)