undera/pylgbst #24: run demo.py with BLED112 on Windows (PyCharm)

mivola, 2019-03-12:
Hi guys,

I'm a complete noon to Python and this whole BLE topic. I bought a BLED112 and think I got it installed correctly as I can connect to the MoveHub using "Bluegiga BLE GUI Tool".

Now I want to run the examples from this project using the PyCharm IDE. I cloned the pylgbst repo from my own fork and started playing around. Unfortunately I get an error when I try to run the demo.py (using right click => "Run/Debug 'demo'":

```
python.exe C:/git/pylgbst/examples/demo.py
INFO:root:Trying get_connection_bluepy
INFO:root:Trying get_connection_bluegiga
INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM3
INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM3
INFO:root:Trying get_connection_gatt
INFO:root:Trying get_connection_gattool
INFO:root:Trying get_connection_gattlib
Traceback (most recent call last):
  File "C:/git/pylgbst/examples/demo.py", line 188, in <module>
    connection = DebugServerConnection()
  File "C:\git\pylgbst\pylgbst\comms\__init__.py", line 146, in __init__
    self.sock.connect(('localhost', port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/git/pylgbst/examples/demo.py", line 191, in <module>
    connection = get_connection_auto()
  File "C:\git\pylgbst\pylgbst\__init__.py", line 58, in get_connection_auto
    raise Exception("Failed to autodetect connection, make sure you have installed prerequisites")
Exception: Failed to autodetect connection, make sure you have installed prerequisites

Process finished with exit code 1
```
According to the output from pygatt.backends.bgapi.bgapi I would assume the Bluetooth adapter is found, but it does not search for any devices. Is there anything I missed during setup/installation?

Thanks a lot!
Michael

undera, 2019-03-15:
Hi,
Indeed, in case of Windows you only need to try the bluegiga way. So the error you see is irrelevant on Windows. You need to concentrate on BLED112. I myself don't have Windows nor BLED112, so I can only guess. 
First thing to do, is instead of automatic connection, use `hub=MoveHub(get_connection_bluegiga())` always, so you won't get irrelevant messages. Then, to get potentially more diagnostics, add `import logging; logging.basicConfig(level=logging.DEBUG); ` before getting MoveHub connection.

Please share the results here.

mivola, 2019-03-15:
Hi Andrey,

I changed the logging and the initialization of the MoveHub as suggested. This changed the error message:
```
DEBUG:comms-pygatt:Trying to connect client to MoveHub with MAC: None
INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM3 - USB Serial Device (COM3)
DEBUG:pygatt.backends.bgapi.util:USB device: USB Serial Device (COM3) VID=0x2458 PID=0x0001 on COM3
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM3
INFO:pygatt.backends.bgapi.bgapi:Resetting and reconnecting to device for a clean environment
DEBUG:pygatt.backends.bgapi.bgapi:Opening connection to serial port (attempt 1)
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: COM3 - USB Serial Device (COM3)
DEBUG:pygatt.backends.bgapi.util:USB device: USB Serial Device (COM3) VID=0x2458 PID=0x0001 on COM3
INFO:pygatt.backends.bgapi.bgapi:Found BLED112 on serial port COM3
DEBUG:pygatt.backends.bgapi.bgapi:Failed to open serial port
Traceback (most recent call last):
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 159, in _open_serial_port
    timeout=0.25)
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\serial\serialwin32.py", line 78, in open
    self._reconfigure_port()
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
Traceback (most recent call last):
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 159, in _open_serial_port
    timeout=0.25)
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\serial\serialwin32.py", line 78, in open
    self._reconfigure_port()
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/git/pylgbst/examples/demo.py", line 194, in <module>
    connection = get_connection_bluegiga()
  File "C:\git\pylgbst\pylgbst\__init__.py", line 12, in get_connection_bluegiga
    return BlueGigaConnection().connect(hub_mac)
  File "C:\git\pylgbst\pylgbst\comms\cpygatt.py", line 27, in connect
    adapter.start()
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 202, in start
    self._open_serial_port()
  File "C:\Users\mvoigt\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pygatt\backends\bgapi\bgapi.py", line 170, in _open_serial_port
    "No BGAPI compatible device detected")
pygatt.exceptions.NotConnectedError: No BGAPI compatible device detected

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/git/pylgbst/examples/demo.py", line 199, in <module>
    connection.disconnect()
NameError: name 'connection' is not defined

Process finished with exit code 1
``` 

[... 44 lines cut ...]
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0f0004320117000000001000000010'
INFO:movehub:Attached LED on port LED
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0f00043a0128000000001000000002'
INFO:movehub:Attached TILT_SENSOR on port TILT_SENSOR
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0f00043b0115000200000002000000'
INFO:movehub:Attached AMPERAGE on port AMPERAGE
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0f00043c0114000200000002000000'
INFO:movehub:Attached VOLTAGE on port VOLTAGE
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0f0004010126000000001000000010'
INFO:movehub:Attached IMOTOR on port C
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'12000101064c45474f204d6f766520487562'
[... cut ...]
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0600453c6d0b'
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0600453c6c0b'
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0600453c6c0b'
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0600453c6c0b'
INFO:pygatt.device:Received notification on handle=0xe, value=0xb'0a00473c000000000000'
INFO:movehub:Voltage: 71%
```
So I have the feeling this is an issue in the related libraries for the BLE/socket/COM communication...

Regards
Michael

mivola, 2019-03-15:
OK, after some more investigation, this issue is caused by https://github.com/peplin/pygatt/issues/208 and the fix is here: https://github.com/peplin/pygatt/pull/181

closed 2019-03-15 by mivola