undera/pylgbst #34: Can't call motors from movehub

krishermans, 2019-08-16:
Hello

my son and I are exploring your beautiful library for Lego Boost. We have a Raspberry Pi and have already managed to make a connection with Lego Boost (Vernie), and used the Color/Distance sensor and changed de LED colors. Next step: motors, but here it fails.

When we try to access the motor_AB (or any other motor), we notice that the type of the variable is `NoneType` instead of `EncodedMotor`, hence we were unable to drive the motors. I suspect de motors are not found upon connecting. We got the following warnings when we connect:

```
WARNING:hub:Have not dedicated class for peripheral type 0x42 on port 0x46
WARNING:hub:Have not dedicated class for peripheral type 0x42 on port 0x46
WARNING:hub:Got only these devices: (None, None, None, LEDRGB on port 0x32, TiltSensor on port 0x3a, Current on port 0x3b, Voltage on port 0x3c)
```
It would much appreciated if you could help us out!

Thanks!
Kris
PS: we are using the standard Vernie robot for our experiments. When using the Lego app, everything works as expected.

undera, 2019-08-16:
Hi,
This might relate to https://github.com/undera/pylgbst/pull/33
A coupe of questions:

1. Did you already upgrade your MoveHub firmware via official app or not?
2. What was the way to install pylgbst library for your case?

krishermans, 2019-08-17:
Hi

I believe this could be the case. We updated MoveHub to latest firmware, and we are using pylgbst-1.0.1 (downloaded the zip file from releases page and installed with pip3).

I'm no Python expert: how could we use your latest sources?

thanks!
Kris

undera, 2019-08-17:
Please upgrade the library, I just made new release. Like this: `pip3 install -U https://github.com/undera/pylgbst/archive/1.1.0.tar.gz`
Let me know if it helps

krishermans, 2019-08-18:
I upgraded according to your instructions. However now we don't get a connection with the MoveHub (it gets discovererd, see logs below). I tried bluepy and pygatt (on Raspberry Pi 3). Pygatt did work before.

```
INFO:root:Trying get_connection_bluepy
INFO:root:Trying get_connection_bluegiga
INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
INFO:root:Trying get_connection_gatt
INFO:root:Trying get_connection_gattool
INFO:pygatt.backends.gatttool.gatttool:Running...
INFO:comms-pygatt:Discovering devices...
INFO:pygatt.backends.gatttool.gatttool:Starting BLE scan
INFO:pygatt.backends.gatttool.gatttool:Discovered 7A:06:02:7A:45:A6 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 58:90:D0:0A:C7:58 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 51:F4:59:FE:28:1E (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 72:6F:F3:F1:50:45 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 5F:EB:50:D9:4F:2F (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered C8:E0:EB:4B:8B:12 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 7C:94:AD:65:B7:24 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 74:05:6A:47:44:B5 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 49:AF:6A:6F:96:75 (None)
INFO:pygatt.backends.gatttool.gatttool:Found 9 BLE devices
INFO:comms-pygatt:Discovering devices...
INFO:pygatt.backends.gatttool.gatttool:Starting BLE scan
INFO:pygatt.backends.gatttool.gatttool:Discovered C6:2A:85:CD:2F:A8 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered name of C6:2A:85:CD:2F:A8 as MI Band 2
INFO:pygatt.backends.gatttool.gatttool:Discovered C8:E0:EB:4B:8B:12 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 74:05:6A:47:44:B5 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 7A:06:02:7A:45:A6 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 58:90:D0:0A:C7:58 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 7C:94:AD:65:B7:24 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 51:F4:59:FE:28:1E (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 5F:EB:50:D9:4F:2F (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 49:AF:6A:6F:96:75 (None)
INFO:pygatt.backends.gatttool.gatttool:Discovered 00:16:53:AA:D4:F5 (None)
**INFO:pygatt.backends.gatttool.gatttool:Discovered name of 00:16:53:AA:D4:F5 as LEGO Move Hub**
```

undera, 2019-08-18:
This is because I changed the lookup name. In my Hub, it's "Move Hub" and I thought that's for everyone. 
However, you can always connect to your hub by MAC, like this:
```python
hub=MoveHub(get_connection_auto(hub_mac='00:16:53:AA:D4:F5'))
```

krishermans, 2019-08-19:
It works now, thanks!

closed 2019-08-19 by krishermans