undera/pylgbst #37: V1.0 not connecting to the Raspberry Pi

Grumpy-Mike, 2019-12-09:
Previously I had a working system on the Raspberry Pi 3 and Pi3+
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

I install the latest version with
pip install https://github.com/undera/pylgbst/archive/1.1.0.tar.gz
And I also tried a pip3 install, both report a successful installation.
then I tried:-
from pylgbst.hub import MoveHub
hub = MoveHub()
for device in hub.peripherals:
    print(device)

nothing is detected, and I have to use Ctrl C to get out of it. The same sort of thing happens with the other demo codes. This also happened before and after a firmware update on my boost hub.
It is the same on the very latest Buster version with python3.7 on a Pi 3 and Pi 4, but with that there could be incompatibilities with the dependencies or OS.

undera, 2019-12-09:
Version 1.1.0 requires new version of hub firmware. Have you upgraded upgraded hub firmware? It is done via official mobile app.

undera, 2019-12-09:
If you did, please provide full log of startup.

Grumpy-Mike, 2019-12-09:
Thank you for your swift response. 
Yes it was after the firmware was updated to 2.0.00.0017. When I run the demo.py code from you, it appears it sees the Lego hub but does not connect. The attached file shows the printout that occurs constantly I hope there is enough of it to see what is happening.
[demoLog.txt](https://github.com/undera/pylgbst/files/3939385/demoLog.txt)

undera, 2019-12-09:
Can you switch it into debug logging and send log with DEBUG messages?
Also, it is known problem that if you have renamed your device, it will fail to connect. To workaround it, please use MAC address to connect to it. See https://github.com/undera/pylgbst#bluetooth-connection-options

Grumpy-Mike, 2019-12-09:
Thanks, I haven't changed any names. I can see you can switch it into debug mode by using
python -c "import logging; logging.basicConfig(level=logging.DEBUG); \
                import pylgbst; pylgbst.start_debug_server()"
But where do I enter this? I have tried the python console which gives me an syntax error and the terminal window which tells me it can't find stuff, both with and without sudo. Sorry to be a pain but I am more a hardware person than a systems level person.

undera, 2019-12-09:
https://github.com/undera/pylgbst/blob/master/examples/demo.py#L185
That line shows setting logging to INFO. You need DEBUG

Grumpy-Mike, 2019-12-09:
Thanks. Attached is what I get. Note again this was all working before upgrading to V1.0 so I am assuming the dependencies I installed to get the older version working is sufficient  for the new one.
[debugMode.txt](https://github.com/undera/pylgbst/files/3939833/debugMode.txt)

undera, 2019-12-09:
I got it. It's the problem of Hub name that I experienced in the past. Please install version 1.1.1, it should help to fix it.

Grumpy-Mike, 2019-12-09:
Thanks upgraded. This is the result of the debug in the demo.py program. Unfortunately still nothing works. When I try the peripherals detected on Hub I get nothing. When I try including GattConnection I get:-
`Traceback (most recent call last):
  File "/home/pi/MagPi/Boost/Part 3/Si-clops.py", line 9, in <module>
    from pylgbst.comms.cgatt import GattConnection
  File "/home/pi/.local/lib/python3.5/site-packages/pylgbst/comms/cgatt.py", line 6, in <module>
    import gatt
ImportError: No module named 'gatt'`
`
I had reinstalled pygat and it is now at version 4.0.5
File of the demo with debug attached
[debugMode.txt](https://github.com/undera/pylgbst/files/3939992/debugMode.txt)

undera, 2019-12-26:
Sorry for late reply.
From your log, I see all devices are attached successfully, but the calls after that don't work. What I see is Python sends command to read device name and never gets reply from device.
I have no idea what causes Hub to not reply, from Python side everything looks correct. The fact you have device attach messages tells that Hub is generally functional and is able to communicate.

One thing you can try is instead of MoveHub() class try to use basic Hub() class. It won't have any simplifications like MoveHub, but will still let you interact with the devices via `peripherals` field.

Another way is to subclass MoveHub and override `_report_status()` method to remove attempt to print device name.

NikolayDA, 2020-01-11:
> Thanks upgraded. This is the result of the debug in the demo.py program. Unfortunately still nothing works. When I try the peripherals detected on Hub I get nothing. When I try including [...]

Hello Grumpy-Mike and undera,

could you make any progress on this issue? I just started to try to run Mike Lego Boost Demo and ran into the same issue but I use different hardware (raspi zero w).

I'm have not much experience with python but if I can support the problem finding and solving please let me know.

br

Nikolay

undera, 2020-01-11:
I don't have access to RaspberryPI, sorry. The issue seem to need some deeper debugging and Python background.

NikolayDA, 2020-01-11:
Спасибо for you quick reply ;-)

I hope the community can help us out.

br nikolay

NikolayDA, 2020-01-12:
Hi undera,

just a quick update because I ran some test during the weekend. I was able to run your demo from pylgbst-master on my raspberry pi zero w and connect to my Lego Boost R2D2 and it started flashing the LED and dancing :-). 

One thing was strange because I had to run the several times until it worked. The first time i connected it I could the Lego Boost Hub in the list of bluetooth devices but the demo crashed. Than I repeated the demo (this time without pressing the button on the lego boost). Result was that the list didn't contain the Boost anymore (so as expected). Than I ran the demo again and pressed the button this time. Result was a successfully running demo. 

Thanks for providing this great lib!

undera, 2020-01-12:
Great

closed 2020-01-12 by undera

NikolayDA, 2020-02-05:
Hello Andrey,

I’d like to connect to one or several powered up hubs at the same time. Do you know if this is possible and could you please give me some hints which extensions woild be needed?

I guess I’d need to add a new class for the powered up hubs since it has only two ports, no internal motor and also no internal sensor?

Thanks again for the nice lib!

br

nikolay

undera, 2020-02-06:
You would just instantiate two Hub classes, with two different connections to them. I guess using hub_mac parameter for connection is the key.