undera/pylgbst #38: Cannot subscribe to both ANGLE and SPEED on boost motor

nakermann1973, 2019-12-23:
When trying to add a subscription to both ANGLE and SPEED for Motor_A, I get an error on the second subscription:

Code:

    self.hub.motor_A.subscribe(test_speed, mode=EncodedMotor.SENSOR_SPEED)
    self.hub.motor_A.subscribe(update_angle, mode=EncodedMotor.SENSOR_ANGLE)

Error:

    ValueError: Port is in active mode MsgPortInputFmtSingle({'payload': bytearray(b''), 'port': 0, 'mode': 1, 'upd_delta': 1, 'upd_enabled': 1}), unsubscribe all subscribers first

I should be able to subscribe to both of these at the same time.

undera, 2019-12-26:
When you say "I should be able to subscribe to both of these at the same time." - where do you take this assumption from? I don't see any possibility for this in LEGO docs. The port is able to be in one mode at the time and the mode defines the notification kind you will get.

nakermann1973, 2019-12-28:
I have assumed that I should be able to, since the lego boost app on my tablet shows speed and angle changing as the motors turn, if I set a variable input to the various sensors.  This assumption may not be valid, however.

undera, 2019-12-28:
If the official app can do it, then it exists. But I can't find anything like that in the mobile app. Can you instruct me how to find it?

Vouzamo, 2020-04-21:
This should be possible by setting up the port in combined input mode. I am close to having this setup in my C# SDK but not getting an acknowledgement reply when I unlock the port.

Perhaps we can solve together?

See: https://github.com/LEGO/lego-ble-wireless-protocol-docs/issues/22