undera/pylgbst #47: Alike #34, ... not dedicated class for peripheral type 0x42 on port 0x46

elegolo, 2020-01-27:
Hi!
Trying Your nice library! Struggled a little to get it work as I am not so very 'pythonic', but now it does.
I am using pylgbst version 1.1.2 via pip3 install pylgbst on Windows10 with BLED112 dongle and BGAPI.
Unfortunately, the motors do not react, the warning as entitled is shown on startup.
I use Vernie from examples, the hub is beeing activated but not reacting on ...motor_AB.start_power(...).
Other peripherals seem to work o.k.
MoveHub firmware (updated via original smartphone app) has reported version 2.0.00.0017.

привет
jo elegolo

undera, 2020-01-27:
Hi,
Can you try changing `start_power` to `start_speed`? Will it help?

elegolo, 2020-01-27:
Hi,
yes, working with start_speed :)
Большое спасибо!
elegolo

closed 2020-01-27 by elegolo