undera/pylgbst #49: Control External Motor and Motors A/B together

hofluk, 2020-02-04:
Is it possible to control the Motor connected to port D (or C) simultaneously with the Motors A and/or B? Did not find any such feature but in the Lego-App it appears that it is possible.

undera, 2020-02-05:
From the LEGO protocol documentation, I see only 2 motors can be grouped together. I didn't experiment to see if motor groups can also be grouped.

hofluk, 2020-02-05:
Thx! Ok. I think even in the app it is done through simulatenous execution of actions on both motors instead of a grouping-function. In python I tried to start the motor A and the external motor on port D to start together with Threads but got some exceptions. Thus, I figured you might have tried that before as well.

windoze, 2020-03-20:
I also have this question.
I just tried and found that the iPad App can make all 3 motors work together, but I cannot figure out how can I do the same with this lib.
According to LEGO documents, MoveHub supports ["Write without Response" property](https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#lego-specific-gatt-service), and there is another paragraph talked about [Flow Control](https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#tx-flow-ctrl).
I guess this feature could be useful.

undera, 2020-03-20:
Can you share the screenshot of the app? 
Regarding "write without response" property, please see `use_profile` parameter of some motor methods, it allows controlling exactly that.

windoze, 2020-03-21:
Here I uploaded a video to youtube

https://youtu.be/2AEqCMrSTBI

I'm using official LEGO Boost App on iPad, the app is as simple as the thing you can see from the screen.

undera, 2020-03-21:
Ok, now I understand that it's just parallel call to 3 motors. This can be done with `use_profile` flags.

windoze, 2020-03-21:
I just checked out the document, seems the "profiles" are used for accelerating/decelerating, I don't quite understand how it relates to parallel call.
Also in your face tracking video at https://youtu.be/WUOa3j-6XfI, I can see the external motor and one of the internal motors are rotating at same time, could you please tell me how did you do that?

Thanks a lot and thank you for the great project.

undera, 2020-03-21:
Face tracker code is here: https://github.com/undera/pylgbst/blob/master/examples/tracker/__init__.py

It is done by simple event handling.

windoze, 2020-03-23:
Yes I saw your code, it starts both motors without setting angle or time, so the command gets response immediately, but this doesn't look like what I saw from the official iPad app, which can do something like calling multiple motor.angled() in parallel, and the code doesn't hang even if the motor is blocked.
I'm digging into the protocol docs now but still don't have any clue.

antoinechauveau, 2022-02-09:
Has there been progress on this issue?

I just hit the same problem as the original poster, trying to run something like `motor_A_and_external.angled()` and finding that it's not doable with pylgbst...

A review of the Lego protocol docs did not reveal a clear path forward either:
- `use_profile` indeed refers to acceleration and deceleration profiles, it does not look related to this use case. Is there anything else in the protocol which would allow pylgbst to implement a non-blocking `motor_A.angled()` call, and an associated `motor_A.has_stopped()`?
- https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#virtual-port-setup sounds like exactly what we need to setup a combined `motor_A_and_external`. But the "remark" in the doc is rather cryptic, I don't grasp whether this functionality is supported by the firmware or not...

antoinechauveau, 2022-02-19:
After looking into this a bit more, I created PR #119 which I think does what @hofluk and @windoze were looking for: non-blocking calls with target time, angle or position.

While this allows simultaneous use of multiple motors, there won't be precise synchronization between them. On my setup I see a gap of about 100ms when attempting to start 2 motors right after each other.

closed 2022-02-20 by undera