undera/pylgbst #56: Program stuck when motor turning command cannot be completed

windoze, 2020-03-17:
I.e. If I use Vernie.move(FORWARD), and hold the motor to stop it running, after a short while the program stuck and no new command can be sent to Boost, it shows error message like:
```
Pending request MsgPortOutput({'payload': b'10110cae0100001414647f03', 'needs_reply': True, 'port': 16, 'is_buffered': False, 'do_feedback': True, 'subcommand': 12, 'params': b'ae0100001414647f03'}) while trying to put MsgPortOutput({'payload': b'02110b0100000032647f03', 'needs_reply': True, 'port': 2, 'is_buffered': False, 'do_feedback': True, 'subcommand': 11, 'params': b'0100000032647f03'})
```
I'm using the pygatt backend on a Raspberry with Raspbian installed.
Is there any way to clear pending requests? Or it could be a backend issue?

undera, 2020-03-18:
Hi,
This is not backend-related, it is logical problem of how Boost device functions. Its protocol requires confirmations of command completions, otherwise the device itself will stop processing commands.

The fix for this is making motor to rotate to desired angle by your hand, or restarting the Hub.

closed 2020-03-18 by undera

windoze, 2020-03-18:
Thanks for your explanation, but it's strange that this situation never happened when I used Lego Boost App on iPad, even if I blocked the motor, the program can still keep running.
Is it possible to use `.timed(second=AAA, speed_primary=XXX...)` to completely avoid this issue?

undera, 2020-03-18:
You can change the code to use "timed" method, but that will lose the movement precision.