undera/pylgbst #59: I can't use pylgbst with Control+ Hub

MehdiMechedal, 2020-05-20:
The library is working but not in a good way, i need to change this class to work with Lego Control+ Hub.
For example i need to use the TiltSensor and the gyro sensor but i can't.

```
"""
    Class implementing Lego Boost's MoveHub specifics
    :type led: LEDRGB
    :type tilt_sensor: TiltSensor
    :type button: Button
    :type current: Current
    :type voltage: Voltage
    :type vision_sensor: pylgbst.peripherals.VisionSensor
    :type port_C: Peripheral
    :type port_D: Peripheral
    :type motor_A: EncodedMotor
    :type motor_B: EncodedMotor
    :type motor_AB: EncodedMotor
    :type motor_external: EncodedMotor
    """
```
The output after running the following code is bellow:

```
@asyncio.coroutine
def connect_hub(mac_address, hci):
    connection = GattConnection(hci)
    connection.connect(mac_address)
    hub = MoveHub(connection)
    return hub

@asyncio.coroutine
def connection():
    print("Looking for Hubs")
    hub1 = yield from asyncio.ensure_future(connect_hub(mac_address_hub1, hci))
    hub2 = yield from asyncio.ensure_future(connect_hub(mac_address_hub2, hci))
    print("Hubs connected")
    return hub1, hub2
```
Output:

```
(Peripheral on port 0x0, Peripheral on port 0x1, Peripheral on port 0x2, Peripheral on port 0x3, LEDRGB on port 0x32, None, Current on port 0x3b, Voltage on port 0x3c)
Got only these devices: (Peripheral on port 0x0, Peripheral on port 0x1, Peripheral on port 0x2, Peripheral on port 0x3, LEDRGB on port 0x32, None, Current on port 0x3b, Voltage on port 0x3c)
```

undera, 2020-05-20:
Hi,
You can't use `MoveHub` class for this purpose, it is specific to Boost's device. But you can use `Hub` class instead, it is generic hub device. Though, MoveHub implements some useful features, like waiting for initial peripheral messages.

The proper way is to subclass the `Hub`  into `ControlPlusHub` and implement device's specifics there. The easiest way is to make full copy of `MoveHub` class and modify it to remove peripheral requirements that are not present in Control+ hub.

I'd be happy to accept this new class as a contribution, since I don't have Control+ hardware myself.

closed 2020-06-27 by undera

MehdiMechedal, 2020-07-11:
I resolved this issue, i have created a ControlPlusHub class which make pulgbst compatible with Control + Hub, can you please give me access to push my code into the project, i'm getting this error
 ```
remote: Permission to undera/pylgbst.git denied to MehdiMechedal.
fatal: unable to access 'https://github.com/undera/pylgbst.git/': The requested URL returned error: 403
```

undera, 2020-07-11:
You should do it via Pull Request. Please read here: https://opensource.com/article/19/7/create-pull-request-github