undera/pylgbst #61: Faulty connection with Raspberry Pi 4

M11R05, 2020-05-26:
Hello Undra,

thanks for creating and supporting the pylgbst libary!
I'm working with a new Raspberry Pi 4B and the latest raspian image and latest version of pylgbst. The system is updated.

I've problems to get the Move Hub run with the Raspi. It starts with the gattlib and bluepy libaries.
[bluetooth_log.txt](https://github.com/undera/pylgbst/files/4684934/bluetooth_log.txt)

In your demo.py I've allready added the mac of the hub manually, please see picture:
[image]
I'm a green horn when it comes to python and hope it's in the right place.

My problem is, that the Raspi seems to connect with the Move Hub and gets its status, but then it starts to search again for ble devices and after a while the system freezes and I have to restart it. Please see the log of the demo.py:
[demo_py log.txt](https://github.com/undera/pylgbst/files/4684956/demo_py.log.txt)

I hope you can help me. Thanks!

undera, 2020-05-27:
Hi
The problem in your code is that you are trying to create MoveHub() object twice.

You need to remove either line 253 or line 260.

M11R05, 2020-05-27:
Hello Undera,

thanks a lot, it worked! The motors are working, tilt sensors seem to correspond and distance is detetcted.
There is only one issue with the LED. They are supposed to change color but there seems to be a problem within the code. Please see my log:
[RGB log.txt](https://github.com/undera/pylgbst/files/4689945/RGB.log.txt)
Could you give me a hind how to fix it?

Thanks Matthias!

undera, 2020-05-27:
It looks like you have subscribed to LED notifications and subscriber function misses the required parameter. Can you show the code you run?

M11R05, 2020-05-27:
Here it is:
[demo.txt](https://github.com/undera/pylgbst/files/4689995/demo.txt)

undera, 2020-05-27:
`def colour_callback(**named):` is wrong. do it as `def colour_callback(named):`

M11R05, 2020-05-27:
Now its working as its supposed to be.
Thanks a lot for the quick and good support!

closed 2020-05-27 by M11R05