undera/pylgbst #64: Connection breaks while still instantiating Movehub()

tiwo, 2020-06-08:
(Thank you for this nice library!)

I'm experiencing connection issues. I believe they could stem from something in the interplay of pylgbst and my Lego Boost Hub. I would be grateful for any pointers on how to proceed.

I've had errors using Debian on some Laptop and thought I'd try again with a Raspberry, which I suspect is used 

- The Lego Boost runs on firmware version 2.0.00.0017,
- and runs fine on the app.
- I am trying to connect using a Raspberry Pi 4
- running the current "Raspbian Lite" [now named Raspberry Pi OS (32-bit) Lite 2020-05-27],
- current lgbst,
- the system Python 3 (3.8, I believe),
- using fresh virtual environments,
- running everything as `root` for now.
- I've tried physical distances between the Raspberry 4 and Lego between 0.2 m and 3 m.

I'm trying to run [this](https://github.com/tiwo/pylgbst-bugreport-202006/blob/93bff3371e0cdee96f88b938f6b14649eeaad7fb/list-peripherals.py#L16-L19) code (from a [bash script](https://github.com/tiwo/pylgbst-bugreport-202006/blob/93bff3371e0cdee96f88b938f6b14649eeaad7fb/test-with-fresh-environments#L48-L57) that automates for a reproducible setup):

`conn = get_connection_auto(hub_mac=hub_mac)
hub = MoveHub(connection=conn)`

- the call to get_connection_auto returns, the second line fails.

The actual error comes from the respective backend ([pip freeze](https://github.com/tiwo/pylgbst-bugreport-202006/blob/93bff3371e0cdee96f88b938f6b14649eeaad7fb/results/bluepy.pip-freeze.txt)): [BTLEDisconnectError: Device disconnected](https://github.com/tiwo/pylgbst-bugreport-202006/blob/93bff3371e0cdee96f88b938f6b14649eeaad7fb/results/bluepy.script#L32-L41), pygatt ([pip freeze](https://github.com/tiwo/pylgbst-bugreport-202006/blob/93bff3371e0cdee96f88b938f6b14649eeaad7fb/results/pygatt.pip-freeze.txt)): [No response received / pygatt.exceptions.NotificationTimeout](https://github.com/tiwo/pylgbst-bugreport-202006/blob/93bff3371e0cdee96f88b938f6b14649eeaad7fb/results/pygatt.script#L70-L80)). However it seems that the Move hub disconnects during initialization of the `MoveHub` instance. During the connection attempt, its light was blue for a second, and according to the logs, the initialization proceeds up to a certain point, and then fails as if the Lego Boost device was disappearing.

undera, 2020-06-08:
Hi,
Since MoveHub initialization is quite simple, it is either low battery, or BLE adapter hardware issue. 
But I can only guess, since I don't have my hands on problematic setup.
What can help in troubleshooting is to enable Python logging with DEBUG level and see for possible pointers which exact operation correlates to this issue.

coezbek, 2020-06-15:
I think I am facing the same issue as @tiwo. With debugging enabled and using all library setups which work (bluez, gatttool, gattlib) sporadically, the furthest I can get is to half a second of blue connected light on the move hub and immediate termination when the first command is sent:

`
pi@raspi:~ $ sudo python boostic3.py
INFO:root:Trying get_connection_gattool
DEBUG:comms-pygatt:Trying to connect client to MoveHub with MAC: 00:16:53:b4:dc:8f
DEBUG:pygatt.backends.gatttool.gatttool:gatttool_cmd=gatttool -i hci0 -I
INFO:comms-pygatt:Discovering devices...
INFO:pygatt.backends.gatttool.gatttool:Running...
INFO:pygatt.backends.gatttool.gatttool:Starting BLE scan
....
INFO:pygatt.backends.gatttool.gatttool:Found 14 BLE devices
DEBUG:comms-pygatt:Devices: [...{'name': u'Chris Hub', 'address': u'00:16:53:B4:DC:8F'}]
...
DEBUG:comms:Checking device name: Chris Hub, MAC: 00:16:53:B4:DC:8F
INFO:comms:Found Chris Hub at 00:16:53:B4:DC:8F
INFO:pygatt.backends.gatttool.gatttool:Connecting to 00:16:53:B4:DC:8F with timeout=5.0
DEBUG:pygatt.device:Looking up handle for characteristic 00001624-1212-efde-1623-785feabcd123
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a05-0000-1000-8000-00805f9b34fb, value handle: 0x3
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a00-0000-1000-8000-00805f9b34fb, value handle: 0x7
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a01-0000-1000-8000-00805f9b34fb, value handle: 0x9
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00002a04-0000-1000-8000-00805f9b34fb, value handle: 0xb
DEBUG:pygatt.backends.gatttool.gatttool:Found characteristic 00001624-1212-efde-1623-785feabcd123, value handle: 0xe
DEBUG:pygatt.device:Found <Characteristic uuid=00001624-1212-efde-1623-785feabcd123 handle=14>
DEBUG:pygatt.backends.gatttool.gatttool:Sending cmd=char-write-req 0x0f 0100
`

This call then just hangs there and does nothing.

Any idea?

tiwo, 2020-06-16:
The message corresponding to `0x0f 0100` doesn't seem as suspicious as i first thought: It is  originating from [this line](https://github.com/undera/pylgbst/blob/dff312534fe277590fd1e82538559897c6252fc7/pylgbst/comms/__init__.py#L47) in the comms submodule, and is part of the BLE stack.

I've excluded EMI and low battery levels as a cause, and have tried three different bluetooth adapters. I too have the <1/2 second of blue light before the disconnection. With pygatt, I see a [timeout error](https://github.com/tiwo/pylgbst-bugreport-202006/blob/93bff3371e0cdee96f88b938f6b14649eeaad7fb/results/pygatt.script#L70) 30 seconds after the disconnect.

coezbek, 2020-06-16:
I tried with another library `bricknil` and I get the same issue. I have opened there a issue as well. 

https://github.com/virantha/bricknil/issues/16

undera, 2020-06-20:
If you get the same behavior with bricknil, it means that problem is not with pylgbst or bricknil, it's somewhere in your hardware or underlying BLE stack.

closed 2020-06-20 by undera