undera/pylgbst #65: Connection error on Raspberry Pi 4

coezbek, 2020-06-15:
I am trying to run the minimal example and run into issues with all offered Bluetooth libraries. I have pasted the debug output below. In my mind it would be nice, if the auto connection would print these by default if all connection types fail with an error.

```
INFO:root:Trying get_connection_bluepy
DEBUG:comms-bluepy:Trying to connect client to MoveHub with MAC: None
INFO:comms-bluepy:Discovering devices...
DEBUG:root:Failed: Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 53, in get_connection_auto
    return fn(controller, hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 37, in get_connection_bluepy
    return BluepyConnection(controller).connect(hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/comms/cbluepy.py", line 96, in connect
    scanner.scan(1)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 852, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 790, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 312, in _mgmtCmd
    raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
BTLEManagementError: Failed to execute management command 'le on' (code: 20, error: Permission Denied)

INFO:root:Trying get_connection_bluegiga
WARNING: No module named pexpect
DEBUG:comms-pygatt:Trying to connect client to MoveHub with MAC: None
INFO:pygatt.backends.bgapi.bgapi:Initialized new BGAPI backend
INFO:pygatt.backends.bgapi.bgapi:Auto-detecting serial port for BLED112
DEBUG:pygatt.backends.bgapi.util:Found 1 serial USB devices
DEBUG:pygatt.backends.bgapi.util:Checking serial USB device: /dev/ttyAMA0 - ttyAMA0
DEBUG:root:Failed: Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 53, in get_connection_auto
    return fn(controller, hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 13, in get_connection_bluegiga
    return BlueGigaConnection().connect(hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/comms/cpygatt.py", line 26, in connect
    adapter.start()
  File "/home/pi/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 203, in start
    self._detect_device_port()
  File "/home/pi/.local/lib/python2.7/site-packages/pygatt/backends/bgapi/bgapi.py", line 147, in _detect_device_port
    raise BGAPIError("Unable to auto-detect BLED112 serial port")
BGAPIError: Unable to auto-detect BLED112 serial port

INFO:root:Trying get_connection_gatt
DEBUG:comms-gatt:Starting DeviceManager...
INFO:comms-gatt:Discovering devices...
DEBUG:comms-gatt:Devices: []
INFO:comms-gatt:Discovering devices...
DEBUG:comms-gatt:Devices: [<gatt.gatt_linux.Device instance at 0xb5871fa8>, <gatt.gatt_linux.Device instance at 0xb5874f08>, <gatt.gatt_linux.Device instance at 0xb5874e18>, <gatt.gatt_linux.Device instance at 0xb58e17d8>, <gatt.gatt_linux.Device instance at 0xb58766c0>, <gatt.gatt_linux.Device instance at 0xb586eb20>, <gatt.gatt_linux.Device instance at 0xb58762b0>, <gatt.gatt_linux.Device instance at 0xb5874f58>, <gatt.gatt_linux.Device instance at 0xb5874058>, <gatt.gatt_linux.Device instance at 0xb5874eb8>, <gatt.gatt_linux.Device instance at 0xb5874c60>, <gatt.gatt_linux.Device instance at 0xb58761c0>, <gatt.gatt_linux.Device instance at 0xb6320f58>, <gatt.gatt_linux.Device instance at 0xb5874fa8>, <gatt.gatt_linux.Device instance at 0xb5876120>]
DEBUG:comms:Checking device name: 69-6D-12-F4-5F-0D, MAC: 69:6d:12:f4:5f:0d
DEBUG:root:Failed: Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 53, in get_connection_auto
    return fn(controller, hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 25, in get_connection_gatt
    return GattConnection(controller).connect(hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/comms/cgatt.py", line 103, in connect
    if self._is_device_matched(address, name, hub_mac):
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/comms/__init__.py", line 52, in _is_device_matched
    if (not hub_mac and name == LEGO_MOVE_HUB) or hub_mac.lower() == address.lower():
AttributeError: 'NoneType' object has no attribute 'lower'

INFO:root:Trying get_connection_gattool
DEBUG:comms-pygatt:Trying to connect client to MoveHub with MAC: None
DEBUG:pygatt.backends.gatttool.gatttool:gatttool_cmd=gatttool -i hci0 -I
DEBUG:root:Failed: Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 53, in get_connection_auto
    return fn(controller, hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 19, in get_connection_gattool
    return GattoolConnection(controller).connect(hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/comms/cpygatt.py", line 26, in connect
    adapter.start()
  File "/home/pi/.local/lib/python2.7/site-packages/pygatt/backends/gatttool/gatttool.py", line 283, in start
    self._con = pexpect.spawn(
NameError: global name 'pexpect' is not defined

INFO:root:Trying get_connection_gattlib
INFO:comms-gattlib:Discovering devices using hci0...
DEBUG:root:Failed: Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 53, in get_connection_auto
    return fn(controller, hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 31, in get_connection_gattlib
    return GattLibConnection(controller).connect(hub_mac)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/comms/cgattlib.py", line 69, in connect
    devices = service.discover(1)
RuntimeError: Set scan parameters failed (are you root?)

Traceback (most recent call last):
  File "boostic3.py", line 6, in <module>
    hub = MoveHub()
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/hub.py", line 200, in __init__
    super(MoveHub, self).__init__(connection)
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/hub.py", line 46, in __init__
    connection = get_connection_auto()
  File "/home/pi/.local/lib/python2.7/site-packages/pylgbst/__init__.py", line 60, in get_connection_auto
    raise Exception("Failed to autodetect connection, make sure you have installed prerequisites")
Exception: Failed to autodetect connection, make sure you have installed prerequisites
```

I think there are several causes:

- The `pylgbst` homepage gives examples with `pip install ...`, but really under Linux that should be `sudo pip install...`, because some of the Python bluetooth packages need root during installation.
- To use `gattool` the dependency `pexpect` needs to be installed using `sudo pip install pexpect`. I would suggest to change the main page to say explicitly that users need to follow the instructions for installing the bluetooth python packages, because those might actually require a lot of system packages (via `apt-get` or similar).
- A lot of the connection types need root, so how about excluding them if python is launched without sudo?
- Last, there is a minor logic bug here:

```
  File "/usr/local/lib/python2.7/dist-packages/pylgbst/comms/__init__.py", line 52, in _is_device_matched
    if (not hub_mac and name == LEGO_MOVE_HUB) or hub_mac.lower() == address.lower():
AttributeError: 'NoneType' object has no attribute 'lower'

```
That should be: 

`    if (not hub_mac and name == LEGO_MOVE_HUB) or **(hub_mac and** hub_mac.lower() == address.lower()):`

undera, 2020-06-20:
Hi,
Feel free to send a PR with your proposed changes

tonka3000, 2020-06-27:
On the [master branch](https://github.com/undera/pylgbst/blob/ba7594a0811e841a64b54f9b960b867a9d832236/pylgbst/comms/__init__.py#L53) this is already fixed. This patch is only required for the actual highest version of `pylgbst` 1.1.2 on pypi.

@undera any plans to push a newer version to pypi?

undera, 2020-06-27:
I have just released version 1.2.0

closed 2020-06-27 by undera