undera/pylgbst #66: Find MoveHub by name

coezbek, 2020-06-15:
Currently it is not documented on the main page that the name of the MoveHub needs to be "LEGO Move Hub" for it to be found by name.

Quote from "pylgbst/comms/__init__.py"

```
 def _is_device_matched(self, address, name, hub_mac):
        log.debug("Checking device name: %s, MAC: %s", name, address)
        if address != "00:00:00:00:00:00":
            if (not hub_mac and name == LEGO_MOVE_HUB) or (hub_mac and hub_mac.lower() == address.lower()):
                log.info("Found %s at %s", name, address)
                return True
        return False
```

Suggestion:
- Allow to provide name also by regex (.*Move Hub.*)
- Allow to pass name (and address?) directly via the Hub constructor so that the advantages of an auto connection with searching by address can be combined.

undera, 2020-06-27:
This is addressed in #68

undera, 2020-06-27:
The design of classes implies that connections may be configured in various ways, so just pass connection function to a hub constructor.

closed 2020-06-27 by undera