undera/pylgbst #69: Raspberry Pi: demo.py sometimes work, sometimes not

tonka3000, 2020-06-28:
Hey @undera ,

I have troubles to get the `demo.py` example from your repo to work always. It worked multiple times (3-4 times), but most of the time the hub switches off `WARNING:hub:Hub switches off`, before an action can be started and the program stuck on that state. The bot turns off after 10 seconds of not interaction which is the default as far as I can tell. I've tried it on my mac, but no bluetooth backend seems to support MacOS so not test possible. For me it looks like the program on the pi is so slow (maybe wait for an event) and reach timeout before.

Hardware: raspberry pi 3 B, Lego Boost Robot
python: 3.7.3
virtual env: yes, with python3 -m venv
Bluetooth backend: bluepy 1.3.0
Firmware-Version: `2.0.00.0017`

Here is the application stdout from `demo.py`. It extremely fast found the hub but it stuck on the line `INFO:hub:Attached peripheral: LEDRGB on port 0x32` for some time, so it seems that the program is waiting for something here.

```sh
INFO:root:Trying get_connection_bluepy
INFO:comms-bluepy:Discovering devices...
INFO:comms-bluepy:Discovering devices...
INFO:comms:Found LEGO Move Hub at 00:16:53:ba:05:69
INFO:hub:Attached peripheral: EncodedMotor on port 0x0
INFO:hub:Attached peripheral: EncodedMotor on port 0x1
INFO:hub:Attached peripheral: VisionSensor on port 0x2
INFO:hub:Attached peripheral: EncodedMotor on port 0x3
INFO:hub:Attached peripheral: EncodedMotor on port 0x10
INFO:hub:Attached peripheral: LEDRGB on port 0x32 //<= stuck for multiple seconds here
INFO:hub:Attached peripheral: TiltSensor on port 0x3a
INFO:hub:Attached peripheral: Current on port 0x3b
INFO:hub:Attached peripheral: Voltage on port 0x3c
WARNING:hub:Have not dedicated class for peripheral type 0x42 on port 0x46
INFO:hub:Attached peripheral: Peripheral on port 0x46
WARNING:hub:Hub switches off
```

Maybe you have an idea how I can debug that problem.

Greetings<br>
Tonka

undera, 2020-06-30:
The proper way to troubleshoot this is to enable DEBUG level logging in demo.py (instead of INFO). Then run it again and share the log.

tonka3000, 2020-07-01:
Hey,

here is the log.

```
INFO:root:Trying get_connection_bluepy
DEBUG:comms-bluepy:Trying to connect client to MoveHub with MAC: None
INFO:comms-bluepy:Discovering devices...
DEBUG:comms:Checking device: LEGO Move Hub, MAC: 00:16:53:ba:01:50
INFO:comms:Found LEGO Move Hub at 00:16:53:ba:01:50
DEBUG:comms-bluepy:Writing to handle 15: b'0100'
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None, None, None, None, None)
DEBUG:comms-bluepy:Incoming notification
DEBUG:hub:Notification on 14: b'0f0004000127000000001000000010'
DEBUG:hub:Decoded message: MsgHubAttachedIO({'payload': b'27000000001000000010', 'port': 0, 'event': 1})
DEBUG:hub:Handling msg with <bound method MoveHub._handle_device_change of <pylgbst.hub.MoveHub object at 0x74886f30>>: MsgHubAttachedIO({'payload': b'27000000001000000010', 'port': 0, 'event': 1})
INFO:hub:Attached peripheral: EncodedMotor on port 0x0
DEBUG:comms-bluepy:Incoming notification
DEBUG:hub:Notification on 14: b'0f0004010127000000001000000010'
DEBUG:hub:Decoded message: MsgHubAttachedIO({'payload': b'27000000001000000010', 'port': 1, 'event': 1})
DEBUG:hub:Handling msg with <bound method MoveHub._handle_device_change of <pylgbst.hub.MoveHub object at 0x74886f30>>: MsgHubAttachedIO({'payload': b'27000000001000000010', 'port': 1, 'event': 1})
INFO:hub:Attached peripheral: EncodedMotor on port 0x1
DEBUG:hub:Waiting for builtin devices to appear: (EncodedMotor on port 0x0, EncodedMotor on port 0x1, None, None, None, None, None)
DEBUG:comms-bluepy:Incoming notification
DEBUG:hub:Notification on 14: b'0f0004020125000000001000000010'
DEBUG:hub:Decoded message: MsgHubAttachedIO({'payload': b'25000000001000000010', 'port': 2, 'event': 1})
DEBUG:hub:Handling msg with <bound method MoveHub._handle_device_change of <pylgbst.hub.MoveHub object at 0x74886f30>>: MsgHubAttachedIO({'payload': b'25000000001000000010', 'port': 2, 'event': 1})
INFO:hub:Attached peripheral: VisionSensor on port 0x2
DEBUG:comms-bluepy:Incoming notification
DEBUG:hub:Notification on 14: b'0f0004030126000000001000000010'
DEBUG:hub:Decoded message: MsgHubAttachedIO({'payload': b'26000000001000000010', 'port': 3, 'event': 1})
DEBUG:hub:Handling msg with <bound method MoveHub._handle_device_change of <pylgbst.hub.MoveHub object at 0x74886f30>>: MsgHubAttachedIO({'payload': b'26000000001000000010', 'port': 3, 'event': 1})
INFO:hub:Attached peripheral: EncodedMotor on port 0x3
DEBUG:comms-bluepy:Incoming notification
DEBUG:hub:Notification on 14: b'090004100227000001'
DEBUG:hub:Decoded message: MsgHubAttachedIO({'payload': b'27000001', 'port': 16, 'event': 2})
DEBUG:hub:Handling msg with <bound method MoveHub._handle_device_change of <pylgbst.hub.MoveHub object at 0x74886f30>>: MsgHubAttachedIO({'payload': b'27000001', 'port': 16, 'event': 2})
INFO:hub:Attached peripheral: EncodedMotor on port 0x10
DEBUG:comms-bluepy:Incoming notification
DEBUG:hub:Notification on 14: b'0f0004320117000000000106000020'
DEBUG:hub:Decoded message: MsgHubAttachedIO({'payload': b'17000000000106000020', 'port': 50, 'event': 1})
DEBUG:hub:Handling msg with <bound method MoveHub._handle_device_change of <pylgbst.hub.MoveHub object at 0x74886f30>>: MsgHubAttachedIO({'payload': b'17000000000106000020', 'port': 50, 'event': 1})
INFO:hub:Attached peripheral: LEDRGB on port 0x32
DEBUG:hub:Waiting for builtin devices to appear: (EncodedMotor on port 0x0, EncodedMotor on port 0x1, EncodedMotor on port 0x10 (ports 0x0 and 0x1 combined), LEDRGB on port 0x32, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (EncodedMotor on port 0x0, EncodedMotor on port 0x1, EncodedMotor on port 0x10 (ports 0x0 and 0x1 combined), LEDRGB on port 0x32, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (EncodedMotor on port 0x0, EncodedMotor on port 0x1, EncodedMotor on port 0x10 (ports 0x0 and 0x1 combined), LEDRGB on port 0x32, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (EncodedMotor on port 0x0, EncodedMotor on port 0x1, EncodedMotor on port 0x10 (ports 0x0 and 0x1 combined), LEDRGB on port 0x32, None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (EncodedMotor on port 0x0, EncodedMotor on port 0x1, EncodedMotor on port 0x10 (ports 0x0 and 0x1 combined), LEDRGB on port 0x32, None, None, None)
[... cut ...]
DEBUG:hub:All devices are present: (EncodedMotor on port 0x0, EncodedMotor on port 0x1, EncodedMotor on port 0x10 (ports 0x0 and 0x1 combined), LEDRGB on port 0x32, TiltSensor on port 0x3a, Current on port 0x3b, Voltage on port 0x3c)
DEBUG:hub:Send message: MsgHubProperties({'payload': b'', 'needs_reply': False, 'property': 1, 'operation': 5, 'parameters': b''})
DEBUG:hub:Waiting for sync reply to MsgHubProperties({'payload': b'0105', 'needs_reply': True, 'property': 1, 'operation': 5, 'parameters': b''})...
DEBUG:comms-bluepy:Writing to handle 14: b'0500010105'
```

undera, 2020-07-01:
So it looks like there is indeed something that makes pylgbst to not receive any incoming messages from BLE after LEDRGB attach message. The library is helpless here, since it simply does not get any communication from underlying BLE stack.
This can be either hardware problem, or underlying library problem, bluepy in your case.  

I would give a try to another connection library, for example pygatt. Again, I don't see anything that would cause such a pause on the library side (I can be wrong, though, since I did not write that part of library).

tonka3000, 2020-07-01:
I tried every other backend, but no of them work. Always with the following error (here for `gatt`), seems like a different kind f problem here.

```
INFO:root:Trying get_connection_bluepy
INFO:root:Trying get_connection_bluegiga
INFO:root:Trying get_connection_gatt
Module 'dbus' not found
Please run: sudo apt-get install python3-dbus
See also: https://github.com/getsenic/gatt-python#installing-gatt-sdk-for-python
INFO:root:Trying get_connection_bleak
INFO:root:Trying get_connection_gattool
INFO:root:Trying get_connection_gattlib
Traceback (most recent call last):
  File "demo.py", line 259, in <module>
    hub = MoveHub(**parameters)
  File "/home/pi/lego/env/lib/python3.7/site-packages/pylgbst/hub.py", line 201, in __init__
    connection = get_connection_auto(hub_name="LEGO Move Hub")
  File "/home/pi/lego/env/lib/python3.7/site-packages/pylgbst/__init__.py", line 75, in get_connection_auto
    raise Exception("Failed to autodetect connection, make sure you have installed prerequisites")
Exception: Failed to autodetect connection, make sure you have installed prerequisites
Exception ignored in: <function Hub.__del__ at 0x76509468>
Traceback (most recent call last):
  File "/home/pi/lego/env/lib/python3.7/site-packages/pylgbst/hub.py", line 52, in __del__
    if self.connection and self.connection.is_alive():
AttributeError: 'MoveHub' object has no attribute 'connection'
```

I also used `gattctl --discover` and get following message

```
Module 'dbus' not found
Please run: sudo apt-get install python3-dbus
See also: https://github.com/getsenic/gatt-python#installing-gatt-sdk-for-python
```

So I run `sudo apt-get install python3-dbus` but it's already installed.

I checked it also on another raspberry pi, same issue there.

Greetings

Tonka

undera, 2020-07-01:
`Please run: sudo apt-get install python3-dbus` - have you tried running that?

undera, 2020-07-01:
Do you have `bluez` installed? It is installed with `sudo apt-get install bluez`

tonka3000, 2020-07-01:
`bluez` is also installed

undera, 2020-07-01:
`/home/pi/lego/env/lib` - are you using some sort of virtualenv? If yes, then you need to install python dbus lib into that virtual env (`pip install dbus-python` or so, might also need some libs, see https://stackoverflow.com/questions/13365697/install-python-dbus-in-virtualenv), or use system-site-packages option of it.

tonka3000, 2020-07-02:
installing dbus-python via `pip install dbus-python` does not work for gatt, but it worked for bleak. With bleak it works most of the time. The strange thing is that I can not really find out under which conditions it does not work (I also check the bluetooth distance, but it did not change that much).