undera/pylgbst #71: Movehub disconnects with Raspberry pi v4.19 and Bluetooth v5.50 after connecting ?

aquenneville, 2020-07-01:
Hi, Fantastic Project! Good work! 
I am using: 
- Raspberry pi v4.19
- Bluetooth v5.50 (PyBluez)
- Python v3.7.3
- installed the python dependencies

When I add the (Lego boost) device and select the MoveHub from the Bluetooth list: I click on the Pair button, 
and a dialog appears: Pairing request sent to the device - waiting for a response. Then I click on the Lego boost green button to turn it on. Then a new dialog appears: Paired successful but this device has no services which can be used with Raspberry pi. 

I have tested my bluetooth with another device (audio speaker) and I have been able to connect to it. I have tried different commands in the shell. I haven't found anything in the /var/log/syslog and dmesg.  
In the bluetoothctl, I have this output:  
[CHG] Device XX:33 Connected: yes
[CHG] Device XX:33 ServicesResolved: yes
[CHG] Device XX:33 Paired: yes
[CHG] Device XX:33 ServicesResolved: no
[CHG] Device XX:33 Paired: no
[CHG] Device XX:33 Connected: no

I have tried to make the bluetooth discoverable, tried different bluetooth commands. 
It seems like the lego boost disconnects as soon as the light on the device turns off. 
The device probably goes into Low energy mode and Raspberry pi is unable to keep the connection.
Not sure exactly what I am doing wrong? Any help would be greatly appreciated. 
I really like the project! Thank you for reading

undera, 2020-07-02:
I don't have any experience with Raspberry Pi, maybe others can help you in this.
From my Linux desktop practice, there is no need in any pairing for Lego Boost. You just start Python program and immediately start the device by pressing green button. And it usually works.

MehdiMechedal, 2020-07-02:
I tried it yesterday with almost same versions, and it was working, can you show your python code please ?

aquenneville, 2020-07-03:
Hi, I am just running demo.py in the project... I thought you needed to pair it. 
I'll try again just by starting the code and pushing the green button. I'll let you know.
Thanks