undera/pylgbst #74: It doesnt work on Raspberry pi

FuchsCrafter, 2020-09-03:
It doesent work on my Rpi 4 model b (Thonny Python IDE).
There is just an error message (in the Demo file):

Traceback (most recent call last):
  File "/home/pi/Downloads/boost/pylgbst-master/examples/demo.py", line 5, in <module>
    from pylgbst import *
ModuleNotFoundError: No module named 'pylgbst'

[image]
 Why it is so?

undera, 2020-09-03:
`No module named...` is pretty basic Python exception, it means Python interpreted is unable to find the library. There can be various reasons for that: you did not install module, module search paths are not configured properly, etc.

FuchsCrafter, 2020-09-04:
> `No module named...` is pretty basic Python exception, it means Python interpreted is unable to find the library. There can be various reasons for that: you did not install module, module search paths are not configured properly, etc.

And what can I so?

reinhard-brandstaedter, 2020-09-08:
Try:
```pip3 install pyglbst``` to install the pyglbst module. Depending how you run you might do it as root

FuchsCrafter, 2020-09-09:
Thanks. I will try it.

FuchsCrafter, 2020-12-24:
It works now. Issue closed.

PS Now I thing im not a big brain.

closed 2020-12-24 by FuchsCrafter