undera/pylgbst #76: Is bleak supported or not?

finik, 2020-09-13:
I am still confused, is bleak supported or not?
I am trying to run it on a mac:
```
from pylgbst.hub import MoveHub
from pylgbst.comms.cbleak import BleakConnection

conn = BleakConnection()
conn.connect()  # you can pass Hub mac address as parameter here, like 'AA:BB:CC:DD:EE:FF'

hub = MoveHub(conn)
```

and I get this:
```
solver.py:5: RuntimeWarning: coroutine 'BleakConnection.connect' was never awaited
  conn.connect()  # you can pass Hub mac address as parameter here, like 'AA:BB:CC:DD:EE:FF'
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
/Users/max/Library/Python/3.7/lib/python/site-packages/pylgbst/hub.py:48: RuntimeWarning: coroutine 'BleakConnection.set_notify_handler' was never awaited
  self.connection.set_notify_handler(self._notify)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
/Users/max/Library/Python/3.7/lib/python/site-packages/pylgbst/comms/__init__.py:46: RuntimeWarning: coroutine 'BleakConnection.write' was never awaited
  self.write(ENABLE_NOTIFICATIONS_HANDLE, ENABLE_NOTIFICATIONS_VALUE)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Got only these devices: (None, None, None, None, None, None, None)
/Users/max/Library/Python/3.7/lib/python/site-packages/pylgbst/hub.py:71: RuntimeWarning: coroutine 'BleakConnection.write' was never awaited
  self.connection.write(self.HUB_HARDWARE_HANDLE, msgbytes)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
```

Mac, Bleak 0.7.1, pylgbst 1.2.0
Thanks

finik, 2020-09-13:
To answer myself, the documentation was wrong, here is the fix:
https://github.com/undera/pylgbst/pull/77

something works now, still experimenting

finik, 2020-09-13:
Update: In order to make it work on a mac, had to do another fix to the actual code:
https://github.com/undera/pylgbst/pull/78

with this, it very consistently connects to the device every even and fails every odd time when experimenting. Seems like there is some state on one of the layers (bleak/BT core?)

finik, 2020-09-13:
fresh demo: https://www.youtube.com/watch?v=F9OxJLWDVfc

undera, 2020-09-14:
The video is WOW!

finik, 2020-09-14:
btw, this whole thing started suddenly working, connection works every time, really fast too, connects almost instantly, I am still not sure what is happening

undera, 2020-09-20:
I guess this works now

closed 2020-09-20 by undera

matti76, 2020-11-11:
Hi all, i also ran into the problems described in [#82 ](https://github.com/undera/pylgbst/issues/82).
Running demo.py shows the following:

`162	INFO	root	Trying get_connection_bluepy
170	INFO	root	Trying get_connection_bluegiga
175	INFO	root	Trying get_connection_gatt
180	INFO	root	Trying get_connection_bleak
1480	INFO	comms-bleak	Discovering devices... Press green button on Hub
11489	INFO	comms	Found LEGO Move Hub at 5E96D149-85EA-42B6-A833-1D0C0364C036
11490	INFO	comms-bleak	Device matched: 5E96D149-85EA-42B6-A833-1D0C0364C036: LEGO Move Hub
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.9/3.9.0_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 950, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@3.9/3.9.0_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 888, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 50, in <lambda>
    self._connection_thread = threading.Thread(target=lambda: asyncio.run(self._bleak_thread()))
  File "/usr/local/Cellar/python@3.9/3.9.0_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/Cellar/python@3.9/3.9.0_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 60, in _bleak_thread
    await bleak.connect(self.hub_mac, self.hub_name)
  File "/usr/local/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 156, in connect
    self._client = BleakClient(self._device.address, self.loop)
TypeError: __init__() takes 2 positional arguments but 3 were given
11733	WARNING	hub	Got only these devices: (None, None, None, None, None, None, None)
Traceback (most recent call last):
  File "/Users/matti/Downloads/pylgbst-master/examples/demo.py", line 259, in <module>
    hub = MoveHub(**parameters)
  File "/usr/local/lib/python3.9/site-packages/pylgbst/hub.py", line 221, in __init__
    self._report_status()
  File "/usr/local/lib/python3.9/site-packages/pylgbst/hub.py", line 238, in _report_status
    name = self.send(MsgHubProperties(MsgHubProperties.ADVERTISE_NAME, MsgHubProperties.UPD_REQUEST))
  File "/usr/local/lib/python3.9/site-packages/pylgbst/hub.py", line 71, in send
    self.connection.write(self.HUB_HARDWARE_HANDLE, msgbytes)
  File "/usr/local/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 96, in write
    raise ConnectionError('Something went wrong, communication threads not functioning.')
ConnectionError: Something went wrong, communication threads not functioning.`

Any ideas how to solve this ?

bibo750four, 2020-11-12:
Hello,
When I run demo.py I do not even get anything like what I see in the log from matti76 in the line below:

```11489 INFO comms Found LEGO Move Hub at 5E96D149-85EA-42B6-A833-1D0C0364C036
11490 INFO comms-bleak Device matched: 5E96D149-85EA-42B6-A833-1D0C0364C036: LEGO Move Hub
```

However I get this warning:
```
520	WARNING	bleak.backends.corebluetooth.CentralManagerDelegate	CentralManagerDelegate is not compliant
```

I also see that the LEGO Hub is discovered because I recognise the UUID, but it is discovered as unknown device:
```
12995	DEBUG	bleak.backends.corebluetooth.CentralManagerDelegate	Discovered device 70774252-9616-48AA-BEF5-F1EC6EB3AED6: Unknown @ RSSI: -53 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
12998	DEBUG	bleak.backends.corebluetooth.CentralManagerDelegate	centralManager_didDiscoverPeripheral_advertisementData_RSSI_
12998	DEBUG	bleak.backends.corebluetooth.CentralManagerDelegate	Discovered device 70774252-9616-48AA-BEF5-F1EC6EB3AED6: Unknown @ RSSI: -54 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
```

If the information can be helpful, I can get a bluetooth connection to the hub using the following:
```
from pylgbst.hub import MoveHub
from pylgbst import get_connection_bleak

conn = get_connection_bleak(hub_mac='70774252-9616-48AA-BEF5-F1EC6EB3AED6')
hub = MoveHub(conn)
```

Can anyone give a hint on where to look for the issue?

Marco

akava, 2020-11-14:
Hi All,

@finik, thanks your fix works. I had to downgrade Bleak to 0.7.1 and setup connection via MAC address (same as @bibo750four ) showed above. 
However it still doesn't work.

@matti76 please downgrade Bleak to 0.7.1 and do a hack with MAC above.

akava, 2020-11-14:
@undera, what OS and what backend are you using? Guess it's the most stable one.
Thanks for the cool library. Your video is amaizing

undera, 2020-11-15:
I'm using PC with Ubuntu 20.04 Linux, with `pygatt` backend. But other backends work fine for me, too. I don't have Mac available, at all.

akava, 2020-11-15:
Thanks @undera, I'll try pygatt on Windows. 
If you have time could you please release latest version of pylgbst? The release available in pip is 1.2.0 released on June, 27.

FYI Regarding Bleak. They updated API starting v0.8 now the BleakClient doesn't control the event loop but rather executes in the external one: https://github.com/hbldh/bleak
I'm not a Python expert so cannot update cbleak.py without significant effort to match the new paradigm.

akava, 2020-11-15:
Ok, I made it work on Mac. Connecting via direct MAC-addr connection. For some reason my Lego Boost got identified either as "Unknown" or "LEGO System  A/S".
Works pretty stable. In order to play with Vernie I had to change Vernie class (from examples) to pass connection to it.
To enable latest version of Bleak to work with pygatt I just removed "self.loop" parameter: BleakClient(self._device.address).

Thanks again! Going to play with it.

huskeypm, 2020-11-22:
@akava How exactly did you make it work with macos? 
I have macos 10.14.6, downgraded to bleak-0.7.1 and am using pylgbst 1.2 
I was able to figure out the mac for my device using the repo's demo.py file:
python demo.py
.....
	Discovered device B6C75D25-356C-46FF-B448-9997B9412B2F: LEGO Move Hub
......

I used this address with the code  to 'successfully' bind to the device, but an exception was thrown:
```
from pylgbst.hub import MoveHub
from pylgbst import get_connection_bleak

addr= "B6C75D25-356C-46FF-B448-9997B9412B2F"
conn = get_connection_bleak(hub_mac=addr)
hub = MoveHub(conn)
```

Traceback (most recent call last):
  File "/Users/peterkekenes-huskey/opt/anaconda3/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
.....
TypeError: encoding without a string argument

The blue light turns on when I run this command, but the code hangs before the hub object is assigned. 

I've posted the full output below. Thanks!

`>>> from pylgbst.hub import MoveHub
>>> from pylgbst import get_connection_bleak
>>> 
>>> addr= "B6C75D25-356C-46FF-B448-9997B9412B2F"
>>> conn = get_connection_bleak(hub_mac=addr)
>>> hub = MoveHub(conn)
2020-11-22 15:53:33,199 bleak.backends.corebluetooth.CentralManagerDelegate WARNING: CentralManagerDelegate is not compliant
2020-11-22 15:53:33,208 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManagerDidUpdateState_
2020-11-22 15:53:33,209 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Bluetooth powered on
2020-11-22 15:53:33,237 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:33,238 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:33,238 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device EBEA2C34-04DC-4B23-AA75-3594DE75E341: Unknown @ RSSI: -48 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataChannel'])>)
2020-11-22 15:53:33,239 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device EBEA2C34-04DC-4B23-AA75-3594DE75E341: Unknown @ RSSI: -48 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-11-22 15:53:33,244 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:33,245 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:33,245 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 93D9306F-5F9C-4441-B927-4B343A18D355: Unknown @ RSSI: -83 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
2020-11-22 15:53:33,245 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 93D9306F-5F9C-4441-B927-4B343A18D355: Unknown @ RSSI: -83 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>)
2020-11-22 15:53:33,336 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:33,337 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:33,337 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device CDE219CB-F53A-4E4E-98EA-8E89A23C1584: Unknown @ RSSI: -94 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-11-22 15:53:33,338 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device CDE219CB-F53A-4E4E-98EA-8E89A23C1584: Unknown @ RSSI: -93 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
2020-11-22 15:53:34,807 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:34,808 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:34,809 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device B6C75D25-356C-46FF-B448-9997B9412B2F: LEGO Move Hub @ RSSI: -47 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
2020-11-22 15:53:34,809 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device B6C75D25-356C-46FF-B448-9997B9412B2F: LEGO Move Hub @ RSSI: -47 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
2020-11-22 15:53:37,827 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2020-11-22 15:53:37,828 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device B6C75D25-356C-46FF-B448-9997B9412B2F: LEGO Move Hub @ RSSI: 79 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
2020-11-22 15:53:39,015 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
[... cut ...]
    name = self.send(MsgHubProperties(MsgHubProperties.ADVERTISE_NAME, MsgHubProperties.UPD_REQUEST))
  File "/Users/peterkekenes-huskey/opt/anaconda3/lib/python3.7/site-packages/pylgbst/hub.py", line 72, in send
    resp = self._sync_replies.get()
  File "/Users/peterkekenes-huskey/opt/anaconda3/lib/python3.7/queue.py", line 170, in get
    self.not_empty.wait()
  File "/Users/peterkekenes-huskey/opt/anaconda3/lib/python3.7/threading.py", line 296, in wait
    waiter.acquire()
KeyboardInterrupt
`

bibo750four, 2020-12-09:
I still cannot make it work. Can someone who managed to make it work please describe what needs to be done? I would like to run the Vernie examples. I am using Bleak 0.7.1 and pylgbst 1.2.0.