undera/pylgbst #82: demo.py not working with bleak on mac os

bibo750four, 2020-11-02:
Hello,

The demo.py fails to connect to the hub, please see the result below. 

```
!debugfile('/pylgbst-master/examples/demo.py')
28215   INFO    root    Trying get_connection_bluepy
28218   INFO    root    Trying get_connection_bluegiga
28236   INFO    pygatt.backends.bgapi.bgapi     Initialized new BGAPI backend
28236   INFO    pygatt.backends.bgapi.bgapi     Auto-detecting serial port for BLED112
28241   INFO    root    Trying get_connection_gatt
28243   INFO    root    Trying get_connection_bleak
28552   INFO    comms-bleak     Discovering devices... Press green button on Hub
28553   WARNING bleak.backends.corebluetooth.CentralManagerDelegate     CentralManagerDelegate is not compliant
28558   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManagerDidUpdateState_
28641   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Bluetooth powered on
28679   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
28681   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
28682   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 70774252-9616-48AA-BEF5-F1EC6EB3AED6: Unknown @ RSSI: -61 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
28685   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 70774252-9616-48AA-BEF5-F1EC6EB3AED6: Unknown @ RSSI: -61 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
29263   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
29266   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 64847CEF-19AE-436C-8B66-39838CA0B3E5: Unknown @ RSSI: -89 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>)
29266   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
29269   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
29269   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device FD37B2DB-896C-40AB-BD58-83A7606CA37A: Unknown @ RSSI: -75 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>)
29271   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device FD37B2DB-896C-40AB-BD58-83A7606CA37A: Unknown @ RSSI: -75 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
29280   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
29282   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 2908C675-B0E1-4769-8056-4BD7B7BD1F41: Unknown @ RSSI: -52 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
29282   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
29284   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 2908C675-B0E1-4769-8056-4BD7B7BD1F41: Unknown @ RSSI: -52 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
29583   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
29585   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 0C3F84D2-9BCA-4B17-9B58-5FCD1EC36F76: Unknown @ RSSI: -63 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>)
29586   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
29588   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 0C3F84D2-9BCA-4B17-9B58-5FCD1EC36F76: Unknown @ RSSI: -64 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataLocalName', 'kCBAdvDataChannel'])>)
30027   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
30030   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 70774252-9616-48AA-BEF5-F1EC6EB3AED6: Unknown @ RSSI: -61 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
30030   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
30032   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 70774252-9616-48AA-BEF5-F1EC6EB3AED6: Unknown @ RSSI: -61 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
30757   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
30760   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device F98B36B6-9691-43C1-B40F-E3EAF47C9ADE: Unknown @ RSSI: -54 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>)
31363   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
31365   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
31366   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 36224CEE-C55A-43FD-BA4C-53888AC16AC6: Unknown @ RSSI: -60 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>)
31368   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 36224CEE-C55A-43FD-BA4C-53888AC16AC6: Unknown @ RSSI: -60 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
31689   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
31691   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 64847CEF-19AE-436C-8B66-39838CA0B3E5: Unknown @ RSSI: -80 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>)
32259   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
32261   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
32262   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device FD37B2DB-896C-40AB-BD58-83A7606CA37A: Unknown @ RSSI: -77 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>)
32264   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device FD37B2DB-896C-40AB-BD58-83A7606CA37A: Unknown @ RSSI: -76 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
33184   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
33185   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 0C3F84D2-9BCA-4B17-9B58-5FCD1EC36F76: Unknown @ RSSI: -68 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>)
33187   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     centralManager_didDiscoverPeripheral_advertisementData_RSSI_
33188   DEBUG   bleak.backends.corebluetooth.CentralManagerDelegate     Discovered device 0C3F84D2-9BCA-4B17-9B58-5FCD1EC36F76: Unknown @ RSSI: -68 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataLocalName', 'kCBAdvDataChannel'])>)
[... cut ...]
    self.assertEqual(bytes(data), last_response[1], 'Verifying response data')
AssertionError: b'\r\x0e\n\r\x0b\x0e\x0e\x0f' != [13, 14, 10, 13, 11, 14, 14, 15] : Verifying response data

----------------------------------------------------------------------
Ran 2 tests in 1.009s

FAILED (failures=1)
```

I am very new to python so it may for sure be that I am doing something wrong. Please let me know if I can do some further test or provide further information.

I am using pylgbst 1.2.0 and bleak 0.7.1

Thank you
Marco

undera, 2020-11-03:
Have you read through #76 ? I suggest doing so.

bibo750four, 2020-11-06:
Thank you for pointing me to the fix. Now the Bleak driver test is passed with OK. However I still do not get a connection when trying the demo.py.
Should it be possible to run the demo as it is or do I have to do something different to allow discovering the hub?
I also am not sure if I updated correctly the library using the last commit. Is it ok to just copy the content of the pylgbst folder to the respective location in the python library structure? I tried to update using a .zip and run pip but it did not work.
Sorry for my beginner-level knowledge :)

undera, 2020-11-07:
It is hard for me to tell what happens in your machine, there are many ways things can go, depending on your exact actions. I encourage you posting comment into issue #76, asking those people to help.

huskeypm, 2020-11-22:
I'm having a similar issue as @bibo750four and posted my comment to #76.  I apologize for being naive, but does the issue need to be reopened or will someone read the comment made to a closed post? 

Thanks,
Pete

undera, 2020-11-23:
@huskeypm Yes, people see comments in closed issues

huskeypm, 2020-11-23:
Thanks. I haven't made much use of github's issue tracker.

undera, 2021-05-19:
Next release would fix it

closed 2021-05-19 by undera