undera/pylgbst #88: Can't get vernie battery level

JeanPeyreMesMots, 2021-03-03:
Hi! 

I'm currently working on a project with my classmates on the Lego Bosst. We're using the current library to make the robot moving, turning, etc....

I have to create a script to get the battery state of the robot while connected via Bluetooth from an Raspberry Pi to it, but I'm stuck for weeks while trying for it :(
I would like just to get the battery percentage, but when I'm launching a script, it always says "Low voltage, check power source (maybe replace battery)", assuming that I've made the robot charged for like 1-2 hours. 

I started searching in the "hub.py" file to see what can be done, and I found this : 
[image]

I don't understand what i don't have the current percentage value returned, instead I have the error I've said before :(
Can someone help me to solve this issue ?

I specify that I have installed the library by using the "pip install -U pylgbst", with all the bluetooth prerequisites, on my Raspberry. 

Thanks you for responding,
Have a great day :)

undera, 2021-03-03:
Hi,
The message of "Low voltage" is printed when response from MoveHub has "not successful" status. There is a chance that overall logic to check `is_ok()` is wrong there. The more important is the message couple of lines above, the one that actually prints voltage level percentage. Does it print the voltage percent for you? What's the number?

JeanPeyreMesMots, 2021-03-04:
No, I don't have the voltage percent :( All I have is the message who tell me "Low voltage, check power source...(etc)"
I got a successful connection to the robot, I can move, turn, go bac... But I don't have the percent :(

undera, 2021-03-04:
In the screenshot you shown above, there is a piece of code to report voltage %

JeanPeyreMesMots, 2021-03-04:
Yes, but i don't have it. All I have in output is the error message I mentioned above

undera, 2021-03-04:
Can you show me full code that you run?

JeanPeyreMesMots, 2021-03-04:
Here it is :

```
from time import sleep

from pylgbst import *
from pylgbst.hub import MoveHub, Voltage, Peripheral
from pylgbst.comms.cgatt import GattConnection

conn = GattConnection("hci0")
if(conn.connect('XX:XX:XX:XX:XX:XX')):
    print("Le robot est connecté")	  # Adresse MAC du Robot Lego Boost
hub = MoveHub(conn)

def callback(value):
    print("Voltage: %s" % value)

# Valeur de départ de la batterie
premiere_valeur = hub.voltage.get_sensor_data(Voltage.VOLTAGE_S)
print(hub.voltage.get_sensor_data(Voltage.VOLTAGE_S))
# Valeur de fin de la batterie
deuxieme_valeur = hub.voltage.get_sensor_data(Voltage.VOLTAGE_L)
print("\r\n")

# Pour la valeur de départ
# Conversion de tuple en string
i = sum(premiere_valeur)
print("Valeur flottante de la première valeur = ", i)
# Conversion de string en entier
valeur_depart = int(i)
print("Valeur entière de la première valeur = ", valeur_depart,"%")

# Même chose pour la valeur finale
y = sum(deuxieme_valeur)
print("Valeur flottante de la deuxième valeur = ", y)
valeur_finale = int(y)
print("Valeur entière de la deuxième valeur = ", valeur_finale,"%")
```

undera, 2021-03-04:
Your code looks ok. Does it print out "valeur" lines? Can you post here example of what it prints?

undera, 2021-03-04:
Also, there is another way to get the voltage percent, it's the direct request to hub for such property:
```        voltage = hub.send(MsgHubProperties(MsgHubProperties.VOLTAGE_PERC, MsgHubProperties.UPD_REQUEST))
        print("Voltage: %s%%" % usbyte(voltage.parameters, 0))
```

JeanPeyreMesMots, 2021-03-04:
I have an error with your code :( : 
"File "etat_batterie.py", line 37, in <module>
    voltage = hub.send(MsgHubProperties(MsgHubProperties.VOLTAGE_PERC, MsgHubProperties.UPD_REQUEST))
NameError: name 'MsgHubProperties' is not defined"

undera, 2021-03-04:
That's general Python import problem. You should import the class before using it.