undera/pylgbst #93: Works with "Control

timg7, 2021-04-10:


timg7, 2021-04-10:
Does pylgbst work with Control+ hubs?
I have demo.py running and it lists the hub, but does not connect:
`11854	INFO	pygatt.backends.gatttool.gatttool	Discovered name of 90:84:2B:62:00:35 as Technic Hub`

undera, 2021-04-10:
In theory, it should be compatible with Control+.
The demo is built for Lego Boost device names. You just need to specify the name of your device (`Technic Hub`) in the connection options.

timg7, 2021-04-11:
Thanks, Andrey.  I have the demo working, so now I have something to build on.  Full installation process on a Raspberry Pi 3 with recent Raspbian:

```
sudo setcap 'cap_net_raw,cap_net_admin+eip' $(which hcitool)  # allows bluetooth scan without root
pip3 install pygatt  # BLE library
pip3 install -U pylgbst  # lego libraries

wget https://github.com/undera/pylgbst/raw/master/examples/demo.py
python3 demo.py -c 'auto://90:84:2B:62:00:35?hub_name="Technic Hub"' -d led_colors
```

nategadzhi, 2022-01-24:
👋 Late to the party here. I'm working on a hobby project with Technic Medium hubs, and they pick up just fine. The technic motors are not automatically recognized, and port mappings on the hub might be a bit different, so I'll try to work on those and do a pull request. It's probably easiest to try and port the specific implementations from node-poweredup.