undera/pylgbst #94: Connecting with Bleak on Win10

sapka12, 2021-04-16:
tried to connect with Win 10 + python3.8

```
from pylgbst import *
from pylgbst.hub import MoveHub

connection = get_connection_bleak(hub_mac='00:__:__:__:__:__')
hub = MoveHub(connection)

for device in hub.peripherals:
    print(device)

```
and having this error:
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\...\Python\Python38\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\...\Python\Python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\...\boost01\venv\lib\site-packages\pylgbst\comms\cbleak.py", line 50, in <lambda>
    self._connection_thread = threading.Thread(target=lambda: asyncio.run(self._bleak_thread()))
  File "C:\Users\...\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\...\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\Users\...\boost01\venv\lib\site-packages\pylgbst\comms\cbleak.py", line 60, in _bleak_thread
    await bleak.connect(self.hub_mac, self.hub_name)
  File "C:\Users\...\boost01\venv\lib\site-packages\pylgbst\comms\cbleak.py", line 156, in connect
    self._client = BleakClient(self._device.address, self.loop)
TypeError: __init__() takes 2 positional arguments but 3 were given
WARNING:hub:Got only these devices: (None, None, None, None, None, None, None)
Traceback (most recent call last):
  File "C:/Users/.../PycharmProjects/boost01/hello.py", line 5, in <module>
    hub = MoveHub(connection)
  File "C:\Users\...\boost01\venv\lib\site-packages\pylgbst\hub.py", line 221, in __init__
    self._report_status()
  File "C:\Users\...\boost01\venv\lib\site-packages\pylgbst\hub.py", line 238, in _report_status
    name = self.send(MsgHubProperties(MsgHubProperties.ADVERTISE_NAME, MsgHubProperties.UPD_REQUEST))
  File "C:\Users\...\boost01\venv\lib\site-packages\pylgbst\hub.py", line 71, in send
    self.connection.write(self.HUB_HARDWARE_HANDLE, msgbytes)
  File "C:\Users\...\boost01\venv\lib\site-packages\pylgbst\comms\cbleak.py", line 96, in write
    raise ConnectionError('Something went wrong, communication threads not functioning.')
ConnectionError: Something went wrong, communication threads not functioning.
```
I could find the mac of the move hub with pure bleak
```
import asyncio
from bleak import BleakScanner

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```
so the connection is possible.

can you please help me to connect with pylgbst too.

my requirements.txt:
```
bleak==0.11.0
enum-compat==0.0.3
pycparser==2.20
pylgbst==1.2.0
pyserial==3.5
pythonnet==2.5.2
```

undera, 2021-04-16:
Maybe @antoinechauveau can comment on this

antoinechauveau, 2021-04-17:
This is a duplicate of #89, you either need the latest development version of this project instead of release 1.2.0, or downgrade Bleak to below version 0.8.

undera, 2021-04-17:
@sapka12 Can you try installing project from the source code?

sapka12, 2021-04-18:
> This is a duplicate of #89, you either need the latest development version of this project instead of release 1.2.0, or downgrade Bleak to below version 0.8.

thanks @antoinechauveau 
I'll try these options.

sapka12, 2021-04-18:
@undera @antoinechauveau tried the development version and it works fine. thanks

closed 2021-04-18 by sapka12