undera/pylgbst #95: Anyone working on support for the Train/Smart Hub?

reinder, 2021-05-09:
Hi All,

I was wondering if anyone is already working on support for the Train Hub? I did some quick experiments today.
Peripheral searching works, it reports a LED, voltage, current and Port A & B. It reports the train motor as dev_type 0x02.

Reinder

undera, 2021-05-10:
Just FYI, here's the script to list all peripherals with all modes:
```python
        descr = {}
        for dev in hub.peripherals.values():
            descr[str(dev)] = dev.describe_possible_modes()

        with open("descr.json", "w") as fhd:
            json.dump(descr, fhd, indent=True)
```

It might help you investigating what exists there. It would be great if you attach resulting JSON file here, just for those curios.

reinder, 2021-05-18:
Basic support for Smart Hub: (I renamed it because that's the bluetooth name of the device)
See: https://github.com/reinder/pylgbst/commit/dbf014242f0a4fe0a700ba9707a714d1af73c889

Train motor is detected, did not look into it further yet.
Quick tested the color sensor and LED, that seems to work. 

Running the `dev.describe_possible_modes()` results in:
`WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)`

Need to investigate this further.

undera, 2021-05-19:
Will you create a PR from your branch, so I can accept it?

reinder, 2021-05-19:
Sure, for this change or when the train moter is implemented too?

undera, 2021-05-19:
Let's start with this changeОтправлено с устройства Galaxy
-------- Исходное сообщение --------От: Reinder Feenstra ***@***.***> Дата: 19.05.2021  14:00  (GMT+03:00) Кому: undera/pylgbst ***@***.***> Копия: Andrey Pokhilko ***@***.***>, Comment ***@***.***> Тема: Re: [undera/pylgbst] Anyone working on support for the Train/Smart Hub? (#95) 
Sure, for this change or when the train moter is implemented too?

undera, 2021-05-19:
I have merged it, I think it is valuable milestone. Feel free to send more PRs if you improve it. I suggest you doing the changes in separate git branches, that would help you avoiding merge conflicts.

closed 2021-05-19 by undera