undera/pylgbst #97: Unhandled port data: MsgPortValueSingle

oehoe, 2021-05-15:
Thank you for this fantastic library! I'm able connect and send commands to 3 motors connected to a technic hub.
However, If I try to get the angle of a motor I get warnings like: WARNING peripherals Unhandled port data: MsgPortValueSingle({'payload': b'00000000', 'port': 3})

The code I used is based on the example in the documentation, however my motors are not recognized bij the library as Motor_A, Motor_B, Motor_C. So I address them via the port number. Is the warning caused by the different addressing?

```
addr = "66F8E420-8D44-43D6-AF9A-CD65EFA3XXXX"
conn = get_connection_bleak(hub_mac=addr)
hub = MoveHub(conn)

my_motor = Motor(port=0x0,parent=hub)
my_motor.start_speed(-0.5)

steering = EncodedMotor(port=0x3,parent=hub)

def callback(angle):
    print("Angle: %s" % angle)

steering.subscribe(callback, mode=EncodedMotor.SENSOR_ANGLE)

steering.angled(30)
```

undera, 2021-05-16:
Hi,

Your code is a bit wrong. You should almost never instantiate peripheral objects yourself. Hub does it once it receives initialization sequence from hardware.

If your hardware is not MoveHub from LEGO Boost, you should use the base `Hub` class (or write own child class of it). Here's how your code would look approximately:

```python
addr = "66F8E420-8D44-43D6-AF9A-CD65EFA3XXXX"
conn = get_connection_bleak(hub_mac=addr)
hub = Hub(conn)

# probably some code here to wait for all required motors to appear in `hub.peripherals` map

my_motor = hub.peripherals[0x0]
my_motor.start_speed(-0.5)

steering = hub.peripherals[0x3]

def callback(angle):
    print("Angle: %s" % angle)

steering.subscribe(callback, mode=EncodedMotor.SENSOR_ANGLE)

steering.angled(30)
```

Please let me know if it helped.

oehoe, 2021-05-16:
Thanks for your reply. I've adjusted the code to 

```
hub = Hub(conn)

Sleep(5)

my_motor = hub.peripherals[0x0]
my_motor.start_speed(-0.5)
```

As the peripheral 0x0 is not recognized as a motor I get a AtrributeError. See screenshot below.

[image]

I'm using the library with python 3.8 on Mac OS with bleak. I had to manually install the library from github because there is a fix on github with bleak which is not in the version on pip. I'll be trying on a raspberry pi to see if I have the same issue.

Thanks for your help.

undera, 2021-05-16:
I have made a commit into dev version that should recognize more devices now. Can you update your library from GitHub and share the log again? The log really helps investigating it.

oehoe, 2021-05-16:
Yes! Now the motors are recognized. It seems there is still an issue with the onboard sensors. I see in the code of hub.py that these are commented out.

[image]

undera, 2021-05-16:
Looks good that we have some progress. Unfortunately, I don't have access to same hardware you have, to implement the sensors.

If you can run this snippet and post here the resulting JSON file, it would help.
```python
descr = {}
for dev in hub.peripherals.values():
            descr[str(dev)] = dev.describe_possible_modes()

with open("descr.json", "w") as fhd:
            json.dump(descr, fhd, indent=True)
```

oehoe, 2021-05-16:
Yes, I tried that yesterday as I saw it in a reply to a different issue but I get an error.

[image]

undera, 2021-05-16:
Does it generate the JSON file?

oehoe, 2021-05-16:
No it keeps printing the last warning. The code doesn't continue.

oehoe, 2021-05-16:
If I do `print(hub.peripherals.values())`
I get the following
dict_values([EncodedMotor on port 0x0, EncodedMotor on port 0x1, EncodedMotor on port 0x3, LEDRGB on port 0x32, Current on port 0x3b, Voltage on port 0x3c, Peripheral on port 0x3d, Peripheral on port 0x60, Peripheral on port 0x61, Peripheral on port 0x62, TiltSensor on port 0x63, Peripheral on port 0x64])

undera, 2021-05-16:
Ok, I see. Somebody that has access to that hardware needs to contribute the sensor implementation. BTW which exactly model do you use?

oehoe, 2021-05-16:
It's the hub which came with 42114 Lego Volvo truck. The hub looks like LEGO 88012 Powered Up Hub. In the battery compartment it says "Model: Hub No. 2". Thanks for your help so far!

[image]

undera, 2021-05-19:
The original issue has been solved

closed 2021-05-19 by undera