undera/pylgbst #100: Mindstorms Inventor does not connect

ChristophGaukel, 2021-08-12:
Hello,
in your file __init__.py I did set the logging to DEBUG, done by: logging.basicConfig(level=logging.DEBUG)
then I did run this program:

from pylgbst import get_connection_gatt

conn = get_connection_gatt(hub_mac='A8:E2:C1:9B:AC:29')
print('connection:', conn)

Its output:
DEBUG:comms-gatt:Starting DeviceManager...
INFO:comms-gatt:Discovering devices...
DEBUG:comms-gatt:Devices: dict_values([<gatt.gatt_linux.Device object at 0x7f093f99f790>, <gatt.gatt_linux.Device object at 0x7f093f03c9d0>, <gatt.gatt_linux.Device object at 0x7f093f03cbb0>, <gatt.gatt_linux.Device object at 0x7f093f03cd90>])
DEBUG:comms:Checking device: Hugo, MAC: 00:16:53:42:2b:99
DEBUG:comms:Checking device: LEGO Hub@Egon, MAC: a8:e2:c1:9b:ac:29
INFO:comms:Found LEGO Hub@Egon at a8:e2:c1:9b:ac:29
INFO:comms-gatt:Waiting for device connection...
DEBUG:comms-gatt:Sleeping...
DEBUG:comms-gatt:Sleeping...
DEBUG:comms-gatt:Sleeping...
DEBUG:comms-gatt:Sleeping...
...

It does never connect. Do you know, what happens?

undera, 2021-08-12:
Hello,
This library is for PoweredUp-compatible devices. It would not work with EV3-devices, because of different technology. Is "Mindstorms Inventor" a kind of PoweredUp device?

ChristophGaukel, 2021-08-12:
Mindstorms Inventor (Set 51515) is quite similar to Spike Prime (Set 45678) and quite different from EV3. Yes, it is a PoweredUp device. At least LEGO tells so.

undera, 2021-08-13:
if it's a PoweredUp device, then the issue might be in the BLE library. I'd suggest using Bleak as the most modern lib.

chey, 2021-08-23:
I am curious if this library does work for Robot Inventor (https://www.lego.com/en-us/product/robot-inventor-51515)

Any luck @ChristophGaukel ?

cpseager, 2021-08-23:
Just for clarity, Spike/RI hub opens a more traditional bluetooth v2.1 virtual serial port and communicates to the computer that way. It does not use BLE in normal use with the LEGO app. It's much more like EV3 than the other powered up hubs in that respect.

However it does also support BLE, so that it can connect to and control other powered up hubs/remotes (and bluetooth joysticks etc.). However you would probably need to program it to do that in python on the hub itself.