undera/pylgbst #101: BLUETOOTH DONGLE LIMITATIONS

ekaggrat, 2021-08-17:
why is the current version limited to the dongle mentioned? Have you tried any of the modern bluetooth built in modules in the recent laptops? 

thanks

undera, 2021-08-17:
The limitation comes from the lack of Python BLE libraries on Windows. 
On Linux/Mac it works with many different BLE adapters.

Today, there is a chance that new Bleak library would work with variety of dongles. Feel free to try it yourself.

ekaggrat, 2021-08-17:
thanks for your reply. I will try it on 3 machines to see which works.. So on a linux/mac machine any dongle would work or is there a bare minimum version needed?

undera, 2021-08-17:
Sorry, I don't know that. I did not test all the possible dongles. I just know that for multiple people it did work.