undera/pylgbst #102: python version

ekaggrat, 2021-08-19:
which version of python is this compatible with? does it work with 3.9?

undera, 2021-08-19:
It supports Python 3.x by design. It is supposed to work fine with 3.9

ekaggrat, 2021-08-20:
thanks for reply and patience. I got it installed and running but now i cant get it to connect using bleak. I tried both on mac os and windows .Under windows i can pair it using the bluetooth app under windows but still i cant get it to connect it to python. Under mac it is not even visible to the os bluetooth app. Anything getting blocked?

undera, 2021-08-20:
I never heard of anything blocking it.

In many cases, the issue is with identifying the device for connection. The most direct way is to specify `hub_mac` in connection properties. This issue https://github.com/undera/pylgbst/issues/103 has good example of such connection configuration.