undera/pylgbst #103: Running two Hubs in parallel blocks (using bleak)

uwedamm, 2021-08-19:
I try to run two Lego Hubs in parallel, as follows
```
# lego ctrl hub
input("conn_ctrl (enable CTRL Hub, than ENTER)")
conn_ctrl = get_connection_bleak(hub_mac='90:84:2B:53:C3:BE',hub_name='LEGO Move Hub1')
hub_ctrl = MoveHub(conn_ctrl)

# lego boost hub
input("conn_boost (enable boost Hub, than ENTER)")
conn_boost = get_connection_bleak(hub_mac='00:16:53:AD:63:4D',hub_name='Move Hub 1')
hub_boost = MoveHub(conn_boost)
```

The initial connection works fine, the LEDs both light blue, but creating the second `MoveHub(conn_boost)` object does not return.
I started kind of debugging (early beginner in python, little experience in object oriented languages) and realized that
the `Hub` class is waiting for a sync-reply, which does not reach the correct MoveHub instance, but the other one. ("Move Hub 1" is waiting for sync-reply, but "Lego Move Hub1" receives the answer

```
DEBUG:hub:Waiting for sync reply to Move Hub 1: MsgHubProperties({'payload': b'0105', 'needs_reply': True, 'property': 1, 'operation': 5, 'parameters': b''})
DEBUG:hub:Uwe: get sync replies: Move Hub 1
DEBUG:comms-bleak:Request: 14 ['0x5', '0x0', '0x1', '0x1', '0x5']
DEBUG:comms-bleak:uwe, write: descriptor: None
DEBUG:comms-bleak:Response: 13 ['0xf', '0x0', '0x1', '0x1', '0x6', '0x4d', '0x6f', '0x76', '0x65', '0x20', '0x48', '0x75', '0x62', '0x20', '0x31']
DEBUG:comms-bleak:uwe set_notify_handler, c: descriptor: None
DEBUG:comms-bleak:uwe _safe_handler, handler= 13, data= bytearray(b'\x0f\x00\x01\x01\x06Move Hub 1')
DEBUG:hub:_notify: Notification on LEGO Move Hub1: 13: b'0f000101064d6f7665204875622031'
DEBUG:hub:Decoded message: LEGO Move Hub1: MsgHubProperties({'payload': b'4d6f7665204875622031', 'needs_reply': False, 'property': 1, 'operation': 6, 'parameters': b'4d6f7665204875622031'})
DEBUG:hub:UWE_notify: got upstream msg: LEGO Move Hub1: MsgHubProperties({'payload': b'4d6f7665204875622031', 'needs_reply': False, 'property': 1, 'operation': 6, 'parameters': b'4d6f7665204875622031'})
DEBUG:hub:UWE_notify: sync_request not set: LEGO Move Hub1
DEBUG:hub:Handling msg with LEGO Move Hub1: <bound method Button._props_msg of Button on port 0x0>: MsgHubProperties({'payload': b'4d6f7665204875622031', 'needs_reply': False, 'property': 1, 'operation': 6, 'parameters': b'4d6f7665204875622031'})
```

Note: Please be not confused, I extended the Logging a bit.

Within cbleak.py, the `resp_queue` is defined at a global place, not within the class itself.
I expect this is due to the fact, that _safe_handler must be a @staticmethod, which cannot access a class member

How does the _processing member of Bleakdriver differentiate the two instances of the MoveHub
(The _notify member of which instance of the Hub will be called?)
Or is this somehow random (like I expect). This would explain, the behavior.

If my assumption above is right, could it be an option to define an array for `resp_queue`, with a separate element for each Hub instance?

Thanks for reading! and any kind of feedback

Uwe

undera, 2021-08-20:
Hi,
On 4th of August, I already moved the queues into connection class: https://github.com/undera/pylgbst/commit/7dc8f19af109844f2e1ebe38c818abaa57ec1061

To try that, you'll need to install the source code version. It is easy, just download the source code ZIP, and do `pip install .` from source directory.

Let me know if it helps your issue.

uwedamm, 2021-08-20:
A bit like "RTFM" :-)
Thank you very much! Works fine now. I will hold you informed, if I experience further issues while experimenting :-)
Btw: Did I say thank you for that great project!!!
It was a pleasure to read your code. I learned a lot about python and am now able to understand, what is going on (at least for parts of it)
Uwe

closed 2021-08-20 by uwedamm

undera, 2021-08-20:
My pleasure