undera/pylgbst #104: Can't connect boost

SvyatoslavNovikov, 2021-09-05:
Hi there. I try connect lego boost on my Mac using Bluetooth ASUS USB-BT400 and next code

`from pylgbst.hub import MoveHub
from pylgbst import get_connection_bleak

conn = get_connection_bleak(hub_mac='00:16:53:AC:8D:83')
hub = MoveHub(conn)

for device in hub.peripherals:
    print(device)`

so, I have next stack trace

`Got only these devices: (None, None, None, None, None, None, None)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/novikovs/PycharmProjects/tryPylgbst/venv39/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 50, in <lambda>
    self._connection_thread = threading.Thread(target=lambda: asyncio.run(self._bleak_thread()))
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/Users/novikovs/PycharmProjects/tryPylgbst/venv39/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 60, in _bleak_thread
    await bleak.connect(self.hub_mac, self.hub_name)
  File "/Users/novikovs/PycharmProjects/tryPylgbst/venv39/lib/python3.9/site-packages/pylgbst/comms/cbleak.py", line 157, in connect
    raise ConnectionError('Device not found.')
ConnectionError: Device not found.`

While code is work I turn on boost hub twice, but device doesn't connect. What I do wrong?

undera, 2021-09-05:
Looks like no communication comes back from the device. Might be caused by BLE adapter, libraries problem or anything else. 
Possible way to get more info is to enable DEBUG level logging and see if anything relevant is printed into console log

geekynils, 2021-09-19:
Interesting I have a similar problem using the Bluetooth chip built into my iMac. It does not report any devices. However I also noticed if I first connect to the hub using [scratch](https://scratch.mit.edu) then close the scratch app and then try to connect with pylgbst (before the hub automatically turns off) it works:

1367    INFO    comms   Found LEGO Move Hub at 41051A14-C962-4E6F-9A1B-77CEA31F90DB
1367    INFO    comms-bleak Device matched: 41051A14-C962-4E6F-9A1B-77CEA31F90DB: LEGO Move Hub
1877    INFO    hub Attached peripheral: EncodedMotor on port 0x0
1888    INFO    hub Attached peripheral: EncodedMotor on port 0x1
1898    INFO    hub Attached peripheral: EncodedMotor on port 0x2
1909    INFO    hub Attached peripheral: VisionSensor on port 0x3
1919    INFO    hub Attached peripheral: EncodedMotor on port 0x10
1930    INFO    hub Attached peripheral: LEDRGB on port 0x32
1941    INFO    hub Attached peripheral: TiltSensor on port 0x3a
1951    INFO    hub Attached peripheral: Current on port 0x3b
1962    INFO    hub Attached peripheral: Voltage on port 0x3c
1972    WARNING hub Have no dedicated class for peripheral type 0x42 (UNKNOWN) on port 0x46
1973    INFO    hub Attached peripheral: Peripheral on port 0x46
2170    INFO    hub b'LEGO Move Hub' on b'001653ac3f59'
2282    INFO    hub Voltage: 78%
2377    INFO    race_car    Setting LED color to: PINK
2676    INFO    race_car    Setting LED color to: PURPLE
2888    INFO    race_car    Setting LED color to: BLUE
3084    INFO    race_car    Setting LED color to: LIGHTBLUE
3280    INFO    race_car    Setting LED color to: CYAN
3488    INFO    race_car    Setting LED color to: GREEN
3690    INFO    race_car    Setting LED color to: YELLOW
3888    INFO    race_car    Setting LED color to: ORANGE
4103    INFO    race_car    Setting LED color to: RED
4303    INFO    race_car    Setting LED color to: WHITE
4504    INFO    race_car    Setting LED color to: NONE
4721    INFO    race_car    Setting LED color to: BLACK

Any idea how I could further debug the issue?

macOS Big Sur, bleak 0.12.1 and pylgbst 1.2.2

azazar, 2022-01-14:
It seems that the problem is a lack of Bluetooth LE support from my USB dongle.

ludi81, 2022-01-16:
on mac os x scratch works fine, but the library does not find the boost. Do you have any idea?

sunnyfanuffel, 2022-02-17:
I think I have a similar issue, as new to Python and this kind of project, I have trouble finding an entry point for my search for the cause. The Move hub connects to my Bluetooth on the Laptop (some mediocre Lenovo laptop with Windows 10). Starting the Move Hub pressing green button and running demo.py results (always) in this:
```
Traceback (most recent call last):
  File "C:\Users\jakob.preisenberger\PycharmProjects\pylgbst\examples\demo.py", line 259, in <module>
    hub = MoveHub(**parameters)
  File "C:\Users\jakob.preisenberger\PycharmProjects\pylgbst\pylgbst\hub.py", line 252, in __init__
    self._report_status()
  File "C:\Users\jakob.preisenberger\PycharmProjects\pylgbst\pylgbst\hub.py", line 269, in _report_status
    name = self.send(MsgHubProperties(MsgHubProperties.ADVERTISE_NAME, MsgHubProperties.UPD_REQUEST))
  File "C:\Users\jakob.preisenberger\PycharmProjects\pylgbst\pylgbst\hub.py", line 96, in send
    self.connection.write(self.HUB_HARDWARE_HANDLE, msgbytes)
  File "C:\Users\jakob.preisenberger\PycharmProjects\pylgbst\pylgbst\comms\cbleak.py", line 101, in write
    raise ConnectionError('Something went wrong, communication threads not functioning.')
ConnectionError: Something went wrong, communication threads not functioning.

Process finished with exit code 1
```

undera, 2022-12-02:
I think this issue happens to a very specific software and hardware combination. We need somebody who has that issue to do a deeper troubleshooting to find the root cause and fix.

fekete-robert, 2022-12-03:
Thanks for replying!

I'm happy to help, but probably need some directions to do so :)

Our current situation is: 
We had an older version of pylgbst using a pygatt and an usb bluetooth dongle working with a classic lego boost move hub that was running an old firmware. This setup was working on a linux machine.

Now we have: 
- updated the move hub to the latest firmware (it's working fine from the lego app and scratch)
- the latest pylgbst with bleak, with the same dongle, and trying to get it work on a windows box

We cannot connect ("Got only these devices: (None, None, None, None, None, None, None)"), seems the move hub isn't detected.
If we use a simple test code (see below) with the specific address of the move hub, and get the "Have no dedicated class for peripheral type 0x42 (UNKNOWN) on port 0x46" error.
`
from pylgbst.hub import MoveHub
from pylgbst import get_connection_bleak

conn = get_connection_bleak(hub_mac='00:16:53:A2:22:59')

hub = MoveHub(conn)
`

undera, 2022-12-03:
The last code snippet is the right place to start from.

If you still have access to Linux machine, it would be good to make the piece of code to work on it, then look what is the difference on Windows. Version of the Python, Bleak might matter.

The message about no dedicated class is kind of harmless, but may point to a problem. In [the sources](https://github.com/undera/pylgbst/blob/f5c5c235cc17ab4e369b235efe92224a4e26350b/pylgbst/messages.py#L257), there is indeed no device type is defined for 0x46 (66 decimal). I wonder if it is because of the latest firmware, or do you have any peripheral connected to Hub, or it's a different kind of Hub.

Last thing you can try is to change `MoveHub` class into more generic `Hub`, and add `time.sleep(30)` at the end of code snippet. The generic Hub would not expect any devices, and you'll be able to see only the initialization sequence. Couple it with debug logging enabled and post here the output - I'll be able to understand your situation better.

closed 2023-10-09 by undera