undera/pylgbst #118: Any plans for adding other hubs, maybe Mindstorms 515151?

TravelTrader, 2022-02-17:
Are there any plans for adding other hubs,
especially Mindstorms 515151??

Thank you!

undera, 2022-02-17:
Hello,
Unfortunately, I don't have access to that set, so I can't contribute such support.

Anyone with access to that hub can contribute it. It will be pretty much copying the code of `MoveHub` class and tailoring it to whatever onboard devices are standard for that hub. 

I can be an advisor in that process.

TravelTrader, 2022-02-17:
Wow, thanks.
I’ll collect informations I have at first.

undera, 2022-02-17:
Very useful is to collect whatever Hub reports itself. Here's my script to do that:
```python
import json
import time
import unittest

from pylgbst.hub import MoveHub
from tests import log

class GeneralTest(unittest.TestCase):
    def setUp(self):
        super(GeneralTest, self).setUp()
        self.hub = MoveHub()
        time.sleep(2)
        log.debug("Created hub: %s", self.hub)
        log.warning("\n\n\n")

    def _describe(self):
        descr = {}
        for dev in self.hub.peripherals.values():
            descr[str(dev)] = dev.describe_possible_modes()

        with open("descr.json", "w") as fhd:
            json.dump(descr, fhd, indent=True)

```

TravelTrader, 2022-02-17:
What worked for me is to use ˋscreenˋ to connect and then ^C to get into the MicroPython environment of the hub. That’s really great but only possible for a connection by USB.

If you’d help me to build up a bluetooth connection to the new Hub that creates a serial port like ˋ/dev/ttyBluetoothˋ then it would be awesome to use the ˋscreenˋ-method like above.

Do you know how to create a serial connection to a bluetooth connected device?

undera, 2022-02-18:
Wow, that's a bit complex for me with the `screen` thing... 

With PoweredUP hubs, usually you can take the `Hub` class and connect to it either using the device name, or knowing its MAC.
If you enable DEBUG level logging, the discovered devices would be printed into console, so you can find out what's the name and MAC of the hub. From there you can copy the `MoveHub` class and start changing it. For example, putting the `DEFAULT_NAME` of hubs of this kind.

Hope that helps you somehow.