undera/pylgbst #123: Connecting to WeDo SmartHub: no characteristic found

erjosito, 2022-10-09:
Hey there, first of all thanks @undera for creating this library! I am trying to connect to a Lego WeDo 2.0 SmartHub. I am a noob, but I read it should be compatible with Boost. I get this stack trace, does it sound like something easy to fix? Thanks!

```
WARNING:pygatt.device:No characteristic found matching 00001624-1212-efde-1623-785feabcd123
Traceback (most recent call last):
  File "./demo2.py", line 258, in <module>
    main()
  File "./demo2.py", line 18, in main
    movehub = SmartHub(conn)
  File "/usr/local/lib/python3.7/dist-packages/pylgbst-1.2.2-py3.7.egg/pylgbst/hub.py", line 359, in __init__
  File "/usr/local/lib/python3.7/dist-packages/pylgbst-1.2.2-py3.7.egg/pylgbst/hub.py", line 74, in __init__
  File "/usr/local/lib/python3.7/dist-packages/pylgbst-1.2.2-py3.7.egg/pylgbst/comms/cpygatt.py", line 54, in set_notify_handler
  File "/usr/local/lib/python3.7/dist-packages/pygatt/device.py", line 206, in subscribe
    self._notification_handles(uuid)
  File "/usr/local/lib/python3.7/dist-packages/pygatt/device.py", line 178, in _notification_handles
    value_handle = self.get_handle(uuid)
  File "/usr/local/lib/python3.7/dist-packages/pygatt/device.py", line 273, in get_handle
    raise exceptions.BLEError(message)
pygatt.exceptions.BLEError: No characteristic found matching 00001624-1212-efde-1623-785feabcd123
Exception ignored in: <function Hub.__del__ at 0x76572108>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pylgbst-1.2.2-py3.7.egg/pylgbst/hub.py", line 79, in __del__
  File "/usr/local/lib/python3.7/dist-packages/pylgbst-1.2.2-py3.7.egg/pylgbst/comms/cpygatt.py", line 47, in disconnect
  File "/usr/local/lib/python3.7/dist-packages/pygatt/backends/gatttool/device.py", line 16, in wrapper
pygatt.exceptions.NotConnectedError: 
```

undera, 2022-10-09:
Hi,
It looks like WeDo type of hub has some different identifiers of hardware. These identifiers are hardcoded here: https://github.com/undera/pylgbst/blob/4e90f9e2ce8cd24640e39d3a2670376aa8401890/pylgbst/comms/__init__.py#L18

If you know the right values, you can override those in the beginning of your script. That would require some experimenting and researching, not too easy.

erjosito, 2022-10-09:
Wow that was quick, thanks for your answer! I will check if Dr. Google can find if anybody has cracked the UUID for the WeDo hardware, otherwise I guess I will try another route. Thanks @undera !

erjosito, 2022-10-09:
Mmh, weird the docs in https://github.com/LEGO/lego-ble-wireless-protocol-docs seem to indicate that WeDo 2.0, Boost and Train share the same hardware UUID...

undera, 2022-10-09:
You may also want to try with different underlying BLE library. Specifically the `bleak` looks the most stable today. Maybe it would behave differently.

erjosito, 2022-10-09:
Ah, found it in https://github.com/pybricks/technical-info/blob/master/assigned-numbers.md: `00001523-1212-efde-1523-785feabcd123`

erjosito, 2022-10-09:
Mmh, now it times out with
```
5432    DEBUG   pygatt.backends.gatttool.gatttool       Sending cmd=char-write-req 0x0f 0100
```
(bleak not working for me on a raspi 3)

undera, 2022-10-10:
This is nice that you have some progress. I would do experiment using a regular computer, to remove potential problematic chain for experiment.