undera/pylgbst #124: Blue Tooth Connection

JaeYoungLee79, 2022-11-06:
Dear Sir

------------------------------
(base) C:\Users\eojin>C:/Users/eojin/anaconda3/python.exe e:/pylgbst-master/examples/demo.py
44      INFO    root    Trying get_connection_bluepy
48      INFO    root    Trying get_connection_bluegiga
120     INFO    pygatt.backends.bgapi.bgapi     Initialized new BGAPI backend
121     INFO    pygatt.backends.bgapi.bgapi     Auto-detecting serial port for BLED112
146     INFO    root    Trying get_connection_gatt
185     INFO    root    Trying get_connection_bleak
270     INFO    comms-bleak     Discovering devices... Press green button on Hub
C:\Users\eojin\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py:142: FutureWarning: The discover function will removed in 
a future version, use BleakScanner.discover instead.
  devices = await discover(timeout=1)
11019   WARNING hub     Got only these devices: (None, None, None, None, None, None, None)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\eojin\anaconda3\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\eojin\anaconda3\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\eojin\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 50, in <lambda>     
    self._connection_thread = threading.Thread(target=lambda: asyncio.run(self._bleak_thread()))      
  File "C:\Users\eojin\anaconda3\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\eojin\anaconda3\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "C:\Users\eojin\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 60, in _bleak_thread
    await bleak.connect(self.hub_mac, self.hub_name)
  File "C:\Users\eojin\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 157, in connect     
    raise ConnectionError('Device not found.')
ConnectionError: Device not found.
----------------------------------------------------------------------------------

How do I do?
Or before starting your example.
How to check connection?

Thanks
Best Regards

JaeYoungLee79, 2022-11-06:
Windows User must have a "[Bluegiga BLED112 Bluetooth Smart Dongle]"?
I'm already use Scratch 3.0 with Lego boost.

undera, 2022-11-06:
To check the connection possibility, you can just use
```python
import logging
import time

from pylgbst import get_connection_auto

logging.basicConfig(level=logging.DEBUG)
conn = get_connection_auto()
conn.enable_notifications()

time.sleep(60) # wait 1 minute to gather info
```

That would give some more debugging information, might help understanding what does not work.

The requirement for BLED112 requirement is written for "the Windows version of `pygatt`".

On Windows, different types of BLE hardware and different versions of libraries are known to have problems. It requires some experimentation and troubleshooting sometimes. I don't have any better solution for now.

JaeYoungLee79, 2022-11-06:
Refer to https://github.com/rphuang/IotDevicesPy
There is the explanation with Install at windows
It works. & I try to check deeply for my thinking. 
Anyway, Robot moving.

I will compare your code.

Thanks
Best Regards

HotDog702, 2023-03-07:
Have got the same issue ((
`runfile('D:/Python/LEGO/pylgbst-master/examples/demo.py', wdir='D:/Python/LEGO/pylgbst-master/examples')
17636	INFO	root	Trying get_connection_bleak
17702	INFO	comms-bleak	Discovering devices... Press green button on Hub
C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
28317	WARNING	hub	Got only these devices: (None, None, None, None, None, None, None)
Exception in thread Thread-6:
Traceback (most recent call last):
  File "C:\Users\HUAWEI\anaconda3\lib\threading.py", line 980, in _bootstrap_inner
    self.run()
  File "C:\Users\HUAWEI\anaconda3\lib\threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 51, in <lambda>
    self._connection_thread = threading.Thread(target=lambda: asyncio.run(self._bleak_thread()))
  File "C:\Users\HUAWEI\anaconda3\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\HUAWEI\anaconda3\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 65, in _bleak_thread
    await bleak.connect(self.hub_mac, self.hub_name, **kwargs)
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 165, in connect
    raise ConnectionError('Device not found.')
ConnectionError: Device not found.`

HotDog702, 2023-03-07:
Tried to do this:
<pre>
`from pylgbst.hub import MoveHub
from pylgbst import get_connection_bleak

conn = get_connection_bleak(hub_name='Move Hub')

hub = MoveHub(conn)

for device in hub.peripherals:
    print(device)`
</pre>
The connection looks to established (blue constant LED on MoveHub), but MoveHub very quickly switched off.
<pre>
`C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
Got only these devices: (None, None, None, None, None, None, None)
Exception in thread Thread-6:
Traceback (most recent call last):
  File "C:\Users\HUAWEI\anaconda3\lib\threading.py", line 980, in _bootstrap_inner
    self.run()
  File "C:\Users\HUAWEI\anaconda3\lib\threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 51, in <lambda>
    self._connection_thread = threading.Thread(target=lambda: asyncio.run(self._bleak_thread()))
  File "C:\Users\HUAWEI\anaconda3\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\HUAWEI\anaconda3\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 66, in _bleak_thread
    await bleak.set_notify_handler((self._safe_handler, self.resp_queue))
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 215, in set_notify_handler
    await self._client.start_notify(MOVE_HUB_HW_UUID_CHAR, c)
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\bleak\__init__.py", line 649, in start_notify
    await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\bleak\backends\winrt\client.py", line 890, in start_notify
    await winrt_char.write_client_characteristic_configuration_descriptor_async(
OSError: [WinError -2147467260] Operation aborted`
</pre>

HotDog702, 2023-03-07:
The result of 
<pre>
```
import logging
import time

from pylgbst import get_connection_auto

logging.basicConfig(level=logging.DEBUG)
conn = get_connection_auto()
conn.enable_notifications()

time.sleep(60) # wait 1 minute to gather info
```
</pre>
Connection seems to established:
<pre>
INFO:root:Trying get_connection_bleak
DEBUG:asyncio:Using selector: SelectSelector
INFO:comms-bleak:Discovering devices... Press green button on Hub
C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
DEBUG:bleak.backends.winrt.scanner:Received 00:00:00:00:00:00: Move Hub.
DEBUG:bleak.backends.winrt.scanner:skipping callback, waiting for scan response
DEBUG:bleak.backends.winrt.scanner:Received 00:00:00:00:00:00: Move Hub.
DEBUG:bleak.backends.winrt.scanner:skipping callback, waiting for scan response
DEBUG:bleak.backends.winrt.scanner:Received 00:00:00:00:00:00: Move Hub.
DEBUG:bleak.backends.winrt.scanner:skipping callback, waiting for scan response
DEBUG:bleak.backends.winrt.scanner:Received 6F:F7:DE:81:9E:77: .
DEBUG:bleak.backends.winrt.scanner:skipping callback, waiting for scan response
DEBUG:bleak.backends.winrt.scanner:Received 00:16:53:AF:A1:FF: .
DEBUG:bleak.backends.winrt.scanner:skipping callback, waiting for scan response
DEBUG:bleak.backends.winrt.scanner:Received 00:16:53:AF:A1:FF: Move Hub.
DEBUG:bleak.backends.winrt.scanner:Received 00:16:53:AF:A1:FF: .
DEBUG:bleak.backends.winrt.scanner:Received 00:16:53:AF:A1:FF: Move Hub.
DEBUG:bleak.backends.winrt.scanner:1 devices found. Watcher status: 3.
DEBUG:comms-bleak:Devices: [BLEDevice(00:16:53:AF:A1:FF, Move Hub)]
DEBUG:comms-bleak:00:16:53:AF:A1:FF: Move Hub
DEBUG:comms:Checking device: Move Hub, MAC: 00:16:53:AF:A1:FF
INFO:comms:Found Move Hub at 00:16:53:AF:A1:FF
INFO:comms-bleak:Device matched: BLEDevice(00:16:53:AF:A1:FF, Move Hub)
DEBUG:bleak.backends.winrt.client:Connecting to BLE device @ 00:16:53:AF:A1:FF
DEBUG:bleak.backends.winrt.client:getting services (service_cache_mode=None, cache_mode=None)...
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000001EF8E932F90>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
**DEBUG:comms-bleak:Connection status: True**
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000001EF8E932F10>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED
DEBUG:bleak.backends.winrt.client:closing requester
DEBUG:bleak.backends.winrt.client:closing session
Exception in thread Thread-6:
Traceback (most recent call last):
  File "C:\Users\HUAWEI\anaconda3\lib\threading.py", line 980, in _bootstrap_inner
    self.run()
  File "C:\Users\HUAWEI\anaconda3\lib\threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 51, in <lambda>
    self._connection_thread = threading.Thread(target=lambda: asyncio.run(self._bleak_thread()))
  File "C:\Users\HUAWEI\anaconda3\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\HUAWEI\anaconda3\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 66, in _bleak_thread
    await bleak.set_notify_handler((self._safe_handler, self.resp_queue))
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\pylgbst\comms\cbleak.py", line 215, in set_notify_handler
    await self._client.start_notify(MOVE_HUB_HW_UUID_CHAR, c)
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\bleak\__init__.py", line 649, in start_notify
    await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
  File "C:\Users\HUAWEI\anaconda3\lib\site-packages\bleak\backends\winrt\client.py", line 890, in start_notify
    await winrt_char.write_client_characteristic_configuration_descriptor_async(
OSError: [WinError -2147467260] Operation aborted
</pre>