undera/pylgbst #128: Crashes from time to time calling bluepy function

smartheit, 2023-01-18:
Hello,
thank you for this cool library.
For a trade show we want to use Vernie controlled by an Embeded Board. Whenever a trigger is detected Vernie should drive a specific route.
Most of the time it works but sometimes we get this error:

ERROR:comms-bluepy:Exception in call dispatcher thread
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/pylgbst-1.2.2-py3.9.egg/pylgbst/comms/cbluepy.py", line 64, in _dispatch_calls
    self._peripheral.waitForNotifications(1.)
  File "/usr/lib/python3.9/site-packages/bluepy/btle.py", line 592, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/lib/python3.9/site-packages/bluepy/btle.py", line 427, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/usr/lib/python3.9/site-packages/bluepy/btle.py", line 351, in _waitResp
    if self._helper.poll() is not None:
AttributeError: 'NoneType' object has no attribute 'poll'

Than the whole application hangs and the only way to recover is to restart and pair Vernie again.
Any Ideas on how to get around this?

Package    Versions:
---------- -------
bluepy     1.3.0
h5py       3.4.0
numpy      1.21.2
pip        21.2.4
pylgbst    1.2.2
pyserial   3.5
setuptools 57.4.0
six        1.16.0

undera, 2023-01-18:
Hi
This may be related to the way you organized your code. Can it be that you initiate the hub object multiple times? 
It would be the best just to share your code snippet here.

undera, 2023-01-18:
Looks like this issue: https://github.com/IanHarvey/bluepy/issues/351 or https://github.com/IanHarvey/bluepy/issues/172

And the bluepy lib looks as unmaintained. Maybe worth trying with Bleak as the most modern one.

smartheit, 2023-01-19:
Hi Andrey,
thanks for your answer.

The test code is based on your example android_remote.py. It is a sever - client test where you can specify the vernie task via command line. The shell script test.sh continuously switches the LED on and off by calling the client "python3 test3_client.py LED,1,0". Sooner or later the creas
[test_vernie.zip](https://github.com/undera/pylgbst/files/10454458/test_vernie.zip)

Regards
Marco

undera, 2023-01-19:
From what I read in those issues in bluepy, this is quite possible. There is an issue of thread safety in the library. Since the library is not quite maintained.
I suggest you to check Bleak first.

Another strategy could be to have the script continuously running. The issue relates exactly to the process of shutting down the bluepy connection.

smartheit, 2023-01-26:
Just to inform you: I am still trying to port bleak onto the system. It requires dbus and this is not available on the embedded board. Thus I need to modify the yocto project in order to get that running.

smartheit, 2023-02-22:
I moved to bleak now. I don't get crashes but Vernie switches off after some time and than pylgbst cannot issue commands anymore. The python application needs to be restarted and Vernie needs to be powered on again. Do you know by coincidence how to avoid this auto switch off?
Thanks
Marco

smartheit, 2023-02-22:
I changed the battery, so 100% is displayed:

INFO:root:Trying get_connection_bleak
INFO:comms-bleak:Discovering devices... Press green button on Hub
/usr/local/lib/python3.10/dist-packages/pylgbst/comms/cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
INFO:comms:Found LEGO Move Hub at 00:16:53:C5:EC:43
INFO:comms-bleak:Device matched: BLEDevice(00:16:53:C5:EC:43, LEGO Move Hub)
INFO:hub:Attached peripheral MOTOR_INTERNAL_TACHO => EncodedMotor on port 0x0
INFO:hub:Attached peripheral MOTOR_INTERNAL_TACHO => EncodedMotor on port 0x1
INFO:hub:Attached peripheral VISION_SENSOR => VisionSensor on port 0x2
INFO:hub:Attached peripheral MOTOR_EXTERNAL_TACHO => EncodedMotor on port 0x3
INFO:hub:Attached peripheral MOTOR_INTERNAL_TACHO => EncodedMotor on port 0x10
INFO:hub:Attached peripheral RGB_LIGHT => LEDRGB on port 0x32
INFO:hub:Attached peripheral TILT_INTERNAL => TiltSensor on port 0x3a
INFO:hub:Attached peripheral CURRENT => Current on port 0x3b
INFO:hub:Attached peripheral VOLTAGE => Voltage on port 0x3c
WARNING:hub:Have no dedicated class for peripheral type 0x42 (UNKNOWN) on port 0x46
INFO:hub:Attached peripheral UNKNOWN => Peripheral on port 0x46
INFO:hub:b'LEGO Move Hub' on b'001653c5ec43'
INFO:hub:Voltage: 100%
Vernie the Robot is ready.

It wouldn't be a problem if the Software could recover when pressing the connection button again but in this case a restart of the program is required.

smartheit, 2023-03-01:
Hi Andrey,
Is there a way to have a kind of ping mechanism in the pylgbst library to detect if Vernie is still up?
With the current setup Vernie goes to sleep and the application does not get it. 
Thanks
Marco

undera, 2023-03-01:
Unfortunately, I don't have access to my hardware piece anymore. So I can't experiment with it to understand the problem and how to fix it.
Hopefully, someone else will pick this up to contribute.

smartheit, 2023-03-01:
So your wife should give you a Easter decoration cutting job so you would need one than ;-)

smartheit, 2023-03-01:
One Idea is that the python app takes control and switches Vernie off so that it is in a defined state.
I saw in the code that there is a function     
```
def switch_off(self):
        self.send(MsgHubAction(MsgHubAction.SWITCH_OFF))
```
Does this function switch off the Hub completely? And if so how can it be called using your API?

undera, 2023-03-02:
Yes, that function switches off the hub. Just call it on Hub instance.

smartheit, 2023-03-25:
Hi Andrey,
the exhibition is over and people loved Vernie reacting on hand gestures. Thanks for the library. I am closing now the issue. However, it happened sometimes that pylgbst lib is not reacting anymore and only a restarted solved that.
Cheers
Marco

closed 2023-03-25 by smartheit