undera/pylgbst #129: How to control a DC (train) motor ?

cuernodegazpacho, 2023-02-10:
I notice that there is no class in pylgbst to represent a simple DC motor (Lego 88011). 

In issue #125 there is a suggestion that the encoded motor class should work with the DC motor, but that is not the case. The `timed` method does nothing. Is there any trick to make it work?

undera, 2023-02-10:
Hi,

I never had that DC motor hardware, but it's not too complex to figure out its capabilities. First thing to do is to get the device type ID, which is printed by pylgbst in log. Then, you can try using `Motor` and `EncodedMotor` classes for that device. The `timed()` method is a property of `Motor` base class.

If that does not help, you can query the hardware capabilities (we have a code snippet for that) and know its true functions and mode IDs. I'm here to help you with that.

cuernodegazpacho, 2023-02-10:
Great! Could you then make available that code snippet? I can work with that, and, in case it works, you may be able to include it in your distribution.

undera, 2023-02-10:
Here it is:
```python
descr = {}
for dev in hub.peripherals.values():
            descr[str(dev)] = dev.describe_possible_modes()

with open("descr.json", "w") as fhd:
            json.dump(descr, fhd, indent=True)
```

You can narrow the loop to only the device of motor. The resulting JSON contains all the HW info.

cuernodegazpacho, 2023-02-11:
The motor shows up associated with `port_A` in the hub, its not in the `peripherals` list. This shouldn't be a problem though. But then the code enters in an infinite loop when I try to call `describe_possible_modes()` on the motor instance:
```
from pylgbst.hub import SmartHub

hub = SmartHub(address='86996732-BF5A-433D-AACE-5611D4C6271D')
motor = hub.port_A

print(motor)
print(motor.describe_possible_modes())
```

gives this:

```
EncodedMotor on port 0x0
WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)
WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)
WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)
WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)
WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)
WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)
WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)
WARNING:hub:Command error: Command 0x22 caused error 0x6: Invalid use (e.g. parameter error(s)
....
....
```
So looks like the `EncodedMotor` class is having trouble in understanding the device. Something associated with its parameters? Do you have any hint on how could I proceed?

undera, 2023-02-11:
Any device gets to the `peripherals` map, under its port ID. 

The `describe_possible_modes()` is not Motor-specific, it comes from the base `Peripheral` class (https://github.com/undera/pylgbst/blob/master/pylgbst/peripherals.py#L177)

The error about command 0x22 might be OK, since it would try to make 256 calls of that command. If possible, please wait until that process will complete. It is made with this "bruteforce" approach because of hardware not always announcing the available modes correctly, thus we scan it.

cuernodegazpacho, 2023-02-11:
Gotcha. I noticed now the loop to 256 in `peripherals.py`. Got the mode description json file. Which, as expected, is filed out to 256 modes, most of which are empty. The motor information is at the beginning of the file. Is there an equivalent file for the  encoded motor device? Or should the information in the json file be copied in the source code (I'm thinking about a subclass like `DCMotor` or `TrainMotor`).

cuernodegazpacho, 2023-02-11:
Got the basics working. I just did this:

```
hub = SmartHub(address='86996732-BF5A-433D-AACE-5611D4C6271D')
motor = hub.port_A
motor.start_power(power_primary=1.0 power_secondary=0.)
sleep(3)
```
Speed is controlled by `power_primary`, between -1.0 and 1.0, negative being reverse.

undera, 2023-02-12:
So can you share that resulting JSON for review? At least its non-empty modes. That would help me understand if the motor is of a known class or not.

cuernodegazpacho, 2023-02-12:
This is the non-empty part. 
```
{
 "mode_count": 1,
 "input_modes": [],
 "output_modes": [
  {
   "Mode": 0,
   "Name": "LPF2-TRAIN",
   "Raw range": [
    -100.0,
    100.0
   ],
   "Percent range": [
    -100.0,
    100.0
   ],
   "SI value range": [
    -100.0,
    100.0
   ],
   "Units": "",
   "Mapping": {
    "input": [],
    "output": [
     "Relative [-1..1]",
     "Absolute [min..max]"
    ]
   },
   "Motor bias": 35,
   "Value encoding": {
    "datasets": 1,
    "type": "8 bit",
    "total_figures": 4,
    "decimals": 0
   }
  }
 ],
 "capabilities": {
  "logically_combinable": false,
  "synchronizable": false,
  "can_output": true,
  "can_input": false
 },
 "modes": [
  {
   "Mode": 0,
   "Name": "LPF2-TRAIN",
   "Raw range": [
    -100.0,
    100.0
   ],
   "Percent range": [
    -100.0,
    100.0
   ],
   "SI value range": [
    -100.0,
    100.0
   ],
   "Units": "",
   "Mapping": {
    "input": [],
    "output": [
     "Relative [-1..1]",
     "Absolute [min..max]"
    ]
   },
   "Motor bias": 35,
   "Value encoding": {
    "datasets": 1,
    "type": "8 bit",
    "total_figures": 4,
    "decimals": 0
   }
  },
  {
   "Mode": 1,
   "Name": "",
   "Raw range": [
    0.0,
    1023.0
   ],
   "Percent range": [
    0.0,
    100.0
   ],
   "SI value range": [
    0.0,
    1.0
   ],
   "Units": "",
   "Mapping": {
    "input": [],
    "output": []
   },
   "Motor bias": 0,
   "Value encoding": {
    "datasets": 1,
    "type": "8 bit",
    "total_figures": 4,
    "decimals": 0
   }
  },
  {
   "Mode": 2
  },
  {
   "Mode": 3
  },
  {
   "Mode": 4
  },
  {

```

undera, 2023-02-12:
Additionally, can you tell me the device type ID that is printed into log for this particular device?

cuernodegazpacho, 2023-02-12:
How do I enable debug logging?

I noticed two things that may be relevant, or not. Maybe they're expected: the motor instance can't be subscribed (or at least I couldn't make it work), and, at every `start_power` call, it prints `WARNING:hub:Notification on port with no device: 0`

undera, 2023-02-12:
`logging.basicConfig(level=logging.DEBUG)` in the beginning of your code would enable debug logging

cuernodegazpacho, 2023-02-12:
I'm not sure where the device ID is, so I attached belo the entire output of my test code.
```
INFO:root:Trying get_connection_bleak
DEBUG:asyncio:Using selector: KqueueSelector
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
INFO:comms-bleak:Discovering devices... Press green button on Hub
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 45B950CB-A8D0-4153-BC68-75272A630D3F: None @ RSSI: -75 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fa7034b8c50>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -51 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fa7034b8c50>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -51 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fa7034b8c50>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 1BA63736-C05F-42AB-8793-566A8EF1518D: None @ RSSI: -72 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fa7034b8c50>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 1BA63736-C05F-42AB-8793-566A8EF1518D: None @ RSSI: -72 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>) and Central: <CBCentralManager: 0x7fa7034b8c50>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:comms-bleak:Devices: [BLEDevice(45B950CB-A8D0-4153-BC68-75272A630D3F, None), BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub), BLEDevice(1BA63736-C05F-42AB-8793-566A8EF1518D, None)]
DEBUG:comms-bleak:45B950CB-A8D0-4153-BC68-75272A630D3F: None
DEBUG:comms:Checking device: None, MAC: 45B950CB-A8D0-4153-BC68-75272A630D3F
DEBUG:comms-bleak:86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub
DEBUG:comms:Checking device: Smart Hub, MAC: 86996732-BF5A-433D-AACE-5611D4C6271D
INFO:comms:Found Smart Hub at 86996732-BF5A-433D-AACE-5611D4C6271D
INFO:comms-bleak:Device matched: BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.client:CentralManagerDelegate  at <CentralManagerDelegate: 0x7fa7034913b0>
DEBUG:bleak.backends.corebluetooth.client:Connecting to BLE device @ 86996732-BF5A-433D-AACE-5611D4C6271D
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didConnectPeripheral_
DEBUG:bleak.backends.corebluetooth.client:Retrieving services...
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didDiscoverServices_
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Services discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving characteristics for service 00001623-1212-EFDE-1623-785FEABCD123
[... cut ...]
DEBUG:hub:Decoded message: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Found matching upstream msg: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Handling msg with <bound method Hub._handle_action of <pylgbst.hub.SmartHub object at 0x7fa704c64820>>: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
WARNING:hub:Hub disconnects
DEBUG:hub:Fetched sync reply: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})

Process finished with exit code 0

```

undera, 2023-02-12:
Hang on, I'll need a bit of time to analyze that information.

cuernodegazpacho, 2023-02-12:
No rush.

undera, 2023-02-12:
Ok, here the first observation from JSON: 

 `output_modes` declares just one mode, while direct scanning reveals the second available mode, just like I told it happens.

We need to understand if the mode 0x0 is power or speed. That would require some experiments and might be hard to tell one from another. Not too critical, we can call it just "run".

Judging by 1023 being a raw range on the mode 0x1 I'd suspect it's rather an output (sensor) mode. Worth a try by issuing a `subscribe()` call with that mode. If it is confirmed to be a sensor mode, I can add a value decode function for proper handling. 

If it happens to not be a sensor mode, we can experiment with sending output commands to it.

I have created a PR #130 to reflect the code draft of `TrainMotor`.

Can you take that code from branch and try running it, calling `.run()` method on motor with values ranging from `-1.0..1.0`, to investigate the first output mode. Also, can you experiment with subscribing to second mode? 

You're officially a LEGO hacker now, your turn to act :).

cuernodegazpacho, 2023-02-12:
The motor is no longer running with the `dc-motor` branch. I can send you the debug output if you like.

On another note, the code in both `dc-motor` and `master` seems to be failing in detecting the peripherals attached to the hub. It finds just the `TrainMotor` instance, as shown in this screen capture of the debugger window:

[image]

I had hacked lines 155-156 in `hub.py` (in master) in order to subtract 1 from the port number. That makes the other peripherals to appear in the hub instance. I'm sure there must be a better way to handle this though.

[image]

undera, 2023-02-13:
Which kind of Hub do you use? This one https://www.lego.com/en-pt/product/hub-88009 ? If yes, then we don't have an existing class to support that Hub, so we need to create one (not too complex).

cuernodegazpacho, 2023-02-13:
> Which kind of Hub do you use? This one https://www.lego.com/en-pt/product/hub-88009 ? If yes, then we don't have an existing class to support that Hub, so we need to create one (not too complex).

Yes, that's the hub I'm using. It seems to be almost identical to the movehub, apart from the extra motors and sensors. I can keep using the movehub class for now, won't impair my project (but see below).

> In your log from above, I see the line DEBUG:hub:All devices are present: (LEDRGB on port 0x32, Current on port 0x3b, Voltage on port 0x3c) which means all onboard devices were detected correctly. I'm a bit confused now about your statement about ports...

When I run the code as is, I get a `KeyError` in line 156 in `hub.py`. I modified that line to use `msg.port-1` instead of `msg.port`. That makes the code execute to completion. I am confused as well with my screen captures in my previous comment, since with this "fix" on `msg.port` I get the class instance correctly initialized, corresponding to the second screen capture. I don't know how I got the first one, in which the internal devices are missing. :-( 

Maybe that 1-indexing issue is all that is different between my hub and the movehub? Because everything seems to be working as expected when I run your tests for the internal devices (led, voltage, current, motor). 

I am sending the debug in the next comment.

cuernodegazpacho, 2023-02-13:
```
/Users/busko/Projects/software/miniconda3/envs/ble/bin/python /Users/busko/Projects/pylgbst/examples/sandbox/test3.py 
INFO:root:Trying get_connection_bleak
DEBUG:asyncio:Using selector: KqueueSelector
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
INFO:comms-bleak:Discovering devices... Press green button on Hub
/Users/busko/Projects/pylgbst/pylgbst/comms/cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device A762FDF4-9932-4DD4-87A6-304B4C86D06E: None @ RSSI: -69 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fc6fe4f9b00>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -49 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fc6fe4f9b00>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -49 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fc6fe4f9b00>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 73C77F5F-EA32-49B9-B376-5A42EF234F62: None @ RSSI: -70 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fc6fe4f9b00>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 73C77F5F-EA32-49B9-B376-5A42EF234F62: None @ RSSI: -69 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>) and Central: <CBCentralManager: 0x7fc6fe4f9b00>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF: [TV] Samsung Q6 Series (65) @ RSSI: -58 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fc6fe4f9b00>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -50 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fc6fe4f9b00>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:comms-bleak:Devices: [BLEDevice(A762FDF4-9932-4DD4-87A6-304B4C86D06E, None), BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub), BLEDevice(73C77F5F-EA32-49B9-B376-5A42EF234F62, None), BLEDevice(CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF, [TV] Samsung Q6 Series (65))]
DEBUG:comms-bleak:A762FDF4-9932-4DD4-87A6-304B4C86D06E: None
DEBUG:comms:Checking device: None, MAC: A762FDF4-9932-4DD4-87A6-304B4C86D06E
[... cut ...]
DEBUG:hub:Decoded message: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Found matching upstream msg: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Handling msg with <bound method Hub._handle_action of <pylgbst.hub.SmartHub object at 0x7fc6ffc64310>>: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
WARNING:hub:Hub disconnects
DEBUG:hub:Fetched sync reply: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})

Process finished with exit code 0

```

undera, 2023-02-13:
What confuses me is that I see only three onboard devices expected to appear on the Hub:
`DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)`
Which would only happen if you use `SmartHub` class. For the `MoveHub` class, you would get the 7 device expectation. 

Looking at the third kind of PoweredUp hub (https://www.lego.com/en-pt/product/technic-hub-88012) I see that it has 4 ports. This means the current `SmartHub` class with 2 ports is the right class to use. Maybe that class needs some corrections to the port IDs written in constants `PORT_LED` , although I see that in the code it should be just fine. The juggle with `-1` would drive the code crazy, for sure.

The most strange message I see now is `WARNING:hub:Notification on port with no device: 0`, which is caused by `DEBUG:hub:Detaching peripheral: TrainMotor on port 0x0`. This may just be the result of the `-1` that leads to removal of the motor object instead of keeping it. In this situation the motor would not work for sure.

Can you remove your `-1` customization and share the log of run?

cuernodegazpacho, 2023-02-13:
> What confuses me is that I see only three onboard devices expected to appear on the Hub:
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
Which would only happen if you use SmartHub class. For the MoveHub class, you would get the 7 device expectation.

Sorry, I forgot to say that I am using the `SmartHub` class, not `MoveHub`. I had started my experiments with the movehub but later found that Smarthub works fine.

cuernodegazpacho, 2023-02-13:
With the original code:
```
/Users/busko/Projects/software/miniconda3/envs/ble/bin/python /Users/busko/Projects/pylgbst/examples/sandbox/test3.py 
INFO:root:Trying get_connection_bleak
DEBUG:asyncio:Using selector: KqueueSelector
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
INFO:comms-bleak:Discovering devices... Press green button on Hub
/Users/busko/Projects/pylgbst/pylgbst/comms/cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 19CCBA47-428F-4DC5-A9B4-019E27B65132: None @ RSSI: -78 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fcba0cd68a0>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -53 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fcba0cd68a0>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -53 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fcba0cd68a0>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:comms-bleak:Devices: [BLEDevice(19CCBA47-428F-4DC5-A9B4-019E27B65132, None), BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub)]
DEBUG:comms-bleak:19CCBA47-428F-4DC5-A9B4-019E27B65132: None
DEBUG:comms:Checking device: None, MAC: 19CCBA47-428F-4DC5-A9B4-019E27B65132
DEBUG:comms-bleak:86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub
DEBUG:comms:Checking device: Smart Hub, MAC: 86996732-BF5A-433D-AACE-5611D4C6271D
INFO:comms:Found Smart Hub at 86996732-BF5A-433D-AACE-5611D4C6271D
INFO:comms-bleak:Device matched: BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub)
DEBUG:bleak.backends.corebluetooth.client:CentralManagerDelegate  at <CentralManagerDelegate: 0x7fcba0cd81b0>
DEBUG:bleak.backends.corebluetooth.client:Connecting to BLE device @ 86996732-BF5A-433D-AACE-5611D4C6271D
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didConnectPeripheral_
DEBUG:bleak.backends.corebluetooth.client:Retrieving services...
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didDiscoverServices_
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Services discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving characteristics for service 00001623-1212-EFDE-1623-785FEABCD123
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didDiscoverCharacteristicsForService_error_
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Characteristics discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 00001624-1212-EFDE-1623-785FEABCD123
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didDiscoverDescriptorsForCharacteristic_error_
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 13
[... cut ...]
  File "/Users/busko/Projects/pylgbst/pylgbst/hub.py", line 93, in send
    assert not self._sync_request, "Pending request %r while trying to put %r" % (self._sync_request, msg)
AssertionError: Pending request MsgPortOutput({'payload': b'00110064', 'needs_reply': True, 'port': 0, 'is_buffered': False, 'do_feedback': True, 'wait_complete': True, 'subcommand': 0, 'params': b'64'}) while trying to put MsgHubAction({'payload': b'02', 'needs_reply': True, 'action': 2})

Process finished with exit code 1

```

undera, 2023-02-13:
Alright, now we're back on track. What happens is that Hub sends a detach event for a port that is not attached. I'm not sure what causes this, maybe it's a way for Hub to tell that there is no device attached at the second port.

I have modified the branch code to gracefully handle that situation. Can you take that updated code and try again?

cuernodegazpacho, 2023-02-13:
Run with latest branch code. The hub never connects.
```
/Users/busko/Projects/software/miniconda3/envs/ble/bin/python /Users/busko/Projects/pylgbst/examples/sandbox/test3.py 
INFO:root:Trying get_connection_bleak
DEBUG:asyncio:Using selector: KqueueSelector
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
INFO:comms-bleak:Discovering devices... Press green button on Hub
/Users/busko/Projects/pylgbst/pylgbst/comms/cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:comms-bleak:Devices: []
/Users/busko/Projects/pylgbst/pylgbst/comms/cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:comms-bleak:Devices: []
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:comms-bleak:Devices: []
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
[... 133 lines cut ...]
Traceback (most recent call last):
  File "/Users/busko/Projects/pylgbst/examples/sandbox/test3.py", line 87, in <module>
    demo(hub_1)
  File "/Users/busko/Projects/pylgbst/examples/sandbox/test3.py", line 49, in demo_motor
    motor.run()
AttributeError: 'NoneType' object has no attribute 'run'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/busko/Projects/pylgbst/examples/sandbox/test3.py", line 100, in <module>
    hub_1.disconnect()
[... cut ...]
    self.send(MsgHubAction(MsgHubAction.DISCONNECT))
  File "/Users/busko/Projects/pylgbst/pylgbst/hub.py", line 98, in send
    resp = self._sync_replies.get()
  File "/Users/busko/Projects/software/miniconda3/envs/ble/lib/python3.10/queue.py", line 171, in get
    self.not_empty.wait()
  File "/Users/busko/Projects/software/miniconda3/envs/ble/lib/python3.10/threading.py", line 320, in wait
    waiter.acquire()
KeyboardInterrupt

Process finished with exit code 130 (interrupted by signal 2: SIGINT)

```

undera, 2023-02-13:
This one looks very much off, no information were received from hub at all. Maybe it got battery down, or something else? It does not seem to start the BLE connection at all. Can you investigate that? The normal session would start with `INFO:comms:Found Smart Hub at 86996732-BF5A-433D-AACE-5611D4C6271D`

cuernodegazpacho, 2023-02-13:
I think the computer Bluetooth went down. It happens randomly. Here is the new run. The motor didn't start.
```
/Users/busko/Projects/software/miniconda3/envs/ble/bin/python /Users/busko/Projects/pylgbst/examples/sandbox/test3.py 
INFO:root:Trying get_connection_bleak
DEBUG:asyncio:Using selector: KqueueSelector
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
INFO:comms-bleak:Discovering devices... Press green button on Hub
/Users/busko/Projects/pylgbst/pylgbst/comms/cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF: [TV] Samsung Q6 Series (65) @ RSSI: -64 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7f9fe55959a0>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 6527589E-7452-4CF0-9BA9-7420C75133EE: None @ RSSI: -79 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7f9fe55959a0>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -42 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7f9fe55959a0>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -42 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7f9fe55959a0>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -42 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7f9fe55959a0>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device A71B2DB2-5E50-42B1-9060-00EA4D45BE73: None @ RSSI: -81 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7f9fe55959a0>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device A71B2DB2-5E50-42B1-9060-00EA4D45BE73: None @ RSSI: -81 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>) and Central: <CBCentralManager: 0x7f9fe55959a0>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device A2CDA0A6-4C24-42A3-A937-9707AFC8E281: None @ RSSI: -77 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7f9fe55959a0>
[... cut ...]
DEBUG:hub:Decoded message: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Found matching upstream msg: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Handling msg with <bound method Hub._handle_action of <pylgbst.hub.SmartHub object at 0x7f9fe6c643a0>>: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Fetched sync reply: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
WARNING:hub:Hub disconnects

Process finished with exit code 0

```

undera, 2023-02-13:
So, now we have the healthy Hub startup. 
Also, the command gets to the motor, and hub confirms that the command is completed and motor is now idle. I understand that it's not something very meaningful, but that's what MsgPortOutputFeedback with status 10 means.

Now it would be a time to experiment with values and modes of motor. What I would do is I'd experiment: from JSON we know that
```
"type": "8 bit",
        "total_figures": 4,
```
which might mean that actually it expects 4-byte value as parameter. So worth trying replacing single-byte `<b` with 4-byte `<i`.

There are also other experiments possible, like specifying larger than 100 as value etc.

undera, 2023-02-13:
BTW I looked in other libs that the train motor uses what they call "WriteDirectMode". I have changed the branch code accordingly, worth trying.

cuernodegazpacho, 2023-02-13:
The motor won't run with `<i`. I tried several values for power, none did it. Then I put the `<b` back, and the motor then responds. Again, I tried several values, they all make it spin at top speed. Except zero, which makes it stop. 

I also tried to replace the params variable by the one we had before. That made the motor spin with variable speeds, forward and reverse, and also stop by setting `param=0`. This is the code that worked:
```
    def power(self, param=1.0):
        """
        Power the motor, with value -1.0..1.0
        """
        # if param != 255:  # A special value of 255 would do a hard-stop.
        #     param = abs_scaled_100(param)
        #
        # params = pack("<i", param)
        # self._write_direct_mode(self.SUBCMD_POWER, params)

        params = b""
        params += pack("<b", abs_scaled_100(param))

        self._write_direct_mode(self.SUBCMD_POWER, params)

    def stop(self):
        self.power(0)
```

undera, 2023-02-13:
Have you tried experimenting with SUBCMD_1? Like subscribing with it as a sensor mode

cuernodegazpacho, 2023-02-13:
This is using the`<i` construct and SUBCMD_1
```
/Users/busko/Projects/software/miniconda3/envs/ble/bin/python /Users/busko/Projects/pylgbst/examples/sandbox/test3.py 
INFO:root:Trying get_connection_bleak
DEBUG:asyncio:Using selector: KqueueSelector
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
INFO:comms-bleak:Discovering devices... Press green button on Hub
/Users/busko/Projects/pylgbst/pylgbst/comms/cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -51 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fe8cadd2170>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -51 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fe8cadd2170>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device EE5D0906-D966-42A5-A9B5-4215A37438BA: None @ RSSI: -68 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fe8cadd2170>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device EE5D0906-D966-42A5-A9B5-4215A37438BA: None @ RSSI: -68 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>) and Central: <CBCentralManager: 0x7fe8cadd2170>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF: [TV] Samsung Q6 Series (65) @ RSSI: -62 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fe8cadd2170>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device F9E2B764-20C4-4B2F-84EB-22983DC74095: None @ RSSI: -71 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fe8cadd2170>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:comms-bleak:Devices: [BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub), BLEDevice(EE5D0906-D966-42A5-A9B5-4215A37438BA, None), BLEDevice(CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF, [TV] Samsung Q6 Series (65)), BLEDevice(F9E2B764-20C4-4B2F-84EB-22983DC74095, None)]
DEBUG:comms-bleak:86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub
DEBUG:comms:Checking device: Smart Hub, MAC: 86996732-BF5A-433D-AACE-5611D4C6271D
INFO:comms:Found Smart Hub at 86996732-BF5A-433D-AACE-5611D4C6271D
INFO:comms-bleak:Device matched: BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub)
DEBUG:bleak.backends.corebluetooth.client:CentralManagerDelegate  at <CentralManagerDelegate: 0x7fe8cade60a0>
DEBUG:bleak.backends.corebluetooth.client:Connecting to BLE device @ 86996732-BF5A-433D-AACE-5611D4C6271D
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
[... cut ...]
DEBUG:hub:Fetched sync reply: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
Traceback (most recent call last):
  File "/Users/busko/Projects/pylgbst/examples/sandbox/test3.py", line 87, in <module>
    demo(hub_1)
  File "/Users/busko/Projects/pylgbst/examples/sandbox/test3.py", line 49, in demo_motor
    motor.power()
  File "/Users/busko/Projects/pylgbst/pylgbst/peripherals.py", line 358, in power
    params = pack("<i", param)
struct.error: required argument is not an integer

Process finished with exit code 1

```

cuernodegazpacho, 2023-02-13:
This is using the `<b` construct and SUBCMD_1. The motor doesn't run in either case:
```
/Users/busko/Projects/software/miniconda3/envs/ble/bin/python /Users/busko/Projects/pylgbst/examples/sandbox/test3.py 
INFO:root:Trying get_connection_bleak
DEBUG:asyncio:Using selector: KqueueSelector
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
INFO:comms-bleak:Discovering devices... Press green button on Hub
/Users/busko/Projects/pylgbst/pylgbst/comms/cbleak.py:150: FutureWarning: The discover function will removed in a future version, use BleakScanner.discover instead.
  devices = await bleak.discover(timeout=1, **kwargs)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManagerDidUpdateState_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF: [TV] Samsung Q6 Series (65) @ RSSI: -62 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fab80f10d90>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -51 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fab80f10d90>
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didDiscoverPeripheral_advertisementData_RSSI_
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub @ RSSI: -51 (kCBAdvData <nsdict_keys(['kCBAdvDataLocalName', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fab80f10d90>
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:'isScanning' changed
DEBUG:comms-bleak:Devices: [BLEDevice(CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF, [TV] Samsung Q6 Series (65)), BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub)]
DEBUG:comms-bleak:CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF: [TV] Samsung Q6 Series (65)
DEBUG:comms:Checking device: [TV] Samsung Q6 Series (65), MAC: CBA44D4D-6706-447F-9E1E-C8B8E0B90CAF
DEBUG:comms-bleak:86996732-BF5A-433D-AACE-5611D4C6271D: Smart Hub
DEBUG:comms:Checking device: Smart Hub, MAC: 86996732-BF5A-433D-AACE-5611D4C6271D
INFO:comms:Found Smart Hub at 86996732-BF5A-433D-AACE-5611D4C6271D
INFO:comms-bleak:Device matched: BLEDevice(86996732-BF5A-433D-AACE-5611D4C6271D, Smart Hub)
DEBUG:bleak.backends.corebluetooth.client:CentralManagerDelegate  at <CentralManagerDelegate: 0x7fab7ddd5cb0>
DEBUG:bleak.backends.corebluetooth.client:Connecting to BLE device @ 86996732-BF5A-433D-AACE-5611D4C6271D
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:centralManager_didConnectPeripheral_
DEBUG:bleak.backends.corebluetooth.client:Retrieving services...
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didDiscoverServices_
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Services discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving characteristics for service 00001623-1212-EFDE-1623-785FEABCD123
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didDiscoverCharacteristicsForService_error_
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Characteristics discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 00001624-1212-EFDE-1623-785FEABCD123
DEBUG:hub:Waiting for builtin devices to appear: (None, None, None)
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:peripheral_didDiscoverDescriptorsForCharacteristic_error_
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 13
[... cut ...]
DEBUG:hub:Decoded message: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Found matching upstream msg: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
DEBUG:hub:Handling msg with <bound method Hub._handle_action of <pylgbst.hub.SmartHub object at 0x7fab7f4b83a0>>: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})
WARNING:hub:Hub disconnects
DEBUG:hub:Fetched sync reply: MsgHubAction({'payload': b'', 'needs_reply': False, 'action': 49})

Process finished with exit code 0

```

cuernodegazpacho, 2023-02-14:
@undera Maybe the second mode is just an artifact of the hub's firmware, that only shows up with these "dumb" peripherals? Maybe it's my own smart hub that has this particular bug. The documentation for [BrickNil](https://virantha.github.io/bricknil/lego_api/lego.html#train-motor) shows only one mode.

undera, 2023-02-14:
Alright. I have committed the code by @cuernodegazpacho for the motor support in #130. I guess we can call this a successful library improvement. Any outstanding issues?

cuernodegazpacho, 2023-02-14:
We need to do 
```
        params = pack("<b", abs_scaled_100(param))
```
for the parameter to fall in the acceptable range. Otherwise, a "out of  range" error is raised. With that modification in place, the code in branch `dc_motor` successfully runs the motor and enables variable speeds. I think that closes the issue!

undera, 2023-02-15:
Cool, I have released 1.3.0 with these changes

closed 2023-02-15 by undera