undera/pylgbst #131: Support for Powered Up Remote Handset (Lego 88010)

cuernodegazpacho, 2023-02-18:
I am trying now to add some support for the PU remote control. Found that it basically behaves as a Hub. The difference with other hubs is that it has no external connectors, and has two internally-connected peripherals, the two button sets at left and right. It has internal LED and voltage peripherals as the other hubs, but no current (which makes sense since it is not driving any external device).

I managed to create a slightly modified version of the `SmartHub` class, with which I can successfully control the LED and take voltage readings. However, the button sets are supposed to be associated with ports 0 and 1 (as per output of the `describe_possible_modes` method), but they don't work. When I associate a `Button` instance to either port 0 or 1, I get no response from any of the buttons in the left and right sets, but I **do** get instead a response from the main (green) hub button. Which is not associated to anything in my code.

I suspect either there is a mixup of the ports somehow, or I'm missing something in the way the messages are passed to/from the BLE interface. 

I confirmed the ports 0 and 1 assignations for this device by looking at what other packages do (bricknil and node-poweredup). I can provide the output of the `describe_possible_modes` method, as well as my dev code (it's in a fork of this project).

cpseager, 2023-02-19:
Remote buttons are a unique sensor type (id 55) so you need to add a new sensor class (not Button) to handle this type of sensor.

cuernodegazpacho, 2023-02-19:
@cpseager thanks for the hint! I noticed that `Button` differs from other sensors by lacking its own implementation of `_decode_port_data`. So I created a subclass of `Peripheral` named `RemoteButton`, where I implement the `subscribe` and `unsubscribe` methods, and I added a bare `_decode_port_data` that, for initial debugging,  just prints its parameter. No difference with what I had before. The subscription mechanism works, but the only button that sends callback notifications is the green hub button, which is not even mapped in the class. I can trace the internals of the bluetooth messaging in the computer, and see that nothing seems to come out of the remote buttons when they are pressed. I checked it with two remotes, so it's not a defect in the hardware. I am at loss on what is going on.

cpseager, 2023-02-20:
On the remote the two buttons are configured as ‘external’ sensors (id 55) on ‘internal’ ports 0 and 1. This makes sense, as in this case you cannot configure them as two dedicated ‘internal’ ports (like Voltage or Button ports) as they would both need to use the same port id number.

Therefore a quick scan of your dev branch shows you have appeared to try to configure the RemoteButton as an internal port (like Button) rather than external (like VisionSensor). In particular your subscribe will need to be something along the lines of

def subscribe(self,callback, mode=0, granularity=1):
   super().subscribe(callback, mode, granularity)

cuernodegazpacho, 2023-02-20:
I created PR #132 to implement the remote.

undera, 2023-02-27:
I have added some comments on the code, please address those.