undera/pylgbst #133: Is pylgbst thread-safe?

cuernodegazpacho, 2023-03-01:
I am trying to implement a blinking function for the on-board LED light in a Smart Hub (aka City Hub). I have the blink code running in a thread. By itself, it works fine. It just runs in an infinite loop, swapping the LED color in between two values, with a sleep delay in between changes. There are termination and status flags that help to manage the loop, stopping it and killing the thread under external command. All works as expected when running by itself.

But when I run other code (outside that thread) that also communicates with the hub, such as to command a motor in port A, there are apparently collisions in the messaging system in pylgbst. These happen at random, but more often the faster the blink rate goes, and eventually kill the entire thing.

The errors I get are always of the type:
```
AssertionError: Pending request MsgPortOutput({'payload': b'3211510009', 'needs_reply': True, 'port': 50, 'is_buffered': False, 'do_feedback': True, 'wait_complete': True, 'subcommand': 81, 'params': b'0009'}) while trying to put MsgPortOutput({'payload': b'0011510000', 'needs_reply': True, 'port': 0, 'is_buffered': False, 'do_feedback': True, 'wait_complete': True, 'subcommand': 81, 'params': b'0000'})
```
although the details of the error, such as port number and payload, may change from run to run.

Is there perhaps a way in pylgbst to get around this?

undera, 2023-03-02:
Your code seem to violate its expectation of strictly one sync command running at a time. I did not block that from happening because there may be combination of sync and non-sync commands. So you can think of lib as non-thread-safe.

To overcome your issue, introduce a lock object that will be used to wrap the LED state change commands as well as other commands. You just need to wrap exact command invocations.

cuernodegazpacho, 2023-03-02:
Thanks! I will do that then.