undera/pylgbst #137: Remote WARNING:peripherals:Attempt to unsubscribe while port value updates are off: RemoteButton on port 0x0

Eggmanplant, 2024-06-26:
WARNING:peripherals:Attempt to unsubscribe while port value updates are off: RemoteButton on port 0x0
This f**** up me when trying to use the remote.

undera, 2024-06-26:
Hi
It's a bit short of a context. Is the warning itself annoying or some functionality does not work?

The warning would normally only produce if you try to call `unsubscribe()` on a port that you did not `subscribe()` first. 

Without seeing your code it is hard to guess the possible reason.

Eggmanplant, 2024-07-02:
The program stops. This error is from the controller demo.

undera, 2024-07-02:
Which controller demo do you mean?

Eggmanplant, 2024-11-02:
I mean demo_handset.py from examples. Sorry for late reply, I was focused on moving to Codeberg.

undera, 2024-11-03:
Looking at the code of that demo, the unsubscribe only happens after 30 seconds of sleep in program. The message would mean that the subscribe were not successfull. 
You'd need to look at detailed debug log of communication to tell what exactly happens.