undera/pylgbst #138: technic hub suport plans?

twin1nap, 2024-11-16:
any plans to add support for the technic hub or any tips for me to add it myself, like where to start?

undera, 2024-11-16:
Theoretically, any PoweredUp hub can be used with this library, use `Hub` class: https://github.com/undera/pylgbst/blob/master/pylgbst/hub.py#L49

You can also subclass it to create a class that handles specifics of some hub type, like done in: https://github.com/undera/pylgbst/blob/master/pylgbst/hub.py#L338

twin1nap, 2024-11-16:
Im still a bit of a beginner in Python, so I dont understand the code fully, so how would I then control a hub using the hub class or make the subclass?

twin1nap, 2024-11-16:
After reading the smarthub subclass a few times, I came to the conclusion I can just copy and paste it mostly and modify it for my own needs, but how would I get the port numbers for all the ports and for the combined ports like port_AB?

undera, 2024-11-17:
The process of discovering port information is quite easy. You just create a piece of code that connects to your Hub, and you see the port and peripheral numbers printed to the console.

twin1nap, 2024-11-17:
I can't get my hub to connect to my laptop for some reason when using the hub class to connect to it. I don't know if I am doing something wrong or something else is the problem, but I can't get it to connect. Here is my code:
`from pylgbst import get_connection_bleak
from pylgbst.hub import Hub

hub_mac = "90:84:2B:4A:12:8D"
conn = get_connection_bleak(hub_mac=hub_mac)
hub = Hub(conn)`

twin1nap, 2024-11-17:
Still just stops after starting it like before for some reason.

twin1nap, 2024-11-17:
I connected the hub using the movehub class and put the logging in a while loop, and now it works. I just have to find a way to get the combined ports for controlling port A and be at the same time.

undera, 2024-11-17:
Actually, the message above might be wrong and you might need to do what is called "Virtual Port": https://lego.github.io/lego-ble-wireless-protocol-docs/#virtual-port-setup

Also see corresponding message type: https://github.com/undera/pylgbst/blob/master/pylgbst/messages.py#L714

twin1nap, 2024-11-17:
Before you responded, I already had a virtual port to work using a past conversation you had that I found [here](https://github.com/undera/pylgbst/issues/122), but I couldn't get it to work when I put it in the subclass I created so that I later could just use port_AB, which then would be the virtual port for ports A and B. Do you know how I could make that work?

undera, 2024-11-17:
You can create an override `__init__` constructor in hub subclass, and call the virtual port right after parent `__init__`. You need to make sure that at the time of creating virtual port, the both motors are already recognized.

undera, 2024-11-17:
Since the type of your Hub uses only externally attached motors, you'd have to create a waiting loop that would check that motors are there, and once this condition is met -- perform the virtual port creation. A similar waiting is done in MoveHub here https://github.com/undera/pylgbst/blob/master/pylgbst/hub.py#L269, though it waits the pre-defined onboard devices.

twin1nap, 2024-11-17:
I got it to work. I'll share the subclass later, when it's fully done, so you can add it to the library if you want to add it. It won't have support for the internal technical hub sensors.

```
Have no dedicated class for peripheral type 0x39 (TECHNIC_MEDIUM_HUB_ACCELEROMETER) on port 0x61
Have no dedicated class for peripheral type 0x3a (TECHNIC_MEDIUM_HUB_GYRO_SENSOR) on port 0x62
Have no dedicated class for peripheral type 0x36 (TECHNIC_MEDIUM_HUB_GEST_SENSOR) on port 0x64
```
and I personally have no need to figure out how to add it as I don't need those sensors myself.

twin1nap, 2024-11-17:
I also just noticed that if I also want to have a virtual port for port_C and port_D, then that will be more complicated because that virtual port will either have the same or a different port depending on whether you already have a virtual port for port_A and port_B.

undera, 2024-11-17:
Cool progress! 
Please share your findings here, it will help others.

twin1nap, 2024-11-18:
After trying a bit, I found out it was a bit harder than I thought. I suspect I have it working at the end of the week.

twin1nap, 2024-11-18:
now added support for a virtual port for ports C and D if you don't have one for Ports A and B, 2 virtual ports will be added as soon as I can access two morots I have that are in a build set. here is the current code(comments at changes):

```
class TechnicHub(Hub): #porvided by twin1nap
    """
    Class implementing Lego TechnicHub specifics
    https://www.lego.com/en-pt/product/technic-hub-88012

    :type led: LEDRGB
    :type current: Current
    :type voltage: Voltage
    :type port_A: Peripheral
    :type port_B: Peripheral
    :type port_C: Peripheral
    :type port_D: Peripheral
    :type port_AB: Virtual Port
    """

    DEFAULT_NAME = "Technic Hub"

    # PORTS
    PORT_A = 0x00
    PORT_B = 0x01
    PORT_C = 0X02
    PORT_D = 0x03
    PORT_VIRTUAL1 = 0X10 # changed name
    PORT_VIRTUAL2 = None # this added. port still unkown, not able to test for it yet
    PORT_LED = 0x32
    PORT_CURRENT = 0x3B
    PORT_VOLTAGE = 0x3C

    def __init__(self, connection=None):
        if connection is None:
            connection = get_connection_auto(hub_name=self.DEFAULT_NAME)

        super().__init__(connection)

        self.button = Button(self)
        self.led = None
        self.port_A = None
        self.port_B = None
        self.port_C = None
        self.port_D = None
        self.port_AB = None
        self.port_CD = None # this added
        self.current = None
        self.voltage = None

        self._wait_for_devices()
        self._wait_for_Port_AB()
        self._wait_for_Port_CD() # this added 

    def _wait_for_devices(self, get_dev_set=None):
        if not get_dev_set:
            get_dev_set = lambda: (self.led, self.current, self.voltage)

        for num in range(0, 100):
            devices = get_dev_set()
            if all(devices):
                log.debug("All devices are present: %s", devices)
                return
            log.debug("Waiting for builtin devices to appear: %s", devices)
            time.sleep(0.1)
        log.warning("Got only these devices: %s", get_dev_set())

    def _wait_for_Port_AB(self, get_dev_set=None):
        if not get_dev_set:
            get_dev_set = lambda: (self.port_A, self.port_B)

        for num in range(0, 10):
            devices = get_dev_set()
            if all(devices):
                log.debug("All devices are present: %s", devices)
                
                cmd_type = MsgVirtualPortSetup.CMD_CONNECT
                ports_to_combine = (self.PORT_A, self.PORT_B)
                msg_combine = MsgVirtualPortSetup(cmd_type, ports_to_combine)
                self.send(msg_combine)

                return
            log.debug("Waiting for devices to appear: %s", devices)
            time.sleep(0.1)
        log.warning("No virtual port AB added due to not enough devices atached to port_A and port_B")

    def _wait_for_Port_CD(self, get_dev_set=None): # this added
        if not get_dev_set:
            get_dev_set = lambda: (self.port_C, self.port_D)

        for num in range(0, 10):
            devices = get_dev_set()
            if all(devices):
                log.debug("All devices are present: %s", devices)
                
                cmd_type = MsgVirtualPortSetup.CMD_CONNECT
                ports_to_combine = (self.PORT_C, self.PORT_D)
                msg_combine = MsgVirtualPortSetup(cmd_type, ports_to_combine)
                self.send(msg_combine)

                return
            log.debug("Waiting for devices to appear: %s", devices)
            time.sleep(0.1)
        log.warning("No virtual port CD added due to not enough devices atached to port_C and port_D")

    

    # noinspection PyTypeChecker
    def _handle_device_change(self, msg):
        super()._handle_device_change(msg)
        if (
                isinstance(msg, MsgHubAttachedIO)
                and msg.event != MsgHubAttachedIO.EVENT_DETACHED
        ):
            port = msg.port
            if port == self.PORT_A:
                self.port_A = self.peripherals[port]
            elif port == self.PORT_B:
                self.port_B = self.peripherals[port]
            elif port == self.PORT_C:
                self.port_C = self.peripherals[port]
            elif port == self.PORT_D:
                self.port_D = self.peripherals[port]
            elif port == self.PORT_VIRTUAL1:
                if self.port_A == None or self.port_B == None: #-|
                    self.port_CD = self.peripherals[port]      # |
                else:                                          # | this added
                    self.port_AB = self.peripherals[port]      #-|
            elif port == self.PORT_LED:
                self.led = self.peripherals[port]
            elif port == self.PORT_CURRENT:
                self.current = self.peripherals[port]
            elif port == self.PORT_VOLTAGE:
                self.voltage = self.peripherals[port]
```

undera, 2024-11-18:
Impressive outcome and progress, thanks for sharing it